realsenseai/librealsense #12958: Installation HELP!!

sai-srinivasan-v, 2024-05-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (-) | 
| Operating System & Version |   {LINUX(ubuntu) | 
| Kernel Version (Linux Only)    |  (6.8.0-31-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0}                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello, Ive been trying to install and have the L515 camera up and running for the entire day.  I have installed the source code from https://github.com/IntelRealSense/librealsense/releases/tag/v2.47.0 if there is someone that can help me install and have this camera running id be extremely grateful.

MartyG-RealSense, 2024-05-24:
Hi @sai-srinivasan-v  Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

sai-srinivasan-v, 2024-05-24:
Hi @MartyG-RealSense  I have done that. The camera is now being read when i enter 'lsusb'. Whats next?

MartyG-RealSense, 2024-05-24:
Is the camera also able to be read if you use the command `rs-enumerate-devices -s`

If it is, can you then run the **RealSense Viewer** command with the command `realsense-viewer`

sai-srinivasan-v, 2024-05-24:
'rs-enumerate-devices -s' returns No device detected. Is it plugged in?

MartyG-RealSense, 2024-05-24:
Thank you.  Which method did you use to build the 2.47.0 SDK from source code?

sai-srinivasan-v, 2024-05-24:
I cloned it from 'https://github.com/IntelRealSense/librealsense.git'  and then 
2.  'sudo ./scripts/setup_udev_rules.sh && sudo apt install -y cmake pkg-config libusb-1.0-0-dev libglfw3-dev'
3. mkdir build && cd build
4.  cmake ..
5.  sudo make install
6 . ./bin/rs-enumerate-devices
During 6 i keep getting an error that says build does not have /bin
I would be happy to follow a new set of instructions from scratch if you can guide me

sai-srinivasan-v, 2024-05-24:
Hi @MartyG-RealSense I followed the guide and this is what i see. 
[image]

MartyG-RealSense, 2024-05-24:
The -j value in the 'make' command, such as **make -j8**, can cause a random crash when building on some computing devices.  You can test whether the j-value is the cause by leaving it out and just using 'make', though the SDK will take longer to build.

`sudo make uninstall && make clean && make && sudo make install`

sai-srinivasan-v, 2024-05-24:
I tried. But i still see the errors or warnings scattered here and there is that ok? If so, what next?

sai-srinivasan-v, 2024-05-25:
Hi @MartyG-RealSense?

MartyG-RealSense, 2024-05-25:
Thanks very much for your patience.

As you are using kernel 6.8, does that mean that you are using the new Ubuntu 24.04?  If you are using 24.04 then the most recent Ubuntu and kernel versions currently supported by the RealSense SDK are Ubuntu 22.04 and kernel 6.5.  

In theory, using the RSUSB = true installation method should mean that the SDK is not dependent on particular Linux versions or kernel versions.  At the time of writing this though there have not yet been reports of other RealSense users using 24.04 and kernel 6.8, so it is not yet known if there is something in 24.04 that might cause problems with installation of the SDK.

sai-srinivasan-v, 2024-05-25:
I also have a M2 Mac. Could you help me with that instead? @MartyG-RealSense

MartyG-RealSense, 2024-05-25:
Installation on an Apple Silicon Mac is currently more difficult than on Linux because RealSense cameras are having issues with recent MacOS versions such as Ventura and Sonoma (RealSense has not officially supported Mac since 2018).  The last MacOS version that RealSense has been able to work with is Monterey - please see https://github.com/IntelRealSense/librealsense/issues/10676#issuecomment-1185203673

A RealSense user did recently succeed in getting the camera working in **Asahi Linux** in a VM on an Apple Silicon Mac and shared an instruction guide.

https://github.com/HasanTheSyrian/librealsense-hasan/blob/master/doc/installation_asahi.md

sai-srinivasan-v, 2024-05-25:
So if i were to run it on ubuntu (which is comparatively easier) id have to downgrade my kernel?
If so what version should i downgrade it to?

MartyG-RealSense, 2024-05-25:
6.5 is currently the newest supported kernel version in SDK 2.55.1.  However, if you were aiming to build SDK version 2.47 then the most recently supported kernel would likely be the older 5.0.2.

If the SDK is built from source code with the **-DFORCE_RSUSB_BACKEND=TRUE** flag included in the CMake build instruction though then the Linux version or kernel version being should should not matter, as an RSUSB build of the SDK bypasses the kernel and is not dependent on a particular Linux version or kernel version.

sai-srinivasan-v, 2024-05-25:
I think i will install SDK 2.54.2 like you mentioned earlier. I have an Ubuntu 22.04 booted on UTM.  Can i follow the instructions on https://github.com/HasanTheSyrian/librealsense-hasan/blob/master/doc/installation_asahi.md ? If its something different please let me know and assistance would be helpful.

MartyG-RealSense, 2024-05-25:
SDK 2.54.2 is compatible with 22.04.  

If you encounter a problem when using the guide then I would recommend posting a comment to HasanTheSyrian at https://github.com/IntelRealSense/librealsense/issues/12503 as they created the guide.

sai-srinivasan-v, 2024-05-25:
I can also reach out to him through this thread right?

MartyG-RealSense, 2024-05-25:
If you put @ in front of HasanTheSyrian then you would tag him into this discussion, yes.

sai-srinivasan-v, 2024-05-25:
after entering 'cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true' 

I get the error message:

-- The CXX compiler identification is unknown
-- The C compiler identification is GNU 11.4.0
CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring incomplete, errors occurred!

@MartyG-RealSense  @HasanTheSyrian  Please Help!

MartyG-RealSense, 2024-05-25:
@HasanTheSyrian  I would clarify that @sai-srinivasan-v is attempting to use your Asahi Linux installation guide on their M2 Mac.

sai-srinivasan-v, 2024-05-25:
NO @MartyG-RealSense  I am using an ubuntu 22.04. I apologise for any confusion that i may have caused.

sai-srinivasan-v, 2024-05-25:
I would also like to let you know that i have fixed the error it threw 2 responses earlier.

MartyG-RealSense, 2024-05-25:
Okay, no problem @sai-srinivasan-v - thanks very much for clarifying my clarification.  :)

sai-srinivasan-v, 2024-05-25:
This is what i see after 
```
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
``` 

[image]

MartyG-RealSense, 2024-05-25:
The config.h error has been reported in the past with librealsense builds.  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-653942842 posts the solution that worked for them, which involves using the command `sudo apt-get install libusb-1.0-0-dev` before building the SDK.

HasanTheSyrian, 2024-05-25:
> I think i will install SDK 2.54.2 like you mentioned earlier. I have an Ubuntu 22.04.4 booted on UTM.  Can i follow the instructions on https://github.com/HasanTheSyrian/librealsense-hasan/blob/master/doc/installation_asahi.md ? If its something different please let me know and assistance would be helpful. 
> 
> 

Docs clearly advise not using VMs

MartyG-RealSense, 2024-05-25:
Thanks so much @HasanTheSyrian for your advice.  The recommendation against VMs dates back to when the RealSense SDK 2.0 was first launched back at the end of 2017 and support in VMs for handling USB devices was less common.  Most people in the present day manage to use RealSense with a VM successfully with Linux as long as the VM being used provides USB emulation.

HasanTheSyrian, 2024-05-25:
> Installation on an Apple Silicon Mac is currently more difficult than on Linux because RealSense cameras are having issues with recent MacOS versions such as Ventura and Sonoma (RealSense has not officially supported Mac since 2018). The last MacOS version that RealSense has been able to work with is Monterey - please see [#10676 (comment)](https://github.com/IntelRealSense/librealsense/issues/10676#issuecomment-1185203673)
> 
> A RealSense user did recently succeed in getting the camera working in **Asahi Linux** in a VM on an Apple Silicon Mac and shared an instruction guide.
> 
> https://github.com/HasanTheSyrian/librealsense-hasan/blob/master/doc/installation_asahi.md

Asahi Linux runs on bare metal, not on a VM. 

Again, credit to the insane work done by Asahi reverse engineering all the drivers.

sai-srinivasan-v, 2024-05-25:
@MartyG-RealSense I think ive gotten past that one as well. But is this alright 
[image]

MartyG-RealSense, 2024-05-25:
The top three warnings about deprecation can be ignored.

The errors make me wonder if GLFW has been installed.

`sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`

sai-srinivasan-v, 2024-05-25:
i think it is
[image]

HasanTheSyrian, 2024-05-25:
> @MartyG-RealSense I think ive gotten past that one as well. But is this alright <img alt="Screenshot 2024-05-25 at 5 02 54 PM" width="1441" src="https://private-user-images.githubusercontent.com/55703998/333796244-76efb45e-8099-485c-a5bc-2a2c6f34e049.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2Mzc1ODgsIm5iZiI6MTcxNjYzNzI4OCwicGF0aCI6Ii81NTcwMzk5OC8zMzM3OTYyNDQtNzZlZmI0NWUtODA5OS00ODVjLWE1YmMtMmEyYzZmMzRlMDQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI1VDExNDEyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg0OWE5NGVhNTRlMmU5ZWE1MjVlOTJmZTQzYWNiYzhiY2ZhZDI2MWZiZDM1M2YxZGIwZmRjY2I3YWRjYWU1ZTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DTGGEfKhec2VBOJn4BxPdmXwn9ohPti8IZ_LpoOHSGM">

Go to the files where there are undefined references and check the libraries. If they are not installed, install them.

Some libraries might not be built for ARM and you might have to build them yourself.

sai-srinivasan-v, 2024-05-25:
how do i do that @HasanTheSyrian ?

MartyG-RealSense, 2024-05-25:
@HasanTheSyrian  Just wanted to add that your willingness to provide help to @sai-srinivasan-v is fantastic  :)

HasanTheSyrian, 2024-05-25:
> how do i do that @HasanTheSyrian ?

The error message tells you where the file is. Go there and look at what libraries it's including. Check if they are installed.

HasanTheSyrian, 2024-05-25:
@MartyG-RealSense btw, did I request to merge my PR? I can't tell I'm not the best at version management.

MartyG-RealSense, 2024-05-25:
@HasanTheSyrian  It looks as though there are some failing checks that are blocking on https://github.com/IntelRealSense/librealsense/pull/12763 though it might be related to some other commits rather than your Asahi guide.

sai-srinivasan-v, 2024-05-25:
Hey @HasanTheSyrian , I'm sorry to keep bothering you i believe these are the libraries? 
[image]

sai-srinivasan-v, 2024-05-25:
examples/software-device/CMakeFiles/rs-software-device.dir/build.make:103: examples/software-device/rs-software-device] Error 1

should i be going into these directories? @HasanTheSyrian  @MartyG-RealSense

MartyG-RealSense, 2024-05-25:
I don't know.  I assume that the error is occurring because of the **-DBUILD_EXAMPLES=true** flag which tells CMake to build the RealSense example programs.

HasanTheSyrian, 2024-05-25:
> Hey @HasanTheSyrian , I'm sorry to keep bothering you i believe these are the libraries? <img alt="Screenshot 2024-05-25 at 5 53 10 PM" width="766" src="https://private-user-images.githubusercontent.com/55703998/333799425-440d68a0-ab0c-41ed-bc71-d2f459f9aae4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2NDA2MjMsIm5iZiI6MTcxNjY0MDMyMywicGF0aCI6Ii81NTcwMzk5OC8zMzM3OTk0MjUtNDQwZDY4YTAtYWIwYy00MWVkLWJjNzEtZDJmNDU5ZjlhYWU0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI1VDEyMzIwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4ZjQ1NGJiYzhlYmE0OWUzMjcxMDc0YTI0MzRjYjAyYjUzNTAzM2FhYmJmOTYxMDQ2MjJjNjY3MTVmZTQwYjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Kg_eVbiu-sRvWeOZry6Gjcg8Ui8lhMeyMhOqkFZB0pQ">

Yes. Look inside `example.hpp` itself. Also, try ignoring these files if you're not going to use that specific example. I don't know how to make CMake do that.

sai-srinivasan-v, 2024-05-25:
this is the example.hpp file. what next? @HasanTheSyrian 
[image]

HasanTheSyrian, 2024-05-25:
> Also, try ignoring these files if you're not going to use that specific example. I don't know how to make CMake do that.

sai-srinivasan-v, 2024-05-25:
Okay i'll ignore it then. i also do not have the realsense viewer.

sai-srinivasan-v, 2024-05-25:
Hi @MartyG-RealSense i somehow brought it to this. but when i try launching realsense-viewer it returns 'command not found' 
[image]

sai-srinivasan-v, 2024-05-25:
Hi @MartyG-RealSense  i got into realsense-viewer but i still cannot connect my camera 
[image]

sai-srinivasan-v, 2024-05-25:
hi @MartyG-RealSense and @HasanTheSyrian i got the camera to work. Thank you guys so much.

sai-srinivasan-v, 2024-05-25:
Feel free to close this thread

MartyG-RealSense, 2024-05-25:
That's excellent news, @sai-srinivasan-v - thanks very much for the update!

@HasanTheSyrian - thanks so much again for your time and generosity in giving advice to @sai-srinivasan-v   :)

closed 2024-05-25 by MartyG-RealSense

sai-srinivasan-v, 2024-06-13:
Hey @MartyG-RealSense  I had some of my friends try to install the Realsense Viewer and even i couldnt help them out. I would like to request you to make an updated step by step tutorial for anyone thats facing this difficulty.  Please do the needful.

MartyG-RealSense, 2024-06-14:
One of my Intel RealSense colleagues is looking at the problems of installation on Apple Silicon 'M' models of Mac to see whether an instruction guide is possible.  There is not news to report about this effort at the time of writing this though.

sai-srinivasan-v, 2024-06-21:
Hey Marty! I hope you're doing good. I got the camera to work except now when i try to open the realsense viewer it threw a segmentation error. And also i have a python script that uses pyrealsense 2 though i can read the camera through the `rs-enumerate-devices` command the python script does not detect the device.

MartyG-RealSense, 2024-06-21:
This segmentation error with realsense-viewer on MacOS is something that other RealSense users with Apple Silicon Macs have experienced.  There is not currently a known solution for this.

At https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-1048069662 a user said "I can run the realsense-viewer from the terminal with sudo, once I add the app manually to the accessibility tab in system preferences/security".

A RealSense user created a pre-built version of the pyrealsense2 wrapper for Mac at https://github.com/IntelRealSense/librealsense/issues/9687 though it has been reported to have problems with recent MacOS versions such as Ventura and Sonoma.

sai-srinivasan-v, 2024-06-21:
These are all issues that are from an Ubuntu Machine.

MartyG-RealSense, 2024-06-21:
You have an M2 Mac but were attempting to use Ubuntu or Asahi Linux earlier in this discussion.  Was it MacOS that you are having the realsense-viewer segmentation error and Python non-detection of the camera on, please?

sai-srinivasan-v, 2024-06-21:
True. But I got it to work on Ubuntu(UTM) and things were fine. And then Realsense threw a segmentation error. Everything i discussed here happening on a Ubuntu Machine. I quit trying to install real sense on my Mac.

MartyG-RealSense, 2024-06-21:
I researched the history of segmentation errors in realsense-viewer on Ubuntu.  In almost all cases there was no solution, unfortunately.  The one user who claimed that they found a solution said that they closed down the Ubuntu terminal (which can be done by selecting Exit) and then restarted the terminal (you can open a terminal with the Ctrl+Alt+T keyboard shortcut).  Then they input the **realsense-viewer** command to launch it.

sai-srinivasan-v, 2024-06-22:
Yeah that simply launches realsense and it does not detect the camera. Also i tried running the python script as a root user and received pipeline = rs.pipeline()
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'
Can you please help me fix this?

sai-srinivasan-v, 2024-06-22:
I moved it into the python directory inside wrappers and it works fine. Are there programs that are written by other users that i can use though? @MartyG-RealSense

[6 more replies]