realsenseai/librealsense #12960: Librealsense 2.55.x breaks L515 but the resulting error in the sdk is "No device detected. Is it plugged in?"

dirksavage88, 2024-05-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.19.125)                                         | 
| Platform                                 | voxl2  |
| SDK Version                            |  2.55.x }                          | 
| Language                            |  C/C++ }                          | 
| Segment			|  Robotics }                          | 

This error that is reported in the 2.55.x releases where the L515 is no longer supported should report an error that is more specific, otherwise users that are installing the latest SDK will attempt to troubleshoot the device connection instead. The error resulting from 'rs-enumerate-devices' should detect an L515 and just report that the model is not supported by 2.55.1 onward. Instead here is the output from rs-enumerate-devices:

`~# rs-enumerate-devices 
No device detected. Is it plugged in?`

Users with docker images pulling the librealsense SDK release without knowing it has dropped L515 support will run into this issue and spend way too much time just to figure out they need to switch to an older librealsense SDK version.

MartyG-RealSense, 2024-05-25:
Hi @dirksavage88  Thanks very much for your request for a message regarding removal of support for L515.  I have highlighted it to my Intel RealSense colleagues.

MartyG-RealSense, 2024-06-04:
Hi again @dirksavage88  My Intel RealSense colleagues are aware of the L515 messaging issue that you raised and are looking at it.

dirksavage88, 2024-06-10:
Thanks for your follow up @MartyG-RealSense

MartyG-RealSense, 2024-06-16:
As the Intel RealSense team is aware of the L515 messagig issue and are looking at it, I will close this case.  Thanks again for the report!

closed 2024-06-16 by MartyG-RealSense