realsenseai/librealsense #12962: Impossible to remove librealsense-dkms

mattiadutto, 2024-05-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Ubuntu 22.04.4  | 
| Kernel Version (Linux Only)    |  6.5.0-1023-oem                                       | 
| Platform                                 | Dell XPS 13 Plus |
| SDK Version                            |  Not aplicable                   | 
| Language                            |  C++ / Python                      | 
| Segment			|  Robot                          | 

### Issue Description
I tried multiple times in the last months to remove the librealsense-dkms package, but unfortunately, I'm not able to remove it moreover I also tried to install some new packages but I'm not able to do it since when I'm installing a new package apt tries to remove the package.

Error when I try to remove _librealsense-dkms_:
```bash
mattia@xps-13-plus:~$ sudo apt remove librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED
  librealsense2-dkms
0 to upgrade, 0 to newly install, 1 to remove and 3 not to upgrade.
1 not fully installed or removed.
After this operation, 10.5 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 505005 files and directories currently installed.)
Removing librealsense2-dkms (1.3.26-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module videodev is in use.
dpkg: error processing package librealsense2-dkms (--remove):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 librealsense2-dkms
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Result of lsmod | grep videodev
```
videodev              364544  7 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov01a10,v4l2loopback,intel_ipu6_isys
mc                     86016  6 v4l2_async,videodev,videobuf2_v4l2,ov01a10,intel_ipu6_isys,videobuf2_common
```

When I try to install a ROS2 package:
```
mattia@xps-13-plus:~$ sudo apt install ros-$ROS_DISTRO-rviz-imu-plugin
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED
  librealsense2-dkms
The following NEW packages will be installed
  ros-humble-rviz-imu-plugin
0 to upgrade, 1 to newly install, 1 to remove and 3 not to upgrade.
1 not fully installed or removed.
Need to get 0 B/170 kB of archives.
After this operation, 9,594 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 505005 files and directories currently installed.)
Removing librealsense2-dkms (1.3.26-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module videodev is in use.
dpkg: error processing package librealsense2-dkms (--remove):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 librealsense2-dkms
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Apt list result:
```
mattia@xps-13-plus:~$ apt list -i | grep realsense

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

librealsense2-dkms/jammy,now 1.3.26-0ubuntu1 all [installed]
ros-humble-librealsense2/jammy,now 2.55.1-1jammy.20240516.151706 amd64 [installed]
ros-humble-realsense2-camera-msgs/jammy,now 4.54.1-1jammy.20240517.160706 amd64 [installed]
ros-humble-realsense2-camera/jammy,now 4.54.1-1jammy.20240517.191635 amd64 [installed]
```

I have already tried what is suggested [here](https://github.com/IntelRealSense/librealsense/issues/10097), but I don't have any /etc/udev/rules/99-librealsense... file.

MartyG-RealSense, 2024-05-26:
Hi @mattiadutto  The recommended method of removing a specific librealsense package or all RealSense related packages is to use a **purge** command, as described at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

The **/etc** rule is installed when building librealsense from source code instead of DKMS packages, so the /etc rule would not be on your computer if you have not performed a source code build of the SDK.

mattiadutto, 2024-05-26:
Ho @MartyG-RealSense thanks for the fast reply I tried and this is the result:
```
mattia@xps-13-plus:~$ dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
(Reading database ... 505005 files and directories currently installed.)
Removing librealsense2-dkms (1.3.26-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module videodev is in use.
dpkg: error processing package librealsense2-dkms (--purge):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: dependency problems prevent removal of ros-humble-realsense2-camera:
 ros-humble-rtabmap-examples depends on ros-humble-realsense2-camera.

dpkg: error processing package ros-humble-realsense2-camera (--purge):
 dependency problems - not removing
dpkg: dependency problems prevent removal of ros-humble-realsense2-camera-msgs:
 ros-humble-realsense2-camera depends on ros-humble-realsense2-camera-msgs.

dpkg: error processing package ros-humble-realsense2-camera-msgs (--purge):
 dependency problems - not removing
dpkg: dependency problems prevent removal of ros-humble-librealsense2:
 ros-humble-realsense2-camera depends on ros-humble-librealsense2.

dpkg: error processing package ros-humble-librealsense2 (--purge):
 dependency problems - not removing
Errors were encountered while processing:
 librealsense2-dkms
 ros-humble-realsense2-camera
 ros-humble-realsense2-camera-msgs
 ros-humble-librealsense2
```

MartyG-RealSense, 2024-05-26:
Have you deleted `lib/udev/rules.d/60-librealsense2-udev-rules.rules` please?  If you have then this should not be done as it breaks linkages to the DKMS packages.

If your udev rule is missing then you could try reinstalling it with the command below.

`sudo apt-get install librealsense2-udev-rules`

mattiadutto, 2024-05-26:
I don't have that file, but I don't think that I removed it manually.

I'm not able to run the suggested command:
```
mattia@xps-13-plus:~$ sudo apt-get install librealsense2-udev-rules
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED
  librealsense2-dkms
The following NEW packages will be installed
  librealsense2-udev-rules
0 to upgrade, 1 to newly install, 1 to remove and 3 not to upgrade.
1 not fully installed or removed.
Need to get 0 B/11.8 kB of archives.
After this operation, 10.5 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 505005 files and directories currently installed.)
Removing librealsense2-dkms (1.3.26-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module videodev is in use.
dpkg: error processing package librealsense2-dkms (--remove):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 librealsense2-dkms
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MartyG-RealSense, 2024-05-26:
I can only find one previous case at https://github.com/IntelRealSense/librealsense/issues/10097 that has the error **Reloading original modules into kernel...modprobe: FATAL: Module videodev is in use** and it says that you have already viewed it.

It may be easier to just wipe everything on the computer including Ubuntu to remove the packages that way and perform a fresh reinstall of Ubuntu and librealsense.  I can appreciate that doing so is not always possible though if you have data that you cannot lose.

mattiadutto, 2024-05-26:
Thanks I would like to avoid wiping everything on my pc, but probably I will do that.

MartyG-RealSense, 2024-05-26:
May I also confirm whether you have applied a kernel patch script to your kernel?  You do not need to do so with a package installation as the patch is already bundled inside the packages.

MartyG-RealSense, 2024-05-26:
If the videodev module is stated to be in use then what happens if you unload it and then reload it with the command below?

`sudo modprobe -r videodev && sudo modprobe videodev`

mattiadutto, 2024-05-26:
This the output:
```
mattia@xps-13-plus:~$ sudo modprobe -r videodev && sudo modprobe videodev
[sudo] password for mattia: 
modprobe: FATAL: Module videodev is in use.
```

MartyG-RealSense, 2024-05-26:
A PC wipe may be unfortunately necessary if there is a hidden tangle somewhere on the system that is preventing this issue from being fixed. 

The alternative would be to leave the librealsense packages alone and build librealsense from source code instead.  A fast way to do this is to use a build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

mattiadutto, 2024-05-26:
I will go for the wipe also because I'm not able to install new packages. Installing the packages from the source each time is a little bit annoying.

MartyG-RealSense, 2024-05-26:
Okay, I wish you the best of luck!

MartyG-RealSense, 2024-06-10:
Hi @mattiadutto  Do you require further assistance with this case, please?  Thanks!

MarioCavero, 2024-06-12:
@mattiadutto In case you did not go for the wipe yet, check out my resolved issue (which I'd say is very similar). 
`lsmod | grep videodev` gives a list of modules. In my case, all of them couldn't be removed as they were being used by `v4l2-relayd` (for you it might be different, but keep checking recursively with `lsmod` to see what is using the '_cannot_be_removed_modules_').
 `sudo systemctl stop v4l2-relayd` allowed me to remove the final modules, `videodev` was not being in use anymore, and `realsense2` managed to get installed and no more dkms problems!

MartyG-RealSense, 2024-06-12:
Thanks so much @MarioCavero for sharing what worked for you in your particular situation!

mattiadutto, 2024-06-14:
Good morning, sorry for the late response, I already did the PC wipe, I just re-installed the librealsense ros2 package and everything went smoothly. I'm not sure but I think I tried to stop the module but didn't help. 
By the way thanks to both of you.

MartyG-RealSense, 2024-06-14:
You are very welcome, @mattiadutto   :)

As you have achieved a solution, I will close this case.  Thanks again @MarioCavero for your help!

closed 2024-06-14 by MartyG-RealSense