realsenseai/librealsense #12968: Sometimes depth stream blocked on single-frame pipe

reuksv, 2024-05-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Ubuntu 20}  |              | 

### Issue Description
A simple code of this kind sometimes (perhaps after hundreds or thousands of iterations) blocks the depth stream and prevents new frames from being received

```
//config enabled only depth stream
pipe->start(*config);
pipe->wait_for_frames();
pipe->stop();
```

In realsense viewer, if you enable the display of depth frames, you will see the message "No frames recieved"
The only thing that helps is the physical plugging of the wire.
I checked on several different sensors and wires.
Is there any way to avoid this behavior?

MartyG-RealSense, 2024-05-28:
How does it behave if you use a **frames** variable with wait_for_frames() and then use the **keep()** instruction to store the frames in the computer's memory?

```
//config enabled only depth stream
pipe->start(*config);
frames = pipe->wait_for_frames();
frames->keep();
pipe->stop();
```

reuksv, 2024-05-29:
> Hi @reuksv How does it behave if you use a **frames** variable with wait_for_frames() and then use the **keep()** instruction to store the frames in the computer's memory?
> 
> ```
> //config enabled only depth stream
> pipe->start(*config);
> frames = pipe->wait_for_frames();
> frames->keep();
> pipe->stop();
> ```

Hello @MartyG-RealSense 
No changes have occurred.

MartyG-RealSense, 2024-05-29:
I would next recommend using an Ubuntu system monitoring tool such as **htop** to monitor the available memory on the computer whilst the program is running and see whether it is progressively declining over time as the program repeatedly iterates.  Such a reduction is known as a **memory leak** and can result in a program becoming unstable or crashing once the free memory becomes low enough.

MartyG-RealSense, 2024-05-30:
Does the issue still occur if you put a value in the brackets of the wait_for_frames instruction such as 1000?

`pipe->wait_for_frames(1000);`

reuksv, 2024-05-31:
The problem still remains

MartyG-RealSense, 2024-05-31:
How does the program behave if the stop instruction is commented out and the pipeline is permitted to run continuously without multiple start-stops?

reuksv, 2024-05-31:
Works good

MartyG-RealSense, 2024-05-31:
If your project requires repeated start-stop cycles instead of running continuously then you could try putting a short **sleep** command before the pipe start line to see whether a small time wait before starting the pipeline also helps to reduce the likelihood of a problem occurring.

MartyG-RealSense, 2024-06-01:
Next, please try placing a camera reset instruction before the pipe start line in your script using the Python code below.  This should have the same effect as performing an unplug-replug of the USB cable.

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

MartyG-RealSense, 2024-06-06:
Hi @reuksv  Do you require further assistance with this case, please?  Thanks!

reuksv, 2024-06-06:
@MartyG-RealSense Unfortunately, I haven't had time to check the last tip yet. Please do not close the ticket yet

MartyG-RealSense, 2024-06-06:
It's no problem at all to keep it open.  Thanks very much for the update!

reuksv, 2024-07-18:
Thank you. I think hardware reset should help, but I managed to test it only in the realsense viewer interface. You can close the ticket.

closed 2024-07-18 by MartyG-RealSense