realsenseai/librealsense #12976: SDK(realsense viewer) vesion change

Rhymakers, 2024-05-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (None) | 
| Operating System & Version |   {Linux (Ubuntu 22)}  | 
| Kernel Version (Linux Only)    |  (6.5.0-35-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  {None}                          | 
| Language                            |  {pcl }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
I have a realsense l515. I want to record data through the realsense viewer, but I saw that l515 is no longer supported in the latest version of SDK(viewer). Can I see the L515 through the viewer by lowering the version of SDK that is already installed?

MartyG-RealSense, 2024-05-29:
Hi @Rhymakers  You could access your L515 by building version 2.54.2 of the SDK from source code using CMake.  The source code can be downloaded as a compressed zip file from the 'Assets' file list at the link below and then extract the SDK code as a folder from the zip and build it.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

Instructions for the build process can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

**********

If you prefer to install from packages then an alternative is to list past versions of the packages with the command **apt-cache showpkg librealsense2** and then use the package installation process that you previously used but use the package names for the 2.54.2 version that you obtain from the showpkg command..

Rhymakers, 2024-05-29:
Thank you for your response!
Actually, I tried another way and succeeded, but I was waiting for a formal response.
I followed the link below with the 2.54.2 zip file.
https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide
I hope it can be a reference to someone.

P.S. I think it's the same way you sent it

MartyG-RealSense, 2024-05-29:
You are very welcome!  It's great to hear that you succeeded.  :)

closed 2024-05-29 by Rhymakers