realsenseai/librealsense #12977: Error when get distance from depth frame

hungtrieu07, 2024-05-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Windows 10  |                                      | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  Python                          | 
| Segment			|  Other                          | 

### Issue Description
Hi, I have an issue when using the camera to capture both RGB frame and depth frame. I'm developing an app for pose measurement on person. My backend are running with Python, frontend are JS. Here is the detail: I have a button on frontend to send a signal to backend. In backend, it will get the RGB frame and depth frame, but sometimes, it will get an error:

```
Traceback (most recent call last):
  File "E:\dev\height-measurement\app.py", line 412, in process_seg_truoc
    knee_dict = measure.distance_bet_2_knee_new(
  File "E:\dev\height-measurement\general_utils.py", line 88, in distance_bet_2_knee_new
    self.calculate_distance(
  File "E:\dev\height-measurement\general_utils.py", line 24, in calculate_distance
    udist = self.depth_frame.get_distance(ixy[0], ixy[1])
RuntimeError: out of range value for argument "x"
```

This is ```calculate_distance.py```:

```
def calculate_distance(self, ixy, xy):
        udist = self.depth_frame.get_distance(ixy[0], ixy[1])
        vdist = self.depth_frame.get_distance(xy[0], xy[1])
        
        min_dist = min(udist, vdist)

        point1 = rs.rs2_deproject_pixel_to_point(
            self.color_intrin, [ixy[0], ixy[1]], min_dist
        )
        point2 = rs.rs2_deproject_pixel_to_point(
            self.color_intrin, [xy[0], xy[1]], min_dist
        )

        dist = math.sqrt(
            math.pow(point1[0] - point2[0], 2)
            + math.pow(point1[1] - point2[1], 2)
            + math.pow(point1[2] - point2[2], 2)
        )

        return dist
```

MartyG-RealSense, 2024-05-29:
Hi @hungtrieu07  I hope that https://github.com/IntelRealSense/librealsense/issues/10380 will be a helpful reference for your problem.

hungtrieu07, 2024-06-06:
> Hi @hungtrieu07 Do you require further assistance with this case, please? Thanks!

Thanks for your reply, I'm fixed that error and now the app run smoothly.

closed 2024-06-06 by hungtrieu07

MartyG-RealSense, 2024-06-06:
It's great to hear that you succeeded.  Thanks very much for the update!