realsenseai/librealsense #12981: Intel 3d camera crash on camera start via usb

anupama86, 2024-05-29:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-05-29:
Hi @anupama86  Does the error appear when the RealSense Viewer first opens, or does it appear when an option is selected?

For example, this error sometimes occurs when trying to access the RGB camera from the **Add Source** option, like in https://github.com/IntelRealSense/librealsense/issues/11338

anupama86, 2024-05-29:
Hi @MartyG-RealSense - It appears when I enable RGB Camera button on i.e. when it starts to stream data from camera. Thank you for sharing the link #11338 - I am observing same error. However, I cross checked the input data in laptop's settings, I am able to see the camera feed from intel's camera. When I use intel's real sense viewer v2.48.0 I get this error. 

Below is the snapshot of camera feed in settings:
[image]

Below is the snapshot of error in viewer:
[image]

MartyG-RealSense, 2024-05-29:
As the **Virtual Driver** message is displayed, is your RealSense camera model is an SR300 please?

When the Platform Camera is selected via the Add Source button, it will typically display a non-RealSense webcam camera such as the built-in webcam on a laptop.  Some laptop cameras work with the SDK and display their RGB image correctly but other laptop webcams do not display and display an error.    There is no way to know in advance whether a particular laptop's built-in webcam will be compatible with the Viewer.

closed 2024-06-14 by MartyG-RealSense