realsenseai/librealsense #12987: pyrealsense2 has no attribute context

anath93, 2024-05-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Ubuntu 22.04
| Kernel Version (Linux Only)    | 5.15                                        | 
| Platform                                 | Jetson Orin,Jetpack 6  |
| SDK Version                            |  2.55.1.0 }                          | 
| Segment			|  Robot                    | 

### Issue Description
Trying to calibrate IMU, using python rs-imu-calibration.py script but unable to.
[image]

Tried other posts for import pyrealsense2.pyrealsense as rs but still same.

anath93, 2024-05-30:
@MartyG-RealSense Installing pip install pyrealsense2 worked but still trying the script for imu calibration gives error 

[image]

SDK built using RUSB

MartyG-RealSense, 2024-05-30:
Hi @anath93  Is the camera detected if you launch the IMU calibration script in sudo admin permissions mode, please?

`sudo python3 rs-imu-calibration.py`

anath93, 2024-05-30:
@MartyG-RealSense 
[image]

anath93, 2024-05-30:
Another weird issue 
[image]

anath93, 2024-05-30:
@MartyG-RealSense the only way Motion module force pause goes away is if I use development branch from librealsense wrapper and I can see gyro values but still unable to use rs-imu calibrate.

MartyG-RealSense, 2024-05-31:
Does it make a difference if you edit the pyrealsense2 import line of the rs-imu-calibration.py script from this:

`import pyrealsense2 as rs`

to this:

`import pyrealsense2.pyrealsense2 as rs`

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py#L10

anath93, 2024-05-31:
No it does not, same result. Tried that already.

MartyG-RealSense, 2024-06-01:
If you have access to the RealSense Viewer tool, what is the value of Y on the 'Accel' panel if you enable the Motion Module in the Viewer's options side-panel?  You can see the Accel values if you hover the mouse cursor over the Accel panel.

The ideal target value of Y Accel is '-9.80'.  If your Viewer shows the Y Accel value as significantly lower than this such as -9.3 or -9.4 then it is evidence that your IMU needs calibration.  If your Y value is closer to -9.80 though, such as -9.75, then your IMU calibration is likely good and does not need recalibrating with rs-imu-calibration.py

MartyG-RealSense, 2024-06-06:
Hi @anath93  Do you require further assistance with this case, please?  Thanks!

anath93, 2024-06-06:
@MartyG-RealSense Thank you for the help. I am closing the ticket here, I am working with your backend team over internal ticket as this is issue and there is workaround.

closed 2024-06-06 by anath93

michelgokan, 2024-08-17:
Did you find a solution to this?