realsenseai/librealsense #12989: Motion Module Doesn't Work

Jiajian-Chang, 2024-05-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455 | 
| Firmware Version                   | release 5.16 | 
| Operating System & Version |   Linux  5.10.120-tegra  | 
| Kernel Version (Linux Only)    |  Linux  5.10.120-tegra  | 
| Platform                                 |  NVIDIA Jetson Orin NX  |
| SDK Version                            |  v2.55.1 }                          | 

### Issue Description
I used the RSUSB method to build the SDK on a Jetson Orin NX with Jetpack 5.1.2.
When I use the realsense-viewer, there is a warning (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11. The RGB camera and Stereo Module work fine but the Motion Module can not work and give Motion Module failure error. I also tried the ROS package and there is no message from the sample topic.

MartyG-RealSense, 2024-05-31:
Hi @Jiajian-Chang  Typically in the RealSense Viewer tool, after the **Motion Module Failure** error has occurred the Motion Module can start successfully if it is enabled a second time.  Is the module able to start if you enable it again after the first failure, please?

I would not recommend enabling depth, RGB and Motion Module simutaneously in the Viewer though.  This is because for some RealSense users the Viewer has a non-fixable issue where enabling three streams that include the Motion Module can cause 'No Frames Received' on one of the streams.  The trigger for the problem is the enabling of the Motion Module, so changing the order that the streams are enabled in does not make a difference.  The ROS wrapper does not have this issue though.

In the RealSense ROS wrapper the IMU topic is disabled by default.  You can enable them by adding the commands below to your launch instruction.

`enable_accel:=true enable_gyro:=true unite_imu_method:=2 `

This should enable the gyro and accel topics and publish them in a single **/imu** topic.

Jiajian-Chang, 2024-05-31:
> Hi @Jiajian-Chang Typically in the RealSense Viewer tool, after the **Motion Module Failure** error has occurred the Motion Module can start successfully if it is enabled a second time. Is the module able to start if you enable it again after the first failure, please?
> 
> I would not recommend enabling depth, RGB and Motion Module simutaneously in the Viewer though. This is because for some RealSense users the Viewer has a non-fixable issue where enabling three streams that include the Motion Module can cause 'No Frames Received' on one of the streams. The trigger for the problem is the enabling of the Motion Module, so changing the order that the streams are enabled in does not make a difference. The ROS wrapper does not have this issue though.
> 
> In the RealSense ROS wrapper the IMU topics are disabled by default. You can enable them by adding the commands below to your launch instruction.
> 
> `enable_accel:=true enable_gyro:=true unite_imu_method:=2 `
> 
> This should enable the gyro and accel topics and publish them in a single **/imu** topic.

I tried the ROS2 wrapper with both unite_imu_method:=2 and unite_imu_method:=1. None of them has message output on /imu topic. There is also no output on /gyro/sample and /accel/ sample, when I make unite_imu_method:=0. The realsense-viewer works fine on my x86 laptop.

MartyG-RealSense, 2024-05-31:
As you are using librealsense 2.55.1, are you using the ROS2 wrapper version **4.55.1** that is the correct match for it, please?

Jiajian-Chang, 2024-05-31:
> As you are using librealsense 2.55.1, are you using the ROS2 wrapper version **4.55.1** that is the correct match for it, please?

I tried 4.55.1 and master branch. Both of them give me 
warning (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
Same performance with realsense-viewer. The RGB camera and depth can work but no output of IMU. I also tried initial_reset:=true. It doesn't work.

MartyG-RealSense, 2024-05-31:
Next, please try resetting your camera to its factory-new default calibration in the RealSense Viewer using instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 as this reset creates a new **calibration table** inside the camera hardware which can clear camera operational problems caused by a corrupted calibration table.

Jiajian-Chang, 2024-05-31:
> Next, please try resetting your camera to its factory-new default calibration in the RealSense Viewer using instructions at [#10182 (comment)](https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487) as this reset creates a new **calibration table** inside the camera hardware which can clear camera operational problems caused by a corrupted calibration table.

I can not write the calibration table. It gives me warning (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11.

I have another D455 with firmware 5.13. That one can work. So I downgrade this one's firmware to 5.13 and it works. Thank you for your help. I think there may be some issue between the latest firmware and Jetson.

anath93, 2024-05-31:
@MartyG-RealSense I had the same issue with 2.55.1.
@Jiajian-Chang  Please try development tag to fix this issue instead of 2.55.1 which is what worked for me.

MartyG-RealSense, 2024-06-01:
Hi @Jiajian-Chang  I'm pleased to hear that you found a workaround that could work for you.

@anath93  Thanks very much for sharing what worked for you in your particular case!

closed 2024-06-14 by MartyG-RealSense