realsenseai/librealsense #12990: Split .bag according to time

linahourieh, 2024-05-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435|  
| Firmware Version                   | Open RealSense Viewer 2.55.1 | 
| Operating System & Version |   Linux (Ubuntu 22.04) / Windows 11  | 
| Kernel Version (Linux Only)    | 6.5.0-35-generic                                  | 
| Platform                                 | PC  |
| SDK Version                            |  legacy / 2.0                        | 
| Language                            |  python }                          | 
| Segment			|                            | 

### Issue Description
I have recorded .bag file with my RealSense viewer (on windows 11), i would like to split this file into 2 files or more according to time, is there a way other than Rosbag to split this file ??

linahourieh, 2024-06-02:
@MartyG-RealSense Thank you for the answer, I would find that inconvienent, it there a form other than .bag that I can use to record the RGB and depth?
Thak you in advance

MartyG-RealSense, 2024-06-02:
If you want to record data as a continuous stream in one or two files instead of many multiples of file then bag is the only available option unless you record the streams to a non-interactive video file such as mpeg or avi.

closed 2024-06-10 by linahourieh