realsenseai/librealsense #12992: Corrupt frames on playback, ROSLZ4_DATA_ERROR

jkenney9a, 2024-05-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.13.00.50 | 
| Operating System & Version |   {Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.4.0-26| 
| Platform                                 | PC  |
| SDK Version                            |  { l2.50.0.3812 }                          | 
| Language                            |  {python }                          | 
| Segment			|  }                          | 

### Issue Description
We are running into an issue where we occasionally get corrupted frames in our .bag files. The way we use the D435 cameras is that we  first record short videos for our experiments. From the recorded video .bag file, we use a python script to extract the color video to feed into a tracking program. Occasionally, when we go to extract the color video the code quits early. When we go to look at these videos that do not fully convert in the realsense viewer, we see an error in the console that says:

`16:09:36 [Error] /home/jenkins/workspace/LRS_SDK_CI_Debian/src/src/media/playback/playback_device.cpp:518 - Failed to read next frame from file: ROSLZ4_DATA_ERROR: malformed data to decompress`

However, we can skip past this frame in the RealSense Viewer by dragging the scroll bar, telling us that it is just this one frame that is corrupted, but that the rest of the file is okay. 

What I'd like to do is just skip this corrupt frame, drop it, and continue with the video extraction. However, I'm struggling with a way to get the python code to do this. If I check for the presence of a frame here, it comes back false. But if I tell the script to 'continue' (i.e., return back to the top of the while loop that is pulling out the frames) it does not seem to go on to the next frame but gets 'stuck' at this corrupted frame. 

Any ideas on how to force the Python script to skip this corrupted frame? I tried to add in a loop to count the number of missing frames and to force the script to go back to the beginning of the while loop, but that doesn't seem to work. It appears to only be one or two corrupted frames. 

I've also tried rosbag reindex and rosbag fix with no luck Reindexing just outputs the video file up to the point that the error occurs and ignores the rest. 

Here is the code that we use for extracting the color videos from a bag file:

```
 pipe = rs.pipeline()
    config = rs.config()
    config.enable_device_from_file(bag_file, repeat_playback=False)
    profile = pipe.start(config)
    
    #Needed so frames don't get dropped during processing:
    profile.get_device().as_playback().set_real_time(False)
    
    #Setup streaming
    config.enable_stream(rs.stream.depth, resolution['x'], resolution['y'],
                             rs.format.z16)
    
    
    if color_video_type.lower() == 'rgb':
        config.enable_stream(rs.stream.color, resolution['x'], resolution['y'], 
                                 rs.format.rgb8)
    else:
        config.enable_stream(rs.stream.color, resolution['x'], resolution['y'], 
                                 rs.format.y16)
    
    #Setup depth stream and info
    if convert_depth_video.lower() == 'y':
        #Setup video codec
        fourcc_depth = cv2.VideoWriter_fourcc(*'XVID')
        depth_filename = filename + '_depth.avi'
        
        #Video writer
        out_depth = cv2.VideoWriter(depth_filename, fourcc_depth, fps, 
                                (resolution['x'], resolution['y']), isColor=True)
        #For colors in depth frame
        colorizer = rs.colorizer()
    
    #Setup color stream and info
    #Setup video codec
    fourcc_color = cv2.VideoWriter_fourcc(*'XVID')

    color_filename = filename + '_color.avi'
    #Video writer
    
    if color_video_type == 'rgb':
        out_color = cv2.VideoWriter(color_filename, fourcc_color, fps, 
                                (resolution['x'], resolution['y']), isColor=True)
    else:
        out_color = cv2.VideoWriter(color_filename, fourcc_color, fps, 
                                (resolution['x'], resolution['y']), isColor=False)

    
    #For saving the aligned depth data
        #Depth scale in meters
    if get_depth_data.lower() == 'y':
        depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
        depth_distance_filename = filename + '_distance.h5'
        f_depth_dist = h5py.File(depth_distance_filename, 'w')
        
        depth_data_group = f_depth_dist.create_group('Camera and Video info')
        depth_data = f_depth_dist.create_dataset('distance', 
                                                 (resolution['y'], resolution['x'], 1),
                                                 dtype='i8', compression='gzip',
                                                 maxshape=(resolution['y'], resolution['x'], None))
        
        #Add depth scale and fps as an attribute to the h5 file
        depth_data_group.attrs['Depth Scale'] = depth_scale
        depth_data_group.attrs['FPS'] = fps
   
    #Setup counter
    i = 0
    print('Processing Video:')
    total_frames = fps * video_length 
    
    #To count missing frames
    missing_frame_count = 0
    
    while True:
    
        frame_present, frameset = pipe.try_wait_for_frames()
        #frame_present, frameset = pipe.poll_for_frames()
        
        #End loop once video finishes
        #Added missing_frame_count to deal with the occasional frame drop
        if not frame_present and missing_frame_count > 10:
            break
        elif not frame_present:
            missing_frame_count +=1
            print('frame missing')
            continue
        
        #Align frames to color image
        align = rs.align(rs.stream.color)
        frameset = align.process(frameset)
        
        #Get frames
        color_frame = frameset.get_color_frame()
        depth_frame_aligned = frameset.get_depth_frame()
        
        color_frame = np.asanyarray(color_frame.get_data())
        
        
        if color_video_type.lower() == 'rgb':
            color_frame = cv2.cvtColor(color_frame, cv2.COLOR_RGB2BGR)
        else:
            #Need to convert from 16 bit grayscale image to 8 bit
            #color_frame = cv2.normalize(color_frame, None, 0, 255, dtype=cv2.CV_8U)
            
            color_frame = color_frame / 256 
            color_frame = color_frame.astype('uint8')
        
        #Get depth data at each point
        if get_depth_data.lower() == 'y':
            depth = np.asanyarray(depth_frame_aligned.get_data()).astype('uint8')
            depth_data.resize((resolution['y'], resolution['x'], i+1))
            depth_data[:,:,i] = depth
        
        #Save frames/depth info
        out_color.write(color_frame)
        if convert_depth_video.lower() == 'y':
            depth_frame_aligned_colorized = np.asanyarray(colorizer.colorize(depth_frame_aligned).get_data())
            out_depth.write(depth_frame_aligned_colorized)
        
        #Update user on progress
        frac_done = i / total_frames
        print('\r','{:>7.1%}'.format(frac_done), 'done', end='')
        i += 1
    
    
    #Wrap things up
    pipe.stop()
    out_color.release()
    output_filenames = [color_filename]
    if convert_depth_video.lower() == 'y':
        out_depth.release()
        output_filenames.append(depth_filename)
    if get_depth_data.lower() == 'y':
        f_depth_dist.close()
        output_filenames.append(depth_distance_filename)
    print('\nBag file succesfully converted') 
    
    return(output_filenames)
```

MartyG-RealSense, 2024-06-01:
Hi @jkenney9a  Usually if a RealSense SDK bag file has corruption in it then the only solution is to re-record the entire bag, unfortunately.

 The issue that you reported regarding one or two corrupted color frames is not usually one that occurs.  When the SDK encounters a bad frame during streaming, it should go back to the last known good frame and continue from that point onwards.

If you are using Python code to record the bag then you could consider increasing the frame queue size to introduce more latency into the pipeline.  Doing so can make streaming more stable (such as causing fewer dropped frames to occur).  The SDK provides a Python example program that demonstrates this called **frame_queue_example.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/frame_queue_example.py#L45

jkenney9a, 2024-06-04:
Hi @MartyG-RealSense ,

Thank you for your quick response. That's unfortunate. Fortunately, this doesn't happen very often for us. Unfortunately we cannot just re-record, as this is experimental data. 

I'll take a look at your frame queue example to hopefully minimize the problem. Would it be expected that using the frame queue would require more resources such that it would slow acquisition? We usually collect data from two cameras simultaneously on our computers and save data onto an SSD to minimize any speed issues.

MartyG-RealSense, 2024-06-04:
I would not expect acquisition to slow.  The main effect on computer resources of increasing the size of the frame queue is that the amount of memory used by the application may increase as the queue has to hold a greater quantity of frames.

closed 2024-06-06 by jkenney9a