realsenseai/librealsense #12994: Whether intel depth quality can only estimate RMS error on the plane?

birard, 2024-06-02:
Ｒequired info
| Camera Model | {D400 } |
| Firmware Version | 5.15.1 |
| Operating System & Version | Win 10 |
| SDK Version | Intel RealSense Viewer v2.54.2 |                       | 

### Issue Description

[image]
[image]

I want to know if depth quality can be used for error estimation in similar images. I know that depth quality is estimated by plane fitting, so is it not allowed to use depth quality on this kind of image? Because I want to use D435 to measure it. The distance is used to convert the resolution, so we must ensure that the distance does not have too big an error.
And I also want to know whether the distance in depth quality is the distance from the center or the distance across the entire plane? In the above image that's two different value about distance and center distance.

MartyG-RealSense, 2024-06-02:
Hi @birard  Plane fitting can be used to estimate the angle of the camera device relative to the floor on RealSense camera models that are not equipped with an IMU.  

Depth error measurement though is mainly based on **RMS Error**, where depth error starts at around zero at the front of the camera and increases linearly as distance from the camera increases.  The section of Intel's camera tuning guide linked to below provides more information about RMS error and manual calculation of it.  The error can also be measured automatically with the Depth Quality Tool that you showed an image of above.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#verify-performance-regularly-on-a-flat-wall-or-target

On the D435 model, RMS error starts to become noticeable at around 3 meters from the camera and beyond.

Any visible depth coordinate on a depth image should have a corresponding depth value, not just the center.  Typically though, the values will be most accurate at the center area of the image and drift in accuracy when moving out towards the edges of the image.

Error is not just determined by distance though.  Environmental factors such as lighting and reflections or the type of material that is being observed and its color may also negatively affect accuracy (dark gray or black materials or ones with low surface detail are more difficult for the camera to read).

birard, 2024-06-02:
Thank you for your response, but what I really want to know is that  the depth quality is projected through the original d435 plane to fit another plane, and that means whether its error value will increase on a plane with ups and downs like above image. The angle calculation will also be incorrect, or even though I am not facing the depth value of a plane, and I can still use the depth quality as the depth indicator?

MartyG-RealSense, 2024-06-02:
Plane fit does not need to be considered in this particular scene.  What will matter most is how much texture the stones have on their surface and how large the stones are.  A surface with some grain and pattern on it will be more easily analyzable for depth information than a surface that is completely plain and smooth, which is why the stones show up well on your depth image.

It will not matter that the stones are randomly up and down.  Not having regular, predictable patterns in the scene can actually be an advantage. The height of the stones is important though.  You can see that the central larger stones show up clearly as individual objects on the image, whilst the smaller stones are rendered as a single plain colour because they are close enough to the ground to be treated as though they are a flat floor.

You may find that changing the Preset in the side panel from 'Custom' to 'Medium Density' enhances the depth image, as it provides a good balance between accuracy and the amount of depth detail on the image.

closed 2024-06-16 by MartyG-RealSense