realsenseai/librealsense #12995: Flicker of frames from D405

DaveBGld, 2024-06-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.15.0-107-generic                                | 
| Platform                                 | PC|
| SDK Version                            |  2.<?>.<?> }                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description

In my application I read frames from the D405 and feed to and inference server to detect semantic segmentation. The results of the segmentation flicker substantially despite that all objects "seen" by the D04 are static, and all frames are identical.

This is the code I am using in a loop:

```
rs2::align alignToColor(RS2_STREAM_COLOR);  
auto dataTemp = videoPipe.wait_for_frames(); 
data = alignToColor.process (dataTemp);   

video = data.get_color_frame();
rs2::depth_frame depth = data.get_depth_frame();  

// Query frame size (width and height)
const int w = video.as<rs2::video_frame>().get_width();
const int h = video.as<rs2::video_frame>().get_height();

// Create OpenCV matrix of size (w,h) from the colorized depth data
cv::Mat image(cv::Size(w, h), CV_8UC3, (void*)video.get_data(), cv::Mat::AUTO_STEP);

// Send image to inference server

```
I display both the original image, which doesn't change and the inference mask, and while the camera image doesn't change, the inference mask is substantially different from frame to frame. 

I isolated this piece of code by saving a frame from the D405, and replacing the code above with:
```
        cv::String imageFileName = "./20240520_151518image.PNG";
        // Read the image
        cv::Mat image = cv::imread(imageFileName);
```

And send the image to the inference server, and results in very stable inference masks.

The camera initialization is:

```
void startRealsense () {
    rs2::colorizer color_map;
    rs2::config configurationA;
    configurationA.enable_stream(rs2_stream::RS2_STREAM_COLOR, 0, 1280, 720, rs2_format::RS2_FORMAT_BGR8, 0);
    configurationA.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 0, 1280, 720, rs2_format::RS2_FORMAT_ANY, 0);
    rs2::pipeline_profile profile = videoPipe.start(configurationA);
    auto depth_sensor = profile.get_device().first<rs2::depth_sensor>();
    depth_scale = depth_sensor.get_depth_scale();
    // Acquire camera parameters
    const rs2::stream_profile color_profile = profile.get_stream(RS2_STREAM_COLOR);
    const rs2::stream_profile depth_profile = profile.get_stream(RS2_STREAM_DEPTH);
    depth_intrin = depth_profile.as<rs2::video_stream_profile>().get_intrinsics();
    color_intrin = color_profile.as<rs2::video_stream_profile>().get_intrinsics();
    depth_extrin_to_color = depth_profile.as<rs2::video_stream_profile>().get_extrinsics_to(color_profile);
    color_extrin_to_depth = color_profile.as<rs2::video_stream_profile>().get_extrinsics_to(depth_profile);
    auto depth_stream = profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
    auto resolution = std::make_pair(depth_stream.width(), depth_stream.height());
    auto i = depth_stream.get_intrinsics();
    auto principal_point = std::make_pair(i.ppx, i.ppy);
    auto focal_length = std::make_pair(i.fx, i.fy);
    rs2_distortion model = i.model;
    rs2_option option_type;

    int exposureValue = thisLaserClientParameters.rsExpopsureValue;
    auto depthSensor = profile.get_device().query_sensors()[0];
    depthSensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, false);
    depthSensor.set_option(RS2_OPTION_EXPOSURE, exposureValue);
 
}
```

Thanks for the help!!

DaveBGld, 2024-06-03:
Still flickering...

The code is now:

```
    configurationA.enable_stream(rs2_stream::RS2_STREAM_COLOR, 0, 1280, 720, rs2_format::RS2_FORMAT_BGR8, 5);
    configurationA.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 0, 1280, 720, rs2_format::RS2_FORMAT_Z16, 5);

```

MartyG-RealSense, 2024-06-04:
Is there flicker if you change the color stream to the infrared stream and use the alternative RGB mode 'RGB from left infrared sensor' feature that D405 supports?

`configurationA.enable_stream(rs2_stream::RS2_STREAM_INFRARED, 1280, 720, rs2_format::RS2_FORMAT_BGR8, 5);`

closed 2024-06-16 by MartyG-RealSense