realsenseai/librealsense #12996: There was a problem opening the viewer with the D455 camera

Denis-Megumi, 2024-06-03:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

Denis-Megumi, 2024-06-03:
I tried installing an earlier version of the SDK, but it still gets this error, and my Depth Quality Tool doesn't open either

MartyG-RealSense, 2024-06-03:
Hi @Denis-Megumi  As you installed an earlier version of the SDK, does the 2.55.1 version of the RealSense Viewer now work if you go back to it?

Denis-Megumi, 2024-06-03:
No, it doesn't work. I uninstalled the 2.54 version and then reinstalled the 2.21 version, and although I was able to open the viewer, I couldn't connect to the camera. Finally, I tried to install the 2.55.1 version of RealSense Viewer, but I still got the above error.

MartyG-RealSense, 2024-06-03:
Does the **rs-capture** run correctly if you launch it using the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

Denis-Megumi, 2024-06-03:
Yes, it works

MartyG-RealSense, 2024-06-03:
Okay, thank you very much.  So the problem is not likely to be with GL graphics support on your computer, since rs-capture requires it in order to work.  

As you have a D455 camera it would be normal for it not to be detected in the 2.21 Viewer as D455 support was not introduced until version 2.38.  The advice about running 2.21 is for older RealSense camera models (D415 and D435).

Do you know which firmware driver version your D455 camera is currently using, please?  If you do not know then you can find it using the **rs-enumerate-devices** tool.  There are instructions at https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-615365409 for this.  To find the firmware version, use the command **rs-enumerate-devices -s** instead of the '-c' in the instructions.

Denis-Megumi, 2024-06-03:
[image]
Here is my firmware driver version

Denis-Megumi, 2024-06-03:
[image]
As you said, I only installed version 2.54.2 of the viewer and I'm still having this issue

MartyG-RealSense, 2024-06-03:
Are you able to update the graphics driver of your Windows computer to confirm whether your issue is occurring due to an outdated driver?

Denis-Megumi, 2024-06-04:
I'm not sure which graphics driver you're talking about, but I've upgraded my NVIDIA GPU driver to the latest version, and I'm still getting the same error

MartyG-RealSense, 2024-06-04:
It sounds as though you have updated the correct driver.

As you have installed the full RealSense SDK, does the same error occur if the Depth Quality Tool is launched?

Denis-Megumi, 2024-06-05:
Now the problem has become a bit more serious, and when I plug in the USB again, I am prompted to update the firmware of the camera. When I finished updating the camera I couldn't connect, and now I can't even run RS-Capture. The presence of the camera was also not detected in the Device Manager, but an unknown USB device appeared (the configuration descriptor was invalid).
[image]
[image]

MartyG-RealSense, 2024-06-05:
May I ask which country you are located in, please?

Denis-Megumi, 2024-06-05:
China

MartyG-RealSense, 2024-06-05:
Thank you very much.  I will discuss your camera non-detection after firmware update with my Intel RealSense colleagues.

Denis-Megumi, 2024-06-07:
Thank you very much for your assistance with the project. I greatly appreciate your help and look forward to hearing from you soon.

MartyG-RealSense, 2024-06-11:
Hi @Denis-Megumi  Please send me an email at the address below so that we can discuss the next step for restoring your camera.

martyx.grover@intel.com

Denis-Megumi, 2024-06-12:
862971459@qq.com.This is my email address You can contact me at this email.

closed 2024-07-24 by MartyG-RealSense