realsenseai/librealsense #12998: RS pipeline.stop() freezes program after long runtime

marcuscop, 2024-06-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435e FRAMOS | 
| Firmware Version                   | HW:1.2_FW:1.14.4.0 | 
| Operating System & Version |   Linux (Ubuntu 22.04.3) | 
| Kernel Version (Linux Only)    |  6.2.0-37-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.50.10                       | 
| Language                            |  python                        | 
| Segment			|  Robot                       | 

### Issue Description
Hi There! I have experienced two issues with long runtime sessions (days/weeks). 1. Sometimes when leaving the rs pipeline open indefinitely, the RGB and depth images suddenly become over-exposed (RGB image is blurry and bright and depth image is mostly garbage). 2. Sometimes when stopping the pipeline the program freezes like reported in https://github.com/IntelRealSense/librealsense/issues/10245#issuecomment-1044328694.

I believe stopping the sensor instead of the pipeline is the recommended solution as seen here (https://github.com/IntelRealSense/librealsense/issues/6459#issuecomment-648795212) but haven't seen a python solution, only C++. Here's an example of what I tried but occasionally run into a "cannot call start() before stop()" error when attempting to call rs.pipeline.start().

```
context = rs.context()
pipeline = rs.pipeline(context)
rs_config = rs.config()
rs_config.enable_device("XXXXXXXXX")
device = rs_config.resolve(pipeline).get_device()
color_sensor = device.query_sensors()[1]
profile = pipeline.start(rs_config)
# multiple days spent in this loop
while(True):
   frames = pipeline.wait_for_frames()
   while(system_stopped): # can be stopped for hours here, normal functionality
      pass
   color_sensor.stop()
   pipeline.start(rs_config) # err: "cannot call start() before stop()"
```

Is there something wrong with this snippet? Or is there a better solution for python?

marcuscop, 2024-06-04:
@MartyG-RealSense Thank you for the response! 

A bit of clarification: In the above snippet, I call pipeline.start() on line 7 before sensor.stop(). Then I attempt to start the pipeline/sensor again by calling pipeline.start() afterwards. In spite of this I still get the error "cannot call start() before stop()" which I think is because there is a internal flag or variable that has not been flipped by sensor.stop(). This puts me in a bit of a predicament because I have to call pipeline.stop() to flip the flag but cannot call pipeline.stop() without risk of freezing the program. Is there any way to work around this?

MartyG-RealSense, 2024-06-05:
Using sensor.stop() has not always worked in all cases that have attempted it.

If you are trying to recover the program after a problem with streaming has occurred then an alternative approach is to perform a reset of the camera with code.  If the reset is completed within 5 seconds then the pipeline should continue onwards from the point that it left off without having to perform a pipe start.  A camera reset will typically take 2 to 3 seconds to complete.

A simple example of Python code for performing a hardware reset is:

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

A slightly more complex example for resetting a specific camera is here:

```
import pyrealsense2 as rs2
ctx = rs2.context()
list = ctx.query_devices()
for dev in list:
serial = dev.query_sensors()[0].get_info(rs2.camera_info.serial_number)
dev.hardware_reset()
```

closed 2024-06-21 by MartyG-RealSense