realsenseai/librealsense #12999: depth error

12581djjd, 2024-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Operating System & Version |   {Win (11)   | 
| Platform                                 | PC |
| Language                            |  {python }                          | 

[image]
The four small blue planes at the top are being measured correctly in terms of depth. However, the depth values for the trapezoidal plane at the bottom are incorrect; the actual depth of the trapezoidal plane is 70 cm, but the D435 measures it as 1.1 to 1.2 meters.

12581djjd, 2024-06-04:
dis = aligned_depth_frame.get_distance(real_xyz[0], real_xyz[1])
                            real_xyz = rs.rs2_deproject_pixel_to_point(depth_intrin, real_xyz, dis)
Both methods of using function calls in Python and viewing with Intel.RealSense.Viewer software result in incorrect depth values for the trapezoidal plane.

MartyG-RealSense, 2024-06-04:
Hello, depth at the bottom of a deep container may be more difficult to read because the bottom is further away from the light source above the container. 
 
The image below is from a past case involving a bucket, where the sides of the bucket are correctly blue (near distance) but the bottom of the bucket is in the pink to red range, which represents a meter or more away on the depth scale of this particular image.  Increasing the lighting level if possible may therefore improve the accuracy of the depth reading for the bottom of the container.

[image]

MartyG-RealSense, 2024-06-10:
Do you require further assistance with this case, please?  Thanks!

12581djjd, 2024-06-11:
The issue is not with a single plane; the distance between the upper and lower planes is problematic. My previous question was incorrect.
[image]

12581djjd, 2024-06-11:
It's not an issue with the light; is calibration required?Did I mistakenly set the parameters for Intel.RealSense.Viewer?

MartyG-RealSense, 2024-06-11:
I don't think that it's a calibration issue.  The deep walls of the container may be confusing the camera when trying to depth-sense the central bottom of the container.  It is similar to how depth readings of a flat wall may become inaccurate if the 90 degree corner between that wall and the adjacent wall is included in the camera's view.

You could try changing the **Depth Units** setting in the Viewer from its default value of '0.001' to **0.0001** to see whether it improves the accuracy of your results. This can be done in the **Stereo Module > Controls** section of the Viewer's options side-panel.

[image]

https://github.com/IntelRealSense/librealsense/issues/10976#issuecomment-1271236193 describes how to implement a depth scale value change in Python.

12581djjd, 2024-06-13:
[image]

12581djjd, 2024-06-13:
[image]

MartyG-RealSense, 2024-06-13:
I note that your Asic Temperature (the internal operating temperature of the camera) is quite hot at 45 degrees C.  Officially a maximum temperature of 35 degrees is recommended, though the camera may be able to go up to 42 degrees C without negative effects.  Beyond 42 degrees though, problems could manifest in the image.

Does accuracy of the four small blue planes improve if you set the Viewer's **Preset** setting near the top of the options side-panel to the 'Medium Density' preset?  Medium Density provides a good balance between accuracy and the amount of detail on the depth image.

12581djjd, 2024-06-17:
I have selected all the modes in Presets, but it did not solve the problem. Could it be that I set the wrong parameters at some point, and they were recorded and have been used ever since? How can I restore the d435 to factory settings?

12581djjd, 2024-06-17:
Or should I lower the working temperature of the D435?

MartyG-RealSense, 2024-06-17:
It would be ideal if the temperature could be lowered.  A camera can become mis-calibrated and need to be recalibrated if it experiences a severe temperature event.

You can reset the camera to its factory-new default calibration in the RealSense Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

You could try disabling the camera's IR emitter component by setting Laser Power to zero to confirm whether or not the IR light projected by the camera onto the container could be making the container more difficult for the camera to read.  The light would be more intense at the top of the container than the bottom because it is nearer to the camera.

closed 2024-07-04 by MartyG-RealSense