realsenseai/librealsense #13001: Frame didn’t arrive within 1000

davidpob99, 2024-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.16.0.1| 
| Operating System & Version |   Linux 20.04.6 LTS  | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                         | 
| Platform                                 | NVIDIA Jetson Orin Nano|
| SDK Version                            |  2.55.1                     | 
| Language                            |  C++                          | 
| Segment			|  others      | 

### Issue Description
We have two cameras connected using a 4.5m 3.2 10Gbps USB-C cable + a USB-C to USB-A 3.2 converter. The cameras are detected properly as USB 3.2. We have a C++ program that work properly for certain amount of time (between 5 and 45 minutes). Then, one of the cameras issues a “Frame didn’t arrive within 1000” exception and it is disconnected from the USB port. We cannot even see it using “rs-enumerate-devices” or “lsusb” call. If the computer is reboot, the problem persists. We have to shutdown the computer (disconnect from the power supply) to detect the camera again. The other camera works smoothly for hours without any issue.

MartyG-RealSense, 2024-06-04:
Hi @davidpob99  If you are using a USB C to C cable then they typically do not work as well with RealSense cameras as USB A to C cables.  Is the camera with the USB-C cable the one that experiences the problem, please?

Cable quality and length are also important factors in stability.  A 4.5 m cable used with RealSense cameras should be high quallity, otherwise performance degradation and disconnections can occur.  It must also be a cable that was designed for data transfer, as an inexpensive one designed only for charging will not be suitable.

davidpob99, 2024-06-05:
Hi @MartyG-RealSense, the confirguration is the following: Cameras are located at 4m of the a Jetson Orin Nano. The Nano is within a protected box with an external female USB-C that connects to the internal USB-A Nano port. The 4.5m USB C to C cable and the USB C to A converter are of high quality 3.2 certified designed for data transfer.

davidpob99, 2024-06-05:
Hi @MartyG-RealSense, the configuration is the following: Cameras are located at 4m of the a Jetson Orin Nano. The Nano is within a protected box with an external female USB-C that connects to the internal USB-A Nano port. The 4.5m USB C to C cable and the USB C to A converter are of high quality 3.2 certified designed for data transfer.

Something that I don’t understand is that the connection works fine for 40 minutes and then the camera is disconnected. The other camera works fine for many hours. Have you experienced similar behaviors?

MartyG-RealSense, 2024-06-05:
Have you tried swapping the cameras?  If the camera that worked fine for hours shows the same disconnection behaviour when used on the other USB cable then it would be evidence that the problem might be in that particular cable.

Also, is the problem camera hot to the touch when you touch the casing?  If it has an overheating problem then you should be able to feel the heat only a few minutes after the camera is started.  

If you have a RealSense model that does not have a plastic outer casing, such as a Depth Module, then you could instead check the internal temperature of the camera in the options side-panel of the RealSense Viewer tool.

If the camera that works fine is disconnected from USB and only the problem camera is run, if it works for hours then it could be an indicator that the Orin Nano is struggling to run two cameras simultaneously.

davidpob99, 2024-06-06:
Thanks for the detailed recommendations. We have tested them and the behaviour of the camera is the same.
Something that I would like to understand is why the camera cannot be started again after the camera raises a try/catch, although it is detected with the “rs-enumerate-devices” call. Concretely, we are executing the following code:

```{c}
if (isActive) {
  try {
    data = pipe.wait_for_frames(1000);
  } catch (rs2::camera_disconnected_error e) {
    logger -> log -> warn(“(1) At camera {}: {}“, * cameraNum, e.what());
    isActive = false;
    return false;
  } catch (rs2::wrong_api_call_sequence_error e) {
    logger -> log -> warn(“(2) At camera {}: {}“, * cameraNum, e.what());
    isActive = false;
    return false;
  } catch (rs2::error e) {
    logger -> log -> warn(“(3) At camera {}: {}“, * cameraNum, e.what());
    logger -> log -> info(“Reconnecting Camera {}“, * cameraNum);
    pipe.stop();
    auto devices = ctx.query_devices();
    try {
      for (auto newDev: devices) {
        if (!registerSN(newDev))
          continue;
        profile = pipe.start(cfgRs);
        isActive = true;
        data = pipe.wait_for_frames(1000);
        break;
      }
    } catch (rs2::error e) {
      logger -> log -> warn(“Cannot reconnect camera {}: {}“, * cameraNum, isActive, e.what());
    }
    return false;
  }
```

This results in the following outcome:
```
[2024-06-06 09:27:33][warning][1866] (3) At camera 0: Frame didn’t arrive within 1000
[2024-06-06 09:27:33][info][1866] Reconnecting Camera 0
[2024-06-06 09:27:33][warning][1866] Cannot reconnect camera 0: acquire: Cannot open ’/dev/video0 Last Error: No such file or directory
[2024-06-06 09:27:34][warning][1866] (2) At camera 0: wait_for_frames cannot be called before start()
```
When executing the “rs-enumerate-devices --compact” on the command line:
```
Device Name          Serial Number    Firmware Version
Intel RealSense D435     244*********    5.16.0.1
Intel RealSense D435     244*********    5.16.0.1
Device info:
  Name             : 	Intel RealSense D435
  Serial Number         : 	244*********
  Firmware Version       : 	5.16.0.1
  Recommended Firmware Version : 	5.16.0.1
  Physical Port         : 	/sys/devices/platform/3610000.xhci/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video12
  Debug Op Code         : 	15
  Advanced Mode         : 	YES
  Product Id          : 	0B07
  Camera Locked         : 	YES
  Usb Type Descriptor      : 	3.2
  Product Line         : 	D400
  Asic Serial Number      : 	308*********
  Firmware Update Id      : 	308*********
  Dfu Device Path        : 	
Device info:
  Name             : 	Intel RealSense D435
  Serial Number         : 	244*********
  Firmware Version       : 	5.16.0.1
  Recommended Firmware Version : 	5.16.0.1
  Physical Port         : 	/sys/devices/platform/3610000.xhci/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video6
  Debug Op Code         : 	15
  Advanced Mode         : 	YES
  Product Id          : 	0B07
  Camera Locked         : 	YES
  Usb Type Descriptor      : 	3.2
  Product Line         : 	D400
  Asic Serial Number      : 	308*********
  Firmware Update Id      : 	308*********
  Dfu Device Path        :
```
If I relaunch the program again (and so the program tries to connect to the camera again), the initialization of the camera provides the following error:
```
[2024-06-06 10:29:28][warning][3677] Init RS camera 0: Frame didn’t arrive within 1000
```
And the “rs-enumerate-devices --compact” returns the following outcome:
```
06/06 10:29:38,843 ERROR [281473475164416] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
 06/06 10:29:38,844 ERROR [281473475164416] (rs.cpp:237) [rs2_create_device( info_list:0xaaaae5904800, index:0 ) Backend] set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
 06/06 10:29:38,844 ERROR [281473475164416] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument “device”
Could not create device - set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error . Check SDK logs for details
Device Name          Serial Number    Firmware Version
Intel RealSense D435     244*********    5.16.0.1
Device info:
  Name             : 	Intel RealSense D435
  Serial Number         : 	244*********
  Firmware Version       : 	5.16.0.1
  Recommended Firmware Version : 	5.16.0.1
  Physical Port         : 	/sys/devices/platform/3610000.xhci/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video6
  Debug Op Code         : 	15
  Advanced Mode         : 	YES
  Product Id          : 	0B07
  Camera Locked         : 	YES
  Usb Type Descriptor      : 	3.2
  Product Line         : 	D400
  Asic Serial Number      : 	308*********
  Firmware Update Id      : 	308*********
  Dfu Device Path        :
```

At this point in time, we cannot have access to the camera, even if rebooting the Orin Nano. We have to shutdown completely the Nano (disconnected from the power supply) to have access again.
I’ve tried using the “hardware_reset()” runtime call to reset the camera status, but the same behavior is observed.

Many thanks for your help

MartyG-RealSense, 2024-06-06:
If **try_wait_for_frames()** is used instead of wait_for_frames() then if the frame delivery time-outs then the program does not exit with an error but instead simply returns a status of 'false'.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

try_wait_for_frames() does not necessarily restore streaming though.  The pipeline could time-out and not exit with an error but still continue running with no new frames arriving.

MartyG-RealSense, 2024-06-14:
Hi @davidpob99  Do you require further assistance with this case, please?  Thanks!

davidpob99, 2024-06-14:
No, thank you very much for your quick response

closed 2024-06-14 by davidpob99