realsenseai/librealsense #13002: librealsense2-dkms error while installing: FATAL: Module videodev is in use for kernel 6.5.0-1003-oem

MarioCavero, 2024-06-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Ubuntu 22.04.4 LTS  | 
| Kernel Version (Linux Only)    |  6.5.0-1003-oem                                      | 
| Platform                                 | PC |
| SDK Version                            | 2.0?                      | 

### Issue Description
I'm following the [installation steps](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) and I am encountering an error for `librealsense2-dkms`. The guide states: 

> Install the libraries (see section below if upgrading packages):
> sudo apt-get install librealsense2-dkms
> sudo apt-get install librealsense2-utils
> The above two lines will deploy librealsense2 udev rules, build and activate kernel modules, runtime library and executable demos and tools.
```
cat /etc/apt/sources.list.d/librealsense.list 
deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo jammy main

```
 I do not have the log for when I did this the first time.
```
sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librealsense2-dkms is already the newest version (1.3.26-0ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
1 not fully installed or removed.
Need to get 0 B/2,332 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y

```
And:
```
sudo apt-get install librealsense2-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libatk-bridge2.0-dev libatk1.0-dev libatspi2.0-dev libcairo2-dev libdatrie-dev libepoxy-dev libfribidi-dev libgdk-pixbuf-2.0-dev libglfw3 libgraphite2-dev libgtk-3-dev libharfbuzz-dev
  libharfbuzz-gobject0 libpango1.0-dev libpixman-1-dev librealsense2 librealsense2-gl librealsense2-udev-rules libthai-dev libxcb-render0-dev libxcb-shm0-dev libxcomposite-dev libxdamage-dev
  libxtst-dev pango1.0-tools wayland-protocols
Suggested packages:
  libcairo2-doc libdatrie-doc libosmesa6 libgraphite2-utils libgtk-3-doc libpango1.0-doc libthai-doc graphicsmagick
The following packages will be REMOVED:
  librealsense2-dkms
The following NEW packages will be installed:
  libatk-bridge2.0-dev libatk1.0-dev libatspi2.0-dev libcairo2-dev libdatrie-dev libepoxy-dev libfribidi-dev libgdk-pixbuf-2.0-dev libglfw3 libgraphite2-dev libgtk-3-dev libharfbuzz-dev
  libharfbuzz-gobject0 libpango1.0-dev libpixman-1-dev librealsense2 librealsense2-gl librealsense2-udev-rules librealsense2-utils libthai-dev libxcb-render0-dev libxcb-shm0-dev libxcomposite-dev
  libxdamage-dev libxtst-dev pango1.0-tools wayland-protocols
0 upgraded, 27 newly installed, 1 to remove and 3 not upgraded.
1 not fully installed or removed.
Need to get 0 B/15.3 MB of archives.
After this operation, 48.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 349442 files and directories currently installed.)
Removing librealsense2-dkms (1.3.26-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module videodev is in use.
dpkg: error processing package librealsense2-dkms (--remove):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 librealsense2-dkms
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

Trying:

> You can remove all RealSense related DKMS packages from your computer with the command below:
> 
> dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

```
dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
(Reading database ... 349442 files and directories currently installed.)
Removing librealsense2-dkms (1.3.26-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module videodev is in use.
dpkg: error processing package librealsense2-dkms (--purge):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
Errors were encountered while processing:
 librealsense2-dkms
```

I have tried stopping videodev with modprobe unsuccesfully. No camera is connected. I first connected an intel real sense d405 (without installing anything), and disconnected it. 

```
 dpkg -l librealsense2-dkms 
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version         Architecture Description
+++-==================-===============-============-=========================================
rH  librealsense2-dkms 1.3.26-0ubuntu1 all          Modified kernel modules for librealsense2
```
And for any other kind of package I install (not realsense related), i get the same. 

In case this info can be useful:
```
sudo dkms status
ipu6-drivers/0.0.0, 6.5.0-1003-oem, x86_64: installed
mbedr_drv/1.0.69, 6.5.0-1003-oem, x86_64: installed
mbedr_drv/1.0.69, 6.5.0-35-generic, x86_64: installed
v4l2loopback/0.12.7, 6.5.0-1003-oem, x86_64: installed
```

 Thanks in advanced!

MartyG-RealSense, 2024-06-05:
Hi @MarioCavero  Another RealSense user at https://github.com/IntelRealSense/librealsense/issues/1288 who recently also experienced the **FATAL: Module videodev is in use** error resolved it with the following procedure:

1.  Use `lsmod` to find what videodev was currently being used by.  For example, their result from lsmod was `videodev v4l2_async, v4l2_fwnode, ov13858`

2.  They then used the value at the end of the sentence to deactivate the module that was using videodev.  For example, in their case the module was called ov13858 so they used it with the **sudo modprobe -r** module deactivation command.

`sudo modprobe -r ov13858`

They reported that afterwards the **librealsense2-dkms** package installed without any issue.

MarioCavero, 2024-06-05:
Thanks for the reply @MartyG-RealSense . I think you mean #12885 . 
`lsmod` for videodev:
  `videodev 364544  7 v4l2_async,v4l2_fwnode,hi556,videobuf2_v4l2,v4l2loopback,intel_ipu6_isys
`
Which according to this solution, should be `intel_ipu6_isys`
`$ sudo modprobe -r intel_ipu6_isys`
Which now results in `lsmod` of:

`videodev  364544  5 v4l2_async,v4l2_fwnode,hi556,v4l2loopback`

I am afraid I cannot see any major changes. Logs below:
```
$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librealsense2-dkms is already the newest version (1.3.26-0ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
1 not fully installed or removed.
Need to get 0 B/2,332 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
```
This seems to do nothing, so I proceed with the following package from the guidelines, `librealsense2-utils`
```
$ sudo apt-get install librealsense2-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libatk-bridge2.0-dev libatk1.0-dev libatspi2.0-dev libcairo2-dev libdatrie-dev libepoxy-dev libfribidi-dev libgdk-pixbuf-2.0-dev libglfw3 libgraphite2-dev libgtk-3-dev libharfbuzz-dev
  libharfbuzz-gobject0 libpango1.0-dev libpixman-1-dev librealsense2 librealsense2-gl librealsense2-udev-rules libthai-dev libxcb-render0-dev libxcb-shm0-dev libxcomposite-dev libxdamage-dev
  libxtst-dev pango1.0-tools wayland-protocols
Suggested packages:
  libcairo2-doc libdatrie-doc libosmesa6 libgraphite2-utils libgtk-3-doc libpango1.0-doc libthai-doc graphicsmagick
The following packages will be REMOVED:
  librealsense2-dkms
The following NEW packages will be installed:
  libatk-bridge2.0-dev libatk1.0-dev libatspi2.0-dev libcairo2-dev libdatrie-dev libepoxy-dev libfribidi-dev libgdk-pixbuf-2.0-dev libglfw3 libgraphite2-dev libgtk-3-dev libharfbuzz-dev
  libharfbuzz-gobject0 libpango1.0-dev libpixman-1-dev librealsense2 librealsense2-gl librealsense2-udev-rules librealsense2-utils libthai-dev libxcb-render0-dev libxcb-shm0-dev libxcomposite-dev
  libxdamage-dev libxtst-dev pango1.0-tools wayland-protocols
0 upgraded, 27 newly installed, 1 to remove and 6 not upgraded.
1 not fully installed or removed.
Need to get 15.3 MB of archives.
After this operation, 48.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libatk1.0-dev amd64 2.36.0-3build1 [96.6 kB]
Get:2 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2 amd64 2.55.1-0~realsense.12474 [2,656 kB]
Get:3 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libxtst-dev amd64 2:1.2.3-1build4 [16.3 kB]
Get:4 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libatspi2.0-dev amd64 2.44.0-3 [71.5 kB]
Get:5 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libatk-bridge2.0-dev amd64 2.38.0-3 [2,954 B]
Get:6 http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpixman-1-dev amd64 0.40.0-1ubuntu0.22.04.1 [280 kB]
Get:7 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2-gl amd64 2.55.1-0~realsense.12474 [744 kB]
Get:8 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libxcb-render0-dev amd64 1.14-3ubuntu3 [19.6 kB]
Get:9 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libxcb-shm0-dev amd64 1.14-3ubuntu3 [6,848 B]
Get:10 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libcairo2-dev amd64 1.16.0-5ubuntu2 [692 kB]
Get:11 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2-udev-rules amd64 2.55.1-0~realsense.12474 [11.8 kB]
Get:12 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2-utils amd64 2.55.1-0~realsense.12474 [8,348 kB]
Get:13 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libdatrie-dev amd64 0.2.13-2 [19.7 kB]
Get:14 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libepoxy-dev amd64 1.5.10-1 [132 kB]
Get:15 http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libfribidi-dev amd64 1.0.8-2ubuntu3.1 [64.8 kB]
Get:16 http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libgdk-pixbuf-2.0-dev amd64 2.42.8+dfsg-1ubuntu0.2 [47.8 kB]
Get:17 http://es.archive.ubuntu.com/ubuntu jammy/universe amd64 libglfw3 amd64 3.3.6-1 [83.2 kB] 
Get:18 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libgraphite2-dev amd64 1.3.14-1build2 [14.7 kB]
Get:19 http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libharfbuzz-gobject0 amd64 2.7.4-1ubuntu3.1 [26.0 kB]
Get:20 http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libharfbuzz-dev amd64 2.7.4-1ubuntu3.1 [553 kB]
Get:21 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libthai-dev amd64 0.1.29-1build1 [26.4 kB]
Get:22 http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 pango1.0-tools amd64 1.50.6+ds-2ubuntu1 [45.8 kB]
Get:23 http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpango1.0-dev amd64 1.50.6+ds-2ubuntu1 [154 kB]
Get:24 http://es.archive.ubuntu.com/ubuntu jammy/main amd64 libxcomposite-dev amd64 1:0.4.5-1build2 [9,326 B]
[... 25 lines cut ...]
`sudo systemctl stop v4l2-relayd`

Now it seems the other two can be stopped.
```
sudo modprobe -r v4l2loopback
sudo modprobe -r videodev
lsmod | grep videodev
```
[... cut ...]
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
```
Running `rs-capture` works as expected! `rs-enumerate-devices` as well. I loaded again the modules. I will try to now install the ros2 realsense needed packages, hopefully it won't cause any errors. Unplugging the realsense camera and trying to open the regular camera with cheese or ffmpeg does not work, but that is a different issue for now. 
Thank you again for the (as always) fast and useful response, it gave me good hints. Feel free to close this issue!

MartyG-RealSense, 2024-06-05:
Ah, thanks very much for pointing out the link's missing number.  I have edited it above.  :)

I'm pleased to hear that you were able to find a solution.  As you suggest, I will close the issue.  Thanks again!

closed 2024-06-05 by MartyG-RealSense