realsenseai/librealsense #13009: Error follow steps DMS

jnapoles, 2024-06-07:
Hi, I try to follow this steps i can't find requirement file to build my DMS sample.

https://dev.intelrealsense.com/docs/dimensional-weight-software-dws

6.3.1 Required Files
Binaries:
o All dlls in the download folder
o Plugins.xml
Header Files:
o DWS_Sdk.h
o DWS_Types.h
o DEBUG_DWS_sdk.h
Additional files under /Configuration/
o BoxMeasure

o Static_mode_config.json
We strongly discourage tampering with the configuration files, as any change in them could deteriorate accuracy and performance.

6.4 Integration
Integration of the SDK was designed to be as simple as possible.

Link the DWS_Sdk.dll to your project.
Add the header files to an include directory in the project.
Make sure the rest of the *.dll files supplied are copied to the build directory (same level as DWS_Sdk.dll)
Do the same with plugins.xml.
Copy the “Configuration” directory to an appropriate location:
The binaries will look for it in your app’s runtime “working directory” (usually where the executable file is found).

Any ide to resolve my problem, thanks so much.

MartyG-RealSense, 2025-08-30:
Hi @jnapoles  Did you ever achieve a solution for this 2024 issue, please?  Thanks!

MartyG-RealSense, 2025-08-30:
Okay, thanks very much for the update.  I appreciate the response.  I will close the issue as it was unsolved and it has been over a year since then.