realsenseai/librealsense #13011: i've been trying to get an intel realsense l515 camera running for a long time, and upon running rs-enumerate-devices i can find that the realsense camera has been discovered, but when i open the viewer the camerra is not getting listed! i'm using a linux ubuntu version 24.04

g-dhanushya-sai, 2024-06-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

g-dhanushya-sai, 2024-06-08:
@MartyG-RealSense please help me with this

MartyG-RealSense, 2024-06-08:
Hi @g-dhanushya-sai  Which version of the librealsense SDK are you using, please?  Support for the L515 camera model was removed in the current latest 2.55.1 SDK version, so version 2.54.2 or older must be used.

Could you try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

g-dhanushya-sai, 2024-06-08:
> Hi @g-dhanushya-sai Which version of the librealsense SDK are you using, please? Support for the L515 camera model was removed in the current latest 2.55.1 SDK version, so version 2.54.2 or older [...]

i've done that! i viewed a couple of other threads, and i tried applying a couple of them....

g-dhanushya-sai, 2024-06-08:
> i've done that! i viewed a couple of other threads, and i tried applying a couple of them....

I'm using version 2.50.0

MartyG-RealSense, 2024-06-08:
Is the camera able to work in the **rs-capture** example?

g-dhanushya-sai, 2024-06-09:
> Is the camera able to work in the **rs-capture** example?

No it doesn't work

it says realsense capture example is not responding! Please help me with this! I'm sorry to leave you hanging with the last response

MartyG-RealSense, 2024-06-09:
It's no trouble at all.

Next, please run **rs-hello-realsense** as this example only displays plain text and does not use graphics.  If the L515 is detected and the example works, this could suggest that your problem is related to displaying graphics.

g-dhanushya-sai, 2024-06-09:
@MartyG-RealSense 
i'm not sure what went wrong, when i type in rs-enumerate-devices it says command not recognized or something along those lines! i'm really new to all this, and it's for a college project, i have no assistance at college either, i'm really not sure, what to do, should i try using ubuntu version 22.04? my usb port version is 3.2 and sdk version librealsense-2.50.0... i'm desperate at this point, please help me with it

g-dhanushya-sai, 2024-06-09:
the enumerate devices command seemed to work earlier, now i'm back to square one

MartyG-RealSense, 2024-06-09:
You are currently using Ubuntu 24.04.  The librealsense SDK does not yet officially support 24.04 even in the current latest 2.55.1 version.  It is hoped that 24.04 will be supported in the next SDK version after 2.55.1.

Support for 22.04 was not officially introduced until SDK 2.53.1.

If librealsense is built from source code with CMake with the build flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction though then librealsense should be able to work with 22.04, as the RSUSB = true method is not dependent on Linux versions or kernel versions and does not require a patch script to be applied to the kernel.

What method did you use to install librealsense version 2.50.0 please?

MartyG-RealSense, 2024-06-16:
Hi @g-dhanushya-sai  Do you require further assistance with this case, please?  Thanks!

g-dhanushya-sai, 2024-06-16:
> Hi @g-dhanushya-sai  Do you require further assistance with this case, please?  Thanks!

Hey @MartyG-RealSense I'm sorry for not responding soon enough! I gave it to someone else who has worked on the camera before, and they said they'd get it done! Thank you for your concern 😊

MartyG-RealSense, 2024-06-16:
Okay, thanks very much for the update!

closed 2024-06-16 by MartyG-RealSense

Erumeldor, 2025-04-02:
> Okay, thanks very much for the update!

Hi @MartyG-RealSense 
Thanks for "-DFORCE_RSUSB_BACKEND=TRUE" - it worked like a charm to compile the lib with version 2.54.2 on Ubuntu 24.04.

Do you have any information on how it would be able to use the ROS wrapper for this SDK version with ROS2 Jazzy?
We do need to use Jazzy, and still have the L515 lidars that we would like to use.

Thanks in advance!

MartyG-RealSense, 2025-04-02:
Hi @Erumeldor  ROS2 Jazzy support was introduced in ROS2 wrapper version 4.56.1.  This is designed to be matched with librealsense SDK beta version 2.56.1.  However, the final SDK version that was validated as supporting L515 officially was **2.50.0**.  2.52.1 supported L515 but had not been validated with it.

You could obtain the source code for wrapper 4.51.1 (which is compatible with librealsense 2.50.0) and try retrofitting into its source code the changes in https://github.com/IntelRealSense/realsense-ros/pull/3149/files and https://github.com/IntelRealSense/realsense-ros/pull/3114/files that added 24.04 and Jazzy support.  The low likelihood of success means that it is likely not worth attempting though, unfortunately.

cemoke, 2025-11-04:
I also need to use ROS2 Jazzy and L515. Is there any update on this?

MartyG-RealSense, 2025-11-04:
Hi @cemoke  Jazzy support was introduced in ROS wrapper 2.56.1, whilst the final wrapper version to support L515 was 2.51.1, which had Humble as the most recently supported ROS2.