realsenseai/librealsense #13013: 2.0 Cable availability

ky047, 2024-06-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (-) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | - |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {- }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello.
I have a few questions to ask
Is there no problem with the D435 camera if I use 2.0 cable?
Please let me know the recommended specifications

MartyG-RealSense, 2024-06-10:
Hi @ky047  RealSense cameras are compatible with USB 2.1.  The range of supported resolution and FPS speed combinations is limited though compared to USB 3.2 and the Infrared 2 stream (right-side infrared) will be unavailable.  A USB2 connection also has a slower data transfer speed than USB3.

ky047, 2024-06-10:
It only affects transmission speed and FPS,
Does the 2.0 cable not affect the recognition of the camera?

MartyG-RealSense, 2024-06-10:
No, it does not affect recognition of the camera.

ky047, 2024-06-10:
Is there any case where 3.0 cable has a high recognition rate or 2.0 cable has a low recognition rate?

MartyG-RealSense, 2024-06-10:
The chances of the cable working successfully with RealSense cameras are reduced if:

1.  The cable is low quality or is an inexpensive one designed only for device charging instead of data transfer.

2.  A USB C to C cable is used instead of USB A to C (full size connector on one end and micro-sized connector on the other end, like the official cable supplied with the camera).

In general though, if a high quality USB A to C cable is being used then the success of the camera being correctly recognized when inserted into the USB port is likely to be more dependent on the computer that is being used with the camera and not the cable.

ky047, 2024-06-11:
Thank you for checking it out
It doesn't matter if you close the case
Have a good day

MartyG-RealSense, 2024-06-11:
You are very welcome.  As you suggested, I will close this case.  Thanks again!

closed 2024-06-11 by MartyG-RealSense