realsenseai/librealsense #13020: Jetson Orin NX - librealsense 2.55.1 - D455 doesn't work properly

yurii-drevnytskyi-dianomix, 2024-06-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D455 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version  | Ubuntu 22.04.3 LTS | 
| Kernel Version (Linux Only)    | 5.15.136-tegra | 
| Platform                                  | NVIDIA Jetson Orin NX, Jetpack 6 nvme |
| SDK Version                           | 2.55.1 | 
| Language                               | C++ | 
| Segment			         | Robot | 

### Issue Description
Librealsense was built from source, following your installation_jetson.md, with libuvc_installation.sh and -DBUILD_WITH_CUDA=true.
Jetpack 6 installed, using SDK manager, standart installation

Building Code:
`git clone https://github.com/IntelRealSense/librealsense.git --branch v2.55.1 --single-branch --depth 1 \`
`&& cd librealsense && scripts/libuvc_installation.sh \`
`&& mkdir build && cd build \`
`&& cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=true -DBUILD_WITH_CUDA=true \`
`&& cd librealsense/build && make -j6 && make install \`
`&& scripts/setup_udev_rules.sh`

After I start RGB stream in realsense-viewer - I'm getting up to 80% frame drop / second
After I start Stereo module too - it drops down to 6% / second
After I start Motion Module - all the streams crash, while continuously logging: 
`messenger-libusb.cpp:42 - control_transfer returned error, index 768, error: Resource temporarily unavailable, number: 11`
`messenger-libusb.cpp:42 - control_transfer returned error, index 1795, error: Resource temporarily unavailable, number: 11`

I found this line in the beginning of installation_jetson.md:
****For Jetson Orin™ with JetPack 6.0 you will need to follow build for MIPI driver as Nvidia released Kernel 5.15 default configuration with disabled HID: [Intel® RealSense™ camera driver for GMSL* interface]****
But I'm not sure if this info is up-to-date and I if need this for my usb3 camera. Without HID IMU stream can't work?

MartyG-RealSense, 2024-06-12:
Hi @yurii-drevnytskyi-dianomix  Yes, HID is necessary for IMU functions.

Although the MIPI reference driver is primarily designed for use with the D457 camera model and a GMSL cable connection (instead of USB), the driver's documentation states that installing it also provides a kernel Image with SENSOR_HID support for RealSense USB cameras with an IMU.

yurii-drevnytskyi-dianomix, 2024-06-12:
Looks like [build for MIPI driver](https://github.com/IntelRealSense/realsense_mipi_platform_driver/tree/master) doesn't support JetPack 6.0 rev. 1 with L4T 36.3 yet, and it's not possible to install Jetpack 6.0 DP with nvidia sdk manager anymore... :(
Are you going to add support for JetPack 6.0 rev. 1 any time soon?

MartyG-RealSense, 2024-06-13:
I have enquired to my Intel RealSense colleagues about your JetPack 6.0 Rev 1 question.

yurii-drevnytskyi-dianomix, 2024-06-13:
Do you think the frame drop issue could be fixed in same way?
Seems like other people have it too: https://github.com/IntelRealSense/librealsense/issues/12566

MartyG-RealSense, 2024-06-14:
My Intel RealSense colleagues advised to look at the **development** branch of the MIPI driver, which added support last week for the official release of JetPack 6.

https://github.com/IntelRealSense/realsense_mipi_platform_driver/tree/dev

MartyG-RealSense, 2024-06-21:
Hi @yurii-drevnytskyi-dianomix  Do you require further assistance with this case, please?  Thanks!

yurii-drevnytskyi-dianomix, 2024-06-21:
All good
Thank you for help

closed 2024-06-21 by yurii-drevnytskyi-dianomix

MartyG-RealSense, 2024-06-22:
You are very welcome, @yurii-drevnytskyi-dianomix - thanks very much for the update!

jeezrick, 2024-07-03:
> All good Thank you for help

Hi, is it working? Do you mind share your solution?

yurii-drevnytskyi-dianomix, 2024-09-19:
@jeezrick, Just follow the guide and IMU should work in the end...
https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/dev/README_JP6.md
PS. Sorry for delayed reply...