realsenseai/librealsense #13021: Camera-IMU extrinsics differences between D455 and D457

micahreich, 2024-06-11:
**Question:** Are the IMU-Camera and IMU-Infra1/2 extrinsics different between the D455 and the D457?

When plugged in, both a D455 and D457 appear as D455 models (they are using the exact same connector):

**Plugged in D455 `rs-enumerate-devices`:**
[image]

**Plugged in D457 `rs-enumerate-devices`:**
[image]

As such, I get the same IMU-Camera/Infra1/Infra2 extrinsics with `rs-enumerate-devices -c` on both camera types. However, after performing Kalibr Camera-IMU calibration, I get two different extrinsics values for each camera type. For the D455, I get values for the  Camera-IMU extrinsics which closely match what appears on `rs-enumerate-devices -c`. However, after calibrating the D457, I get a value which is different from that reported by `rs-enumerate-devices -c`

After performing Camera-IMU calibration for the stereo pair on the D457:
```yaml
cam0:
  T_cam_imu:
  - [0.9999931427800174, -0.0016892304941353018, 0.003295586940126516, 0.09949318287810802]
  - [0.0017030070662288125, 0.9999898053065498, -0.004181991140537492, 0.0049061493111220355]
  - [-0.0032884889956671256, 0.004187574871550686, 0.9999858249279432, -0.02955888788570322]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [1]
  camera_model: pinhole
  distortion_coeffs: [-0.01103729693207687, -0.0021985270178093096, -0.0002236915291331411, -0.0018945544301806865]
  distortion_model: radtan
  intrinsics: [389.8620950178092, 388.91800429584464, 325.3888036295283, 239.46701931772418]
  resolution: [640, 480]
  rostopic: /camera/infra1/image_rect_raw
  timeshift_cam_imu: 0.004843498803838562
cam1:
  T_cam_imu:
  - [0.9999983261285631, -0.0017033519994751791, -0.0006680808616477473, 0.004439738538511637]
  - [0.001700283684319089, 0.9999881269878522, -0.0045667185943291935, 0.004896249708055788]
  - [0.0006758516587642526, 0.004565575023240513, 0.9999893493179026, -0.02911243065289644]
  - [0.0, 0.0, 0.0, 1.0]
  T_cn_cnm1:
  - [0.9999921445556381, 2.4634895347942934e-06, -0.003963687796743136, -0.09516983706587909]
  - [-3.988430939995851e-06, 0.9999999259872779, -0.00038472007252608666, -2.087431575104303e-05]
  - [0.003963686555625934, 0.0003847328592740065, 0.9999920705533214, 4.997549897848379e-05]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [0]
  camera_model: pinhole
  distortion_coeffs: [-0.01219707044444792, -0.0022219835518296786, -0.00026273576500083953, -0.0023201942128480654]
  distortion_model: radtan
  intrinsics: [390.4607337101762, 389.8851228594401, 327.0788360457071, 239.6763331121837]
  resolution: [640, 480]
  rostopic: /camera/infra2/image_rect_raw
  timeshift_cam_imu: 0.004854616068191737
```

After performing Camera-IMU calibration for the stereo pair on the D455:
```yaml
cam0:
  T_cam_imu:
  - [0.9999576721050568, 0.002813023587983025, 0.008760188156069906, 0.031684661051907696]
  - [-0.002873968824050705, 0.999971702726438, 0.006952269383966502, -0.009562342343691004]
  - [-0.008740383368862088, -0.006977151616691762, 0.9999374605713517, -0.026536371869680546]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [1]
  camera_model: pinhole
  distortion_coeffs: [-0.007464439673137539, -0.003239692692021013, -0.0027135375803431913, -0.0013127396299211385]
  distortion_model: radtan
  intrinsics: [389.96490426865773, 389.2168790608396, 321.37671021599266, 237.9700394936505]
  resolution: [640, 480]
  rostopic: /camera/infra1/image_rect_raw
  timeshift_cam_imu: 0.005097332223784535
cam1:
  T_cam_imu:
  - [0.999978140120791, 0.003283281298222775, 0.005739280833026187, -0.06325355907658448]
  - [-0.003326555765563399, 0.9999659890609481, 0.007546836959830824, -0.009635047925568378]
  - [-0.005714307246044626, -0.007565764024631763, 0.9999550519435468, -0.026441387657387436]
  - [0.0, 0.0, 0.0, 1.0]
  T_cn_cnm1:
  - [0.999995326278841, 0.0004491834172957683, -0.0030241783548732998, -0.09501402751903543]
  - [-0.0004473753332999051, 0.9999997208109012, 0.0005985260404348031, -4.265060619781037e-05]
  - [0.003024446358527868, -0.0005971703002915169, 0.9999952480446329, -6.680791949993797e-06]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [0]
  camera_model: pinhole
  distortion_coeffs: [-0.007165804851485035, -0.0027169873179884456, -0.002832920072061058, -0.0015160926166674147]
  distortion_model: radtan
  intrinsics: [389.6872556636675, 389.1806491537163, 322.64071851096037, 237.79788394086225]
  resolution: [640, 480]
  rostopic: /camera/infra2/image_rect_raw
  timeshift_cam_imu: 0.005098617652466733
```

The extrinsics reported by both cameras is: 
[image]
Note: I don't see an "Accel" to "Infrared 2" transform on the D457, but I do on the D455?

So: are the extrinsics indeed different between the two camera models, and `rs-enumerate-devices -c` doesn't correctly reflect that? @MartyG-RealSense I would appreciate your answer here!

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455/D457 | 
| Firmware Version                   | D455: 5.15.0.2, D457: 5.13.1.53 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | x86-64 Thinkpad Laptop |
| SDK Version                            | 2.55.1.0         | 
| Language                            |  C++/ROS/Kalibr                         | 
| Segment			|  Robot                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-06-12:
Hi @micahreich  When a D457's physical mode switch is pushed into the 'USB' position in order to use it on a USB cable instead of a GMSL cable then the camera is detected as a D455 and has the same performance and features as a D455.

Each individual RealSense camera unit has unique intrinsics and extrinsics due to the manufacturing process in the factory.

I have an Accel to Infrared 2 extrinsic reading from `rs-enumerate-devices - c` on my own D457, detected as D455 on a USB connection.  Infrared 2 will not be accessible if the camera is being detected as being on a USB 2.1 connection instead of USB 3.2.

micahreich, 2024-06-12:
Thanks for the reply @MartyG-RealSense. I understand that each RealSense camera will have different intrinsics/extrinsics due to manufacturing tolerances, but for each D455, for example, the IMU should be roughly in the same place on the board. I'm wondering: is it true that the D457 IMU is indeed in a different location (I believe it to be behind the right imager, rather than somewhere between the left and right imager for the D455) than in the D455 IMU? I just want to find out if my IMU-Camera extrinsic calibration results are reasonable or not.

Could you elaborate more on:
> Infrared 2 will not be accessible if the camera is being detected as being on a USB 2.1 connection instead of USB 3.2

The IR2 stream is unavailable? Would you know why the D457 is being detected as USB 2.1 even though the D455 is detected as USB 3.2, and I'm using the exact same cable and port?

MartyG-RealSense, 2024-06-12:
If a D457 is being detected as USB 2.1 then it is usually because the micro-sized end of the USB cable in the base of the camera needs to be unplugged, turned around the other direction and re-inserted (USB Type-C cables are two-way insertion at the micro-sized end).  It should be detected as USB 3.2 when the connector is reversed.

MartyG-RealSense, 2024-06-18:
Hi @micahreich  Do you require further assistance with this case, please?  Thanks!

micahreich, 2024-06-18:
@MartyG-RealSense I was hoping you could answer whether the D457 camera-IMU extrinsics are indeed different from D455 extrinsics (beyond differences that occur in manufacturing)?

closed 2024-07-04 by MartyG-RealSense