realsenseai/librealsense #13023: pipeline = rs.pipeline() RuntimeError: failed to convert special folder: errno=42

atong-atong, 2024-06-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D405 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {win10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python3.8}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I installed Conda on my computer and created two Python environments, one is version 3.7 and the other is version 3.8. When I run Pyralsense2 on version 3.7 of Python, it can run normally and call the D405 camera. However, when I run Pyralsense2 on version 3.8 of Python, I will report an error pipeline=rs. pipeline()
RuntimeError: failed to convert special folder: errno=42

MartyG-RealSense, 2024-06-12:
Hi @atong-atong  This error may occur if your Windows computer does not have a Documents folder.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/12910#issuecomment-2100569892

atong-atong, 2024-06-12:
There is a Documents folder, but I have found the reason for the error. The reason is a version issue. I downloaded the latest version 2.55.1.6486 directly from pip install pyrealsene2, which will cause this error. Regardless of whether there is a Doc folder or not, the solution is to switch to a lower version, pip install pyrealsense2==2.53.1.4623, After installing the lower version, both Python 3.8 and Python 3.7 can run normally. This should be a bug, and I hope you can solve it

MartyG-RealSense, 2024-06-12:
Ideally there should only be one Python 3 version on the computer so that the **pip install pyrealsense2** installation process can detect which wheel (.whl) package to use, as there are separate pyrealsense2 .whl files for each Python 3 version supported.  

https://pypi.org/project/pyrealsense2/#files

So given that **pip install pyrealsense2** is intended to detect one particular Python 3 version and use the appropriate .whl file to install the wrapper for that Python version, it may not be a bug.  I'm pleased to hear though that you found a workaround for your particular two-Python setup.

atong-atong, 2024-06-13:
I installed two versions of Python using Conda, so it shouldn't be related to what you said. I also tried to test it on a computer with only one version of Python, and found that version 2.55.1.6486 of pyrealsene2 cannot start properly, but it can start normally after using version 2.53.1.4623. Therefore, there may be a bug in version 2.55.1.6486

MartyG-RealSense, 2024-06-13:
There have been a few recent cases of this error on SDK 2.55.1 at https://github.com/IntelRealSense/librealsense/issues/12929 and https://github.com/IntelRealSense/librealsense/issues/12941 and https://github.com/IntelRealSense/librealsense/issues/13027

However, they all involved the following message:

************

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:

RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:{):
failed to convert special folder: errno=42

*************

Your version of this error apparently occurs in relation to the pyrealsense2 wrapper rather than the RealSense Viewer tool.

I will discuss this case with my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2024-06-14:
My Intel RealSense colleagues provided advice about the errno=42 problem.

In SDK 2.55.1 there was a pipeline context crash problem due to an uninitialized json variable.  The issue was fixed on the development branch of the SDK and also included in the 2.55 release.  But sometimes 2.55 still has a problem due to mixed libraries. 

Clean the folders and make sure that you have the correct version of the SDK libraries.  Builds made from the development branch didn’t have the problem after the fix.

https://github.com/IntelRealSense/librealsense/tree/development

********

Another possible cause of the error is that the SDK uses a json file to store settings.  Sometimes, due to an app not cleanly closing down, those json files are empty and the SDK will crash on the next time that it is launched.  The json files can be found at the following location on a Windows computer:

`C:\Users\[user]\AppData\Roaming`

The folder contains files like 'realsense-config.json'.  The workaround is to delete the empty json files at this location and then relaunch.

closed 2024-06-28 by MartyG-RealSense