realsenseai/librealsense #13025: not recugnize my camera

naorwaiss, 2024-06-12:
hi i use nevida jetson nano 
pytohn 3.9.12
sdk 2.48
cmake 3.27
the installation going well and i can use pyrealsnes2 
the problem is that i run code to view the camera:

fhsbg1@fhsbg1-desktop:~$ python3 test2.py
Traceback (most recent call last):
  File "/home/fhsbg1/test2.py", line 183, in <module>
    loop.run_until_complete(process_frames(queue))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/home/fhsbg1/test2.py", line 67, in process_frames
    pipeline.start(config)
RuntimeError: No device connected

the real problem is that when i run the command:
v4l2-ctl --list-devices

this is detect me the camera:

thanks for the help

MartyG-RealSense, 2024-06-12:
Hi @naorwaiss  Is the camera detected if you run the script in sudo admin permissions mode, please?

`sudo python3 test2.py`

naorwaiss, 2024-06-12:
fhsbg1@fhsbg1-desktop:~$ sudo python3 test2.py
Traceback (most recent call last):
  File "/home/fhsbg1/test2.py", line 183, in <module>
    loop.run_until_complete(process_frames(queue))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/home/fhsbg1/test2.py", line 76, in process_frames
    cv2.namedWindow("Video Stream", cv2.WINDOW_NORMAL)
cv2.error: OpenCV(4.10.0) /io/opencv/modules/highgui/src/window.cpp:1284: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvNamedWindow'

this is somthing that i dont enconter with

MartyG-RealSense, 2024-06-12:
Please create a test script with the code at the link below and run the script file with sudo.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

If the script works correctly then it could indicate that the problem is in your test2.py script.

naorwaiss, 2024-06-12:
fhsbg1@fhsbg1-desktop:~$ sudo python3 test3.py 
/home/fhsbg1/test3.py:23: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if y%20 is 19:
Traceback (most recent call last):
  File "/home/fhsbg1/test3.py", line 5, in <module>
    pipeline = rs.pipeline()
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

this is the from the link you send

MartyG-RealSense, 2024-06-13:
Which firmware driver version is installed in your camera, please?  You can find the firmware version using the command below: 

`rs-enumerate-devices -s`

The firmware version that is recommended to be used with SDK 2.48.0 is **5.12.14.50**.  Using a newer firmware version with SDK 2.48.0 could result in errors.

naorwaiss, 2024-06-13:
ok so how i can downgrade my camera to lower frimware?

MartyG-RealSense, 2024-06-13:
It would depend on which RealSense camera model you have.  You could downgrade to 5.12.14.50 if you have a D415 or D435 / D435i camera.  

You should not downgrade though if you have a D455 camera as the minimum recommended firmware for that type of camera is 5.13.0.50.  It would be okay to use 5.13.0.50 with a D455 if you use SDK version 2.50.0 (the version after 2.48.0).

naorwaiss, 2024-06-13:
ok i got d435 and 435 with the newest version 
what the prucedure to downgrade to 5.12.14.50?

MartyG-RealSense, 2024-06-13:
1.  Please visit the firmware releases page at the link below and scroll down the list of firmware versions until you reach 5.12.14.50.

https://dev.intelrealsense.com/docs/firmware-releases-d400

[image]

2.  Download the firmware **.bin** file for 5.12.14.50 to your computer.

3.  Open to the RealSense Viewer and go to the **More** option near the top of the Viewer's options side-panel.  Select the **Update Firmware** option from the drop-down menu.

4.  A file selection box will pop up.  Navigate to the location on your computer that you downloaded the firmware .bin file to and select the file.  The firmware downgrade to 5.12.14.50 should now automatically begin.

dmipx, 2024-06-18:
Hi.
First of all - share output of `v4l2-ctl --list-devices`
Then, verify kernel output while reconnecting camera device and share with us
1.disconnect camera
2.`dmesg -C`
3.connect camera
4.`dmesg`

You kernel should have support iio-hid in case you have camera with IMU module, like d435i.
verify video nodes:
`ls /dev/video*`
in case video0 is camera first video node, run capture test with
`v4l2-ctl -d 0 --stream-mmap`

MartyG-RealSense, 2024-06-25:
Hi @naorwaiss  Do you require further assistance with this case, please?  Thanks!

naorwaiss, 2024-06-25:
#how to download all the thing to the jestson and simple tasks :

# drone-camera-forproject

-----------------------------------------------step 1 -------------------------------------------------------------
# python3.9.12 installation

sudo apt update
sudo apt upgrade
sudo apt install wget
sudo apt install -y build-essential zlib1g-dev libncurses5-dev libgdbm-dev libnss3-dev libssl-dev libreadline-dev libffi-dev libsqlite3-dev libbz2-dev
wget https://www.python.org/ftp/python/3.9.12/Python-3.9.12.tgz

# python3.9.12 installation
tar -xf Python-3.9.12.tgz
cd Python-3.9.12
./configure --enable-optimizations
make -j4  # You can change the number after -j to match the number of CPU cores for faster compilation. (this take like 30 min)
sudo make altinstall

# after the installation finish - make the python as defount (python and python3)
sudo update-alternatives --install /usr/bin/python3 python3 /usr/local/bin/python3.9 1
sudo update-alternatives --install /usr/bin/python python /usr/local/bin/python3.9 1

alternativ - check if the instalation finish - python --version /python3 --version.

#fix pip problem
sudo apt update
sudo apt install python3.9-distutils --- this command do some problem , ignore the problem 
wget https://bootstrap.pypa.io/get-pip.py
sudo python3.9 get-pip.py
sudo python3.9 -m pip install --upgrade pip

------------------------------------------------------------------step 2------------------------------------------------

# install cmake

cmake --version  #check you defoult camke version - remmber it to 

sudo apt-get install libcurl4-openssl-dev
sudo apt-get install curl   ##install curl
mkdir ~/cmake-source
cd ~/cmake-source
curl -LO https://cmake.org/files/v3.27/cmake-3.27.4.tar.gz
tar -xzvf cmake-3.27.4.tar.gz

# build and inastall cmake

cd cmake-3.27.4
./bootstrap
make -j2  # You can change the number after -j to match the number of CPU cores for faster compilation. (the bootstrap and make take 30 - 60 min)
sudo make install

#check the version again
cmake --version

-------------------------------------------------------step 3 --------------------------------------------------------------------------

# intelrealsense
#now go to intelreal sense git downlad and take some source file to install the sdk
#from here:  https://github.com/IntelRealSense/librealsense/releases/
need to take the source file with zip variation

unzip librealsense-2.48.0.zip -d /home/drone/naorp/cam

#got to this cd 
https://stackoverflow.com/questions/62134563/how-to-give-permission-to-intel-realsense-camera-on-ubuntu
mkdir build
cd build
sudo apt-get install libxinerama-dev
sudo apt-get install libxcursor-dev
#run this line to use the cmake
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/local/bin/python3.9 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=false

# installation
make -j2
sudo make install

# bashrc
vim ~/.bashrc
#use i to insert data
#to exit use esc and then :wq

#put this at bashrc - mabey need to fix it - lets see what at my jetson but

export PATH=$PATH:/usr/local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.9/site-packages
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.9
export PYTHONPATH=$PYTHONPATH:/home/drone/naorp/cam/librealsense-2.48.0/build/wrappers/python

#after exit
source ~/.bashrc

-----------------------------------------------------------------------step 4------------------------------------------------

need to do this aldo to get udev bitch
and check the camera frimware 
https://stackoverflow.com/questions/62134563/how-to-give-permission-to-intel-realsense-camera-on-ubuntu

#lets start 
clone https://github.com/IntelRealSense/librealsense
#cd to the clone 
cd librealsense
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/

#run this 2 command to update the ne udev rule 
sudo udevadm control --reload-rules
sudo udevadm trigger

check the realsense-veiwer to cehck if you got ditect camera 

-----------------------------------------------------end of camera installation-------------------------------------------------
-----------------------------------------------------------drone installation --------------------------------------------------
--------------------------------------------------------------step 1 -----------------------------------------------------------
# mavlink instalation
#at this point we can install the mavproxy
#can see this here also https://www.youtube.com/watch?v=nIuoCYauW3s

i do some organize installation list for jetson nano.
thanks for the help

MartyG-RealSense, 2024-07-04:
Thanks so much @naorwaiss for your guide!

Do you need anything else from us, please?  Thanks again!