realsenseai/librealsense #13027: error

zxcvbnmhkfghffgh, 2024-06-13:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:nullptr):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-06-13:
A cause of a **failed to convert special folder: errno=42** error may be if your Windows computer does not have a **Documents** folder.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/12910#issuecomment-2100569892

MartyG-RealSense, 2024-06-14:
My Intel RealSense colleagues have provided further advice about the errno=42 error at https://github.com/IntelRealSense/librealsense/issues/13023#issuecomment-2167504813

closed 2024-06-28 by MartyG-RealSense