realsenseai/librealsense #13028: RealSense, No device connected

mrortach, 2024-06-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | v2.50.0 | 
| Operating System & Version |   Windows 11  | 
| Platform                                 | PC  |
| SDK Version                            |  v2.50.0   | 
| Language                            |  opencv/python                     | 
| Segment			| YOLOv10                  | 

### Error starting RealSense pipeline: No device connected

```import pyrealsense2 as rs
import numpy as np
import cv2
import torch
from ultralytics import YOLO

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Load YOLO model
model = YOLO(yolov10x.pt)

try:
    # Start streaming and catch potential errors
    try:
        print("Starting RealSense pipeline...")
        pipeline.start(config)
        print("RealSense pipeline started successfully.")
    except RuntimeError as e:
        print(f"Error starting RealSense pipeline: {e}")
        pipeline = None

    if pipeline:
        while True:
            # Wait for a coherent pair of frames: depth and color
            frames = pipeline.wait_for_frames()
            color_frame = frames.get_color_frame()
            if not color_frame:
                continue

            # Convert images to numpy arrays
            color_image = np.asanyarray(color_frame.get_data())

            # Perform object detection
            results = model(color_image)

            # Display results
            for result in results:
                boxes = result.boxes
                for box in boxes:
                    x1, y1, x2, y2 = map(int, box.xyxy[0])
                    conf = box.conf[0]
                    cls = int(box.cls[0])
                    label = f'{model.names[cls]} {conf:.2f}'
                    cv2.rectangle(color_image, (x1, y1), (x2, y2), (255, 0, 0), 2)
                    cv2.putText(color_image, label, (x1, y1 - 10), cv2.FONT_HERSHEY_SIMPLEX, 0.9, (255, 0, 0), 2)

            # Show the image
            cv2.imshow('RealSense', color_image)
            if cv2.waitKey(1) & 0xFF == ord('q'):
                break
finally:
    # Stop streaming if the pipeline was started
    if pipeline:
        print("Stopping RealSense pipeline...")
        pipeline.stop()
        print("RealSense pipeline stopped.")
    # Check if destroyAllWindows exists before calling it
    if hasattr(cv2, 'destroyAllWindows'):
        cv2.destroyAllWindows()

mrortach, 2024-06-14:
When installing pyrealsense2, it is crucial that the version matches the SDK version exactly. For example, if the latest SDK version is 2.54.2.5684, then you should install pyrealsense2 with the same version using the command pip install pyrealsense2==2.54.2.5684.

MartyG-RealSense, 2024-06-14:
Thanks very much @mrortach for adding the details of your solution here!

MartyG-RealSense, 2024-06-28:
Case closed due to solution achieved and no further comments received.

closed 2024-06-28 by MartyG-RealSense

eyast, 2025-01-10:
Hi - thanks for the note! I have SDK 2.50.0.3785, but that version can't be found when trying to install pyrealsense2 - should i look at another SDK /pip package? thanks!

MartyG-RealSense, 2025-01-10:
Hi @eyast  The name of the pyrealsense2 pip package for use with SDK 2.50.0 is **2.50.0.3812**, installed with the command `pip install pyrealsense2==2.50.0.3812`

https://pypi.org/project/pyrealsense2/2.50.0.3812/

eyast, 2025-01-12:
works like a charm thank you @MartyG-RealSense  - I had to downgrade Python to 3.9 then installed 2.50.0.3812. Thanks for your help, appreciated !!

MartyG-RealSense, 2025-01-12:
You are very welcome, @eyast - It's great to hear that you achieved a solution!

Official pyrealsense2 pip install support for Python 3.10 was added in librealsense 2.53.1.  3.11 support was added in 2.54.1 and Python 3.12 in 2.56.1.  

That is for pip installation on PC computers though.  For pip install on Arm-based computing boards, Python 3.10 is the maximum Python version supported at the time of writing this.

If a recent Python version does not yet have pip package support then you can build the pyrealsense2 wrapper from source code instead and it should work.  So if you need to use librealsense 2.50.0 and cannot upgrade to a newer librealsense version then you may need to build pyrealsense2 from source code instead of performing pip install if you want to use newer Python versions than 3.10.

eyast, 2025-01-13:
Hi @MartyG-RealSense  thanks for your help, I was able to build the wrapper for my version of python and it works, thank you!
One area I didn't manage to get working is intellisense (the ability to hover on a namespace in my IDE to read the docstrings etc). Do you have any thoughts on what I should do to get this working?

eyast, 2025-01-13:
Amazing - thanks @MartyG-RealSense  !!! This is a lot better than what I did (installed `mypy`, then ran `stubgen -m pyrealsense2`.. I am very grateful for your help, I now have pyrealsense2 working with type hints on Python 3.10. thank you so much !!

MartyG-RealSense, 2025-01-13:
Excellent news - thanks very much for the update!  :)