realsenseai/librealsense #13030: realsense2.dll compilation problems

mrortach, 2024-06-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense SR305 | 
| Firmware Version                   | v2.50.0 | 
| Operating System & Version | Windows 11  | 

Severity	Code	Description	Project	File	Line	Suppression State	Details
Warning	LNK4199	/DELAYLOAD:/realsense2d.dll ignored; no imports found from /realsense2d.dll	rs-depth-quality	C:***\librealsense-master\build\tools\depth-quality\LINK	1		

Severity	Code	Description	Project	File	Line	Suppression State	Details
Warning	LNK4199	/DELAYLOAD:/realsense2-gld.dll ignored; no imports found from /realsense2-gld.dll	rs-depth-quality	C***\build\tools\depth-quality\LINK	1		

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md#windows

[image]

MartyG-RealSense, 2024-06-13:
Hi @mrortach  It is not necessary to compile realsense2.dll when using the Windows version of the RealSense SDK.  If you have installed the full RealSense SDK on Windows with the **Intel.RealSense.SDK-WIN10** installer file then you can find a pre-built version of this dll at the location below on your Windows computer.

`C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin > x64`

[image]

You can also find a pre-built version of pyrealsense2.pyd in this folder for use with Python 3.7 only.

mrortach, 2024-06-13:
Ok then why am I getting this error when testing with yolov10: Starting RealSense pipeline...
Error starting RealSense pipeline: No device connected
Why can't the device be found? I don't understand.

MartyG-RealSense, 2024-06-13:
I see that you have posted at the **ultralytics** GitHub at https://github.com/ultralytics/ultralytics/issues/13515 and they have responded with advice.

mrortach, 2024-06-13:
Problem not solved.

MartyG-RealSense, 2024-06-13:
Does it make a difference if you launch your script in **sudo** admin permissions mode?  For example, `sudo python3 test.py`

I also note that your script does not have an `import pyrealsense2 as rs` line in its header.  So even if you have successfully installed the RealSense Python compatibility wrapper (pyrealsense2), the script may not know what to do with the RealSense-specific Python instructions if pyrealsense2 has not been not imported into your script.

mrortach, 2024-06-13:
I missed it while adding the message (import pyrealsense2 as rs) I edited the message. sudo python3 test.py I'm using Windows, sudo will not work. 'sudo' is not recognized as an internal or external command,
operable program or batch file.

MartyG-RealSense, 2024-06-13:
For some RealSense users the SR300 / SR305 cameras had an issue where they could disconnect immediately upon activation if a certain amount of power was drawn from them.  You can test whether your SR305 has this issue when used with Python by setting **Laser Power** to zero.  This can be done by adding the code below to the line immediately after the pipe start instruction.

```
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
depth_sensor.set_option(rs.option.laser_power, 0)
```

mrortach, 2024-06-13:
And this part confuses me.
[image]

I don't know if the code is correct, but I got an error again.

Starting script...
Starting pipeline...
RuntimeError: No device connected

```python
import cv2
import pyrealsense2 as rs
import torch
import numpy as np  # numpy kütüphanesini import ediyoruz
import supervision as sv  # type: ignore
from ultralytics import YOLOv10

print("Starting script...")

# Derinlik ve renk akışlarını yapılandır
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

try:
    print("Starting pipeline...")
    pipeline_profile = pipeline.start(config)
    print("Pipeline started successfully.")
    
    # Lazer gücünü sıfıra ayarla
    device = pipeline_profile.get_device()
    depth_sensor = device.query_sensors()[0]
    depth_sensor.set_option(rs.option.laser_power, 0)
except RuntimeError as e:
    print(f"RuntimeError: {e}")
    exit(1)

# YOLOv10 modelini yükle
model = YOLOv10('yolov10x.pt')
print("Model loaded.")

try:
    while True:
        # Tutarlı bir çift çerçeve bekleyin: derinlik ve renk
        frames = pipeline.wait_for_frames()
        color_frame = frames.get_color_frame()
        if not color_frame:
            continue

        # Görüntüleri numpy dizilerine dönüştür
        color_image = np.asanyarray(color_frame.get_data())

        # Nesne tespiti yap
        results = model(color_image)

        # Sonuçları görüntüle
        results.show()

finally:
    # Yayını durdur
    pipeline.stop()
    print("Pipeline stopped.")

MartyG-RealSense, 2024-06-13:
Firmware for the SR300 / SR305 camera models is not downloadable.  It can only be updated in the RealSense Viewer.  However, if your SR305 is working normally in the Viewer then there is no need to change the firmware version.  Even with the latest RealSense camera models, they will sometimes work fine in the Viewer but not be detected by pyrealsense2 for no clear reason.

Is there any difference if you remove 'config' from the brackets of the pipe start instruction so that the program ignores the custom configuration lines?  The SR305 camera will use 640x480 and 30 FPS by default anyway.

`pipeline_profile = pipeline.start() `

mrortach, 2024-06-13:
Starting script...
Starting pipeline...
RuntimeError: No device connected

```
python
import cv2
import pyrealsense2 as rs
import torch
import numpy as np  # numpy kütüphanesini import ediyoruz
import supervision as sv  # type: ignore
from ultralytics import YOLOv10

print("Starting script...")

# Derinlik ve renk akışlarını yapılandır
pipeline = rs.pipeline()

# config satırlarını kaldırıyoruz
# config = rs.config()
# config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
# config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

try:
    print("Starting pipeline...")
    # config'siz başlatma
    pipeline_profile = pipeline.start()
    print("Pipeline started successfully.")
    
    # Lazer gücünü sıfıra ayarla
    device = pipeline_profile.get_device()
    depth_sensor = device.query_sensors()[0]
    depth_sensor.set_option(rs.option.laser_power, 0)
except RuntimeError as e:
    print(f"RuntimeError: {e}")
    exit(1)

# YOLOv10 modelini yükle
model = YOLOv10('yolov10x.pt')
print("Model loaded.")

try:
    while True:
        # Tutarlı bir çift çerçeve bekleyin: derinlik ve renk
        frames = pipeline.wait_for_frames()
        color_frame = frames.get_color_frame()
        if not color_frame:
            continue

        # Görüntüleri numpy dizilerine dönüştür
        color_image = np.asanyarray(color_frame.get_data())

        # Nesne tespiti yap
        results = model(color_image)

        # Sonuçları görüntüle
        results.show()

finally:
    # Yayını durdur
    pipeline.stop()
    print("Pipeline stopped.")

MartyG-RealSense, 2024-06-13:
Next, please try starting the camera without the profile reference.

`pipeline.start(config)`

mrortach, 2024-06-13:
Starting script...
Starting pipeline...
RuntimeError: No device connected

```

import cv2
import pyrealsense2 as rs
import torch
import numpy as np
import supervision as sv  # type: ignore
from ultralytics import YOLOv10

print("Starting script...")

# Initialize the pipeline
pipeline = rs.pipeline()

try:
    print("Starting pipeline...")
    # Start the pipeline without a configuration
    pipeline_profile = pipeline.start()
    print("Pipeline started successfully.")
    
    # Set laser power to zero
    device = pipeline_profile.get_device()
    depth_sensor = device.query_sensors()[0]
    depth_sensor.set_option(rs.option.laser_power, 0)
except RuntimeError as e:
    print(f"RuntimeError: {e}")
    exit(1)

# Load the YOLOv10 model
model = YOLOv10('yolov10x.pt')
print("Model loaded.")

try:
    while True:
        # Wait for a consistent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        color_frame = frames.get_color_frame()
        if not color_frame:
            continue

        # Convert images to numpy arrays
        color_image = np.asanyarray(color_frame.get_data())

        # Perform object detection
        results = model(color_image)

        # Display results
        results.show()

finally:
    # Stop the pipeline
    pipeline.stop()
    print("Pipeline stopped.")

mrortach, 2024-06-13:
Isn't it related to the Python version? I am using 3.11 and Python 3.10.14 is selected as the environment.

MartyG-RealSense, 2024-06-13:
If you built the Python wrapper from source code then the Python version should not matter.  

Versions would matter if the wrapper was installed from packages with **pip install pyrealsense2** as the pip install method currently only supports Python 3.7 to Python 3.10 on Arm devices.

mrortach, 2024-06-13:
I have a Raspberry Pi 5 and I had the same problem. How can I solve this problem? I need to test it. If it works well, I am thinking of buying the new version, but I cannot test it.

MartyG-RealSense, 2024-06-14:
There have been very few reports at the time of writing this regarding using librealsense with Raspberry Pi 5, but the early indications so far are that a Pi 5 has the same kind of problems that Pi 4 RealSense users experience.  An example of such a report is at github.com/IntelRealSense/realsense-ros/issues/3093

mrortach, 2024-06-14:
Name: pyrealsense2
Version: 2.55.1.6486
Summary: Python Wrapper for Intel Realsense SDK 2.0.
Home-page: https://github.com/IntelRealSense/librealsense
Author: Intel(R) RealSense(TM)
Author-email: realsense@intel.com
License: Apache License, Version 2.0
Location: c:\users\mrort\anaconda3\envs\esneksan_yolov8\lib\site-packages
Requires: 
Required-by:

pip install pyrealsense2==2.50.0.3819
ERROR: Could not find a version that satisfies the requirement pyrealsense2==2.50.0.3819 (from versions: 2.53.1.4623, 2.54.1.5216, 2.54.1.5217, 2.54.2.5684, 2.55.1.6486)
ERROR: No matching distribution found for pyrealsense2==2.50.0.3819

[image]

MartyG-RealSense, 2024-06-14:
It's great to hear that you were successful!

closed 2024-06-21 by MartyG-RealSense