realsenseai/librealsense #13032: Noisy gyroscope

B-Paweekorn, 2024-06-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.16.00.01  | 
| Operating System & Version |  Ubuntu 22.04   | 
| Kernel Version (Linux Only)    | 6.5.0-35-generic                                    | 
| Platform                                 | PC  |
| SDK Version                            | SDK2.0 (v2.50.0)                            | 

### Issue Description
My gyroscope data seems very noisy, as shown in the video. Is there any solution to fix this?

https://github.com/IntelRealSense/librealsense/assets/122732439/8abdabfa-346c-4e9c-9c94-be2286141dd8

MartyG-RealSense, 2024-06-13:
Hi @B-Paweekorn  RealSense SDK version 2.55.1 added an option to set the sensitivity of the gyro with the programming instruction **RS2_OPTION_GYRO_SENSITIVITY**.  However, the sensitivity is not currently adjustable in the RealSense Viewer tool.  During tests in the Viewer that I performed, it seemed to be the up-down tilt action that contributed most to the gyro axes spinning very fast.  Left-right turn of the camera produced more stable axis movement.

MartyG-RealSense, 2024-06-21:
Hi @B-Paweekorn  Do you require further assistance with this case, please?  Thanks!

B-Paweekorn, 2024-06-24:
Yes, please. If there are any additional solutions or if you could help identify the root cause of the problem, I would appreciate it.

MartyG-RealSense, 2024-06-24:
I compared the movement of the gyro axes in the Viewer to the movement represented in the **rs-motion**  C++IMU example program.  The movement was significantly more stable in rs-motion.  

The C++ code of rs-motion includes an 'alpha' value that can be configured to weight the IMU data towards the gyro (at the risk of drift) or towards the accelerometer (at the risk of higher sensitivity to disturbances).  So it may be that rs-motion's apparent high stability means that the issue with the strongly fluctuating axes is in the RealSense Viewer specifically.

MartyG-RealSense, 2024-06-28:
Hi @ljysimple  If you are experiencing an over-sensitive gyro then the gyro sensitivity feature added in librealsense 2.55.1 may be helpful if it is possible for you to update your SDK version.

Issues in SDK versions older than the current latest one that require code changes to be solved cannot be retroactively fixed by Intel though, unfortunately.

A key difference betwen the Linux and Windows versions of librealsense is that the Linux version runs on a **UVC** backend whilst the Windows version uses a different backend called **Microsoft Media Foundation**.

ljysimple, 2024-06-28:
Thank you. The 2.50 version I downloaded mainly corresponds to the 2.32-ros version. Is there any corresponding realsense-ros version for the 2.55 version?

MartyG-RealSense, 2024-06-28:
2.51.1 can be used with wrapper 2.3.2 as it is very similar to 2.50.0.  Development on the RealSense ROS1 wrapper ceased after 2.50.0 and it has not been updated for more recent librealsense versions.

Development of the RealSense ROS2 wrapper continues and there is a ROS2 wrapper for 2.55.1.

ljysimple, 2024-06-28:
My version is ros1. After 2.51 is installed, it can support 2.3.2？

MartyG-RealSense, 2024-06-28:
Yes, 2.51.1 is compatible with 2.3.2 as 2.51.1 only had minor changes from the previous version and so remains compatible with 2.3.2.

ljysimple, 2024-06-28:
Thank you for your answer, I will try to solve it as soon as possible

MartyG-RealSense, 2024-07-04:
Hi @B-Paweekorn and @ljysimple  Do either of you have an update about this case that you can provide, please?  Thanks!

closed 2024-07-18 by MartyG-RealSense