realsenseai/librealsense #13036: D405 IR-Cut-Filter

fredwrnr, 2024-06-14:
Hey guys,
we use a D405 module and want to use IR light to use in the dark. 
I know that the camera has a IR-cut-filter but I need to know if its on the cover glass or the lens? I know it voids the warraty but would removing the cover glass make the camera ir-sensitive?

Thanks in advance!

MartyG-RealSense, 2024-06-14:
Hi @fredwrnr  The D405 camera contains a **D401** module.  The specification details for the IR sensor on the D401 module list the IR-Cut filter.  This indicates that the filter is likely to be on the lens of the IR sensor and that removing the front glass would not remove the IR-Cut filter.

[image]

MartyG-RealSense, 2024-06-28:
Case closed due to no further comments received.

closed 2024-06-28 by MartyG-RealSense

danlevy100, 2024-07-21:
Hello,
I share a similar issue: I want to use an external IR dot projector to improve measurements of featureless objects with a D405/D401. I would love to get some instructions on how to remove the IR cut filter. Many thanks!

MartyG-RealSense, 2024-07-21:
Hi @danlevy100  There is not information available about removing the IR filter from a sensor.  If you are planning to use an external projector  though then you could instead consider using a **visible** projector instead of an IR pattern one to add texture to a plain surface.

Information about use of visible projectors can be found in Intel's projector guide at the link below.

https://dev.intelrealsense.com/docs/projectors#6-the-consumer-front-projector

Another alternative to using the dot pattern is to increase light illumination in the real-world scene, as RealSense 400 Series cameras can make use of any kind of visible light.

danlevy100, 2024-07-22:
Thanks for the quick response!
I've been looking for compact visible light dot projectors, but unfortunately have not found anything as compact and cheap as exists for IR. Since I have uniform-color surfaces, I am not sure if increasing the ambient light would work so well, but I'll give it a shot.

danlevy100, 2024-07-24:
Thanks Marty! Let's see how that works out.