realsenseai/librealsense #13037: 报的什么错

admin123yan, 2024-06-14:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25501, json_settings:{):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-06-14:
Hi @admin123yan  My Intel RealSense colleagues provided advice about the errno=42 problem.

In SDK 2.55.1 there was a pipeline context crash problem due to an uninitialized json variable.  The issue was fixed on the development branch of the SDK and also included in the 2.55 release.  But sometimes 2.55 still has a problem due to mixed libraries. 

Clean the folders and make sure that you have the correct version of the SDK libraries.  Builds made from the development branch didn’t have the problem after the fix.

https://github.com/IntelRealSense/librealsense/tree/development

Another possible cause of the error is that the SDK uses a json file to store settings.  Sometimes, due to an app not cleanly closing down, those json files are empty and the SDK will crash on the next time that it is launched.  The json files can be found at the following location on a Windows computer:

`C:\Users\[user]\AppData\Roaming`

The folder contains files like 'realsense-config.json'.  The workaround is to delete the empty json files at this location and then relaunch.

***************

嗨@admin123yan 我的 Intel RealSense 同事提供了有关 errno=42 问题的建议。

在 SDK 2.55.1 中，由于未初始化的 json 变量，存在管道上下文崩溃问题。该问题已在 SDK 的开发分支上修复，也包含在 2.55 版本中。但有时 2.55 仍然会因为混合库而出现问题。

清理文件夹并确保您拥有正确版本的 SDK 库。修复后，从开发分支构建的版本没有出现问题。

https://github.com/IntelRealSense/librealsense/tree/development

错误的另一个可能原因是 SDK 使用 json 文件来存储设置。有时，由于应用程序未干净关闭，这些 json 文件为空，SDK 将在下次启动时崩溃。 json 文件可以在 Windows 计算机上的以下位置找到：

`C:\Users\[user]\AppData\Roaming`

该文件夹包含“realsense-config.json”等文件。解决方法是删除此位置的空 json 文件，然后重新启动。

MartyG-RealSense, 2024-06-21:
Do you require further assistance with this case, please?  Thanks!

您需要进一步协助处理此案吗？谢谢！

MartyG-RealSense, 2024-06-28:
Case closed due to no further comments received.

由于未收到进一步评论，案件结案。

closed 2024-07-04 by MartyG-RealSense

MartyG-RealSense, 2024-12-04:
My Intel RealSense colleagues have provided some further advice about the errno=42 error.

Given that the problem seems to be related to certain charcters being used in a Windows user account name, I asked whether creating a secondary Windows user account with a name that has no special characters might help.  This was their response. 

You can try to run it as Administrator instead of switching the account.  On Windows 11, it also has a feature to run the app as a different user.  That option can be enabled with the instructions at the link below:
 
https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/

*******************

我的英特尔 RealSense 同事提供了一些有关 errno=42 错误的进一步建议。

鉴于该问题似乎与 Windows 用户帐户名称中使用的某些字符有关，我询问创建一个名称中没有特殊字符的辅助 Windows 用户帐户是否有帮助。这是他们的回复。

您可以尝试以管理员身份运行它，而不是切换帐户。在 Windows 11 上，它还具有以其他用户身份运行应用程序的功能。可以按照以下链接中的说明启用该选项：

https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/