realsenseai/librealsense #13038: Intel reaslsense FPGA support

Nimaro76, 2024-06-14:
Hello, 
I hope you are well. I have a question regarding the use of intel realsense with FPGA. is it possible to bypass the use of intel realsense SDK by parsing the bytes coming from the USB3 connection? does intel have documents that have specified the parsing procedure and initialization bytes per camera ? 

Best regards

MartyG-RealSense, 2024-06-14:
Hi @Nimaro76  In past cases involving using RealSense cameras with FPGA devices, those devices have had the ability to run a Linux OS such as petalinux, as described at https://github.com/IntelRealSense/librealsense/issues/12136

Does the FPGA that you are using have the ability to act like a computing device and run a Linux OS, please?

If you want to directly access the data generated by the RealSense camera hardware at the sensor level then the **Low Level Device API** provided by Intel may be worth investigating.

https://dev.intelrealsense.com/docs/api-architecture#low-level-device-api

If you need to completely avoid use of the SDK then I recall a case from 2017 where somebody aimed to connect a RealSense camera (the SR300 model) to an Arduino board via a serial port connection.

https://forum.arduino.cc/t/intel-real-sense-camera-arduino/377517

The link below has further discussion about the possibilities of using a serial cable connection.

https://community.intel.com/t5/Items-with-no-label/Can-I-use-lib-realsense-on-the-NIOS-II-embedded-processor/m-p/702693

I cannot recall a past example of accessing RealSense data as bytes parsed from a USB3 connection without using the SDK though.

Nimaro76, 2024-06-14:
Indeed the board can run petalinux. But the problem I have is also targeted to compute power as I have seen severe drop in performance when extracting pointcloud from arm vs intel cpu. So I was hoping to solve this by utilizing FPGA fabric

MartyG-RealSense, 2024-06-14:
If you could use an Nvidia Jetson board (which is Arm based) then the RealSense SDK can substantially accelerate point clouds by using its CUDA support to offload processing from the CPU onto the Jetson's graphics GPU.

If you are able to write an application in C++ and your Arm device has a GPU then you can alternatively do a CPU to GPU work offload with the SDK's *GLSL Processing Blocks** feature.  GLSL is 'vendor neutral' and so can work with any GPU brand, though there may not be a noticeable improvement on low-power computing devices.

The example C++ program 'rs-gl' demonstrates processing a point cloud with GLSL.

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

dmipx, 2024-06-18:
If you are planning to use FPGA for data capture it is better to use GMSL camera like D457 and have MIPI data stream as input to FPGA processing block.

Nimaro76, 2024-06-18:
Thank you for your reply . From my understanding this method requires a deserializer of the GMSL ? So another board connected to FPGA?

dmipx, 2024-06-18:
Yes, you will need max9296 deserializer board.

closed 2024-07-04 by MartyG-RealSense