realsenseai/librealsense #13039: How can I use pyrealsense2 with python 12?

mrortach, 2024-06-15:
How can I use pyrealsense2 with python 12?

MartyG-RealSense, 2024-06-15:
Hi @mrortach  At the time of writing this the **pip install pyrealsense2** method does not yet support Python 3.12 (though the support is planned to be implemented), so the pyrealsense2 wrapper must be compiled from source code instead.  This can be done with CMake at the same time as building librealsense, or after librealsense has been installed. 

If you need pyrealsense2 as a pip package installation for Python 3.12 then a method of creating a custom pip package yourself is suggested at https://github.com/IntelRealSense/librealsense/issues/12601#issuecomment-1905554344

mrortach, 2024-06-15:
I still don't know enough to do this.

MartyG-RealSense, 2024-06-15:
If you are not experienced with building with CMake then the simplest approach available may be to use an edited version of the **libuvc backend** build script.

The installation instructions for the libuvc backend procedure can be found here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The procedure builds librealsense using a pre-made script called **libuvc_installation.sh**.  To build librealsense and the pyrealsense2 wrapper, you could make an edit to the CMake build instruction on line 46 of the script.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

Change line 46 from this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

To this:

`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.12`

mrortach, 2024-06-15:
Only I use windows. Actually, I learned a lot while compiling it, but I couldn't finish it.

MartyG-RealSense, 2024-06-15:
You could build the librealsense SDK and pyrealsense2 wrapper together from source code with CMake instead of using the **Intel.RealSense.SDK-WIN10** installer file by following the instructions at the link below.  That would enable you to include the flags BUILD_PYTHON_BINDINGS:bool=true and PYTHON_EXECUTABLE=/usr/bin/python3.12

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

Alternatively, you can build the pyrealsense2 wrapper from source code separately after librealsense has been installed by the installer file.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#windows

mrortach, 2024-06-15:
I keep getting errors.

[image]

MartyG-RealSense, 2024-06-15:
I am not familiar with the **LNK4098** error, though a RealSense user at the link below who also experienced it was offered advice about resolving it.

https://stackoverflow.com/questions/72741520/how-can-i-solve-error-lnk2019-with-glfw-in-librealsense-example-code-project-ca

MartyG-RealSense, 2024-06-21:
Hi @mrortach  Do you require further assistance with this case, please?  Thanks!

mrortach, 2024-06-21:
https://github.com/IntelRealSense/librealsense/issues/13040 Actually, the problem was partially solved, but I still had a problem because pyrealsense2 could not be installed according to version 2.54.2.5684. While trying to solve it, more problems arose.