realsenseai/librealsense #13040: Building pyrealsense2 on Windows: Cannot open input file 'realsense2-gl.lib'

mrortach, 2024-06-15:
Body:
I'm trying to build pyrealsense2 from source on Windows using Visual Studio 2022. I followed the instructions provided in the Intel RealSense GitHub repository.

Here are the steps I took:

1. Cloned the repository:

`git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
git checkout v2.54.2`

2. Created and navigated to the build directory:

`mkdir build
cd build`

3. Ran CMake with the following command:

`cmake .. -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE="C:/***/AppData/Local/Programs/Python/Python312/python.exe"
`

4. Opened the generated .sln file in Visual Studio 2019 and built the solution using Build > Build Solution.

However, I'm encountering the following errors:

`LNK1181: cannot open input file 'realsense2-gl.lib'
LNK4199: /DELAYLOAD:realsense2d.dll ignored; no imports found from realsense2d.dll
LNK4199: /DELAYLOAD:realsense2-gl.dll ignored; no imports found from realsense2-gl.dll
LNK4199: /DELAYLOAD:realsense2.dll ignored; no imports found from realsense2.dll
`

[image]

Here is a screenshot of the error list in Visual Studio:

I have confirmed that realsense2.dll and realsense2-gl.dll are present in the C:\Windows\System32\librealsense\build\Release directory.

Additional Configuration:
In Configuration Properties > Linker > General > Additional Library Directories, I have added:

`C:\Windows\System32\librealsense\build
C:\Windows\System32\librealsense\build\Release`

In Configuration Properties > Linker > Input > Additional Dependencies, I have added:

`realsense2.lib
realsense2-gl.lib
`

Despite this, the build fails with the errors mentioned above. How can I resolve this issue and successfully build pyrealsense2?

System Information:
OS: Windows 11
Visual Studio 2022
Python 3.12
Any help or suggestions would be greatly appreciated. Thank you!

MartyG-RealSense, 2024-06-15:
Hi @mrortach  As you are using Visual Studio, I wonder whether you need to set up linkages to realsense2.dll.  If the full Rhe RealSense SDK has been installed on Windows using the **Intel.RealSense.SDK-WIN10** installer file then the SDK provides three **.props** property sheet files for doing this.  They can be found at `C: > Program Files (x86) > Intel RealSense SDK 2.0`

[image]

Once the SDK is installed and you can access the three props files in the SDK folder, the link below has a C++ Visual Studio tutorial shared by a RealSense user for creating a starter project using the props files.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

mrortach, 2024-06-20:
I tried to compile pyrealsense2 version 2.54.2.5684 for Python 3.12 using the method you suggested, but it didn't work.

MartyG-RealSense, 2024-06-21:
Another approach to installation may be to install the 2.54.2 SDK without pyrealsense2 by using the **Intel.RealSense.SDK-WIN10** installer file and de-selecting the Python wrapper component on the installer's configurartion options so that the Python 3.7 version of the wrapper is not installed.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

Then once the 2.54.2 SDK is installed, build the Python wrapper separately from source code for Python 3.12.

mrortach, 2024-06-21:
I partially did it yesterday, but today I couldn't get it done.

MartyG-RealSense, 2024-06-21:
What errors do you get when building the Python wrapper on Windows separately after the SDK is installed, please?

MartyG-RealSense, 2024-06-28:
Hi @mrortach  Do you require further assistance with this case, please?  Thanks!

mrortach, 2024-06-28:
I still need help, I haven't tried the last shared things. I'll look as soon as I have time.

MartyG-RealSense, 2024-06-28:
Okay, that's no problem at all.  Thanks very much for the update.  I look forward to your next report.  Good luck!

closed 2024-07-24 by MartyG-RealSense