realsenseai/librealsense #13043: FetchContent compatibility?

panpawel88, 2024-06-17:
Hello,

I wanted to add Realsense library as a dependency to my project. Realsense library is added by `FetchContent` cmake feature. However cmake generation fails on my side.

Here is my simple `CMakeLists.txt` file:

```
cmake_minimum_required(VERSION 3.28)
project(realsense_test)

include(FetchContent)

FetchContent_Declare(
        librealsense
        GIT_REPOSITORY "https://github.com/IntelRealSense/librealsense"
        GIT_TAG "8dab5816bef30d56bc185c21132892fe41fbc146"
)

FetchContent_MakeAvailable(librealsense)

set(CMAKE_CXX_STANDARD 17)

add_executable(realsense_test main.cpp)
target_link_libraries(realsense_test librealsense)
```

Error log:
```
// ...
CMake Error at cmake-build-debug/_deps/librealsense-src/tools/CMakeLists.txt:28 (include):
  include could not find requested file:

    /home/pawel/CLionProjects/realsense-test/CMake/opengl_config.cmake
// ...
```

Correct path for `opengl_config.cmake` is `/home/pawel/CLionProjects/realsense-test/cmake-build-debug/_deps/librealsense-src/CMake/opengl_config.cmake`.

Is Realsense library compatible with `FetchContent`? In https://github.com/IntelRealSense/librealsense/issues/12150 somebody wanted to clone with `FetchContent`, so I would expect that it should be working.

MartyG-RealSense, 2024-06-25:
Hi @panpawel88  Do you require further assistance with this case, please?  Thanks!

panpawel88, 2024-06-25:
Hello @MartyG-RealSense , 

I couldn't get it working by FetchContent, but I've added realsense dependency by using vcpkg. My issue is solved.

Thanks

closed 2024-06-25 by MartyG-RealSense

mprat, 2024-08-30:
I actually think the issue is because the opengl configuration in librealsense is using `CMAKE_SOURCE_DIR` rather than `PROJECT_SOURCE_DIR`; I've submitted PR #13311 to fix this issue, since I was running into the same problem.

MartyG-RealSense, 2024-08-30:
Thanks very much @mprat for contributing a PR!