realsenseai/librealsense #13044: Bricked L515 after 2 hours of usage

Sr4l, 2024-06-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | can't open camera | 
| Operating System & Version |  Linux (Ubuntu 20.04, 22.04), Windows 10, Windows 11 | 
| Kernel Version (Linux Only)    | multiple                                         | 
| Platform                                 | PCs  |
| SDK Version                            |  2.50                        | 
| Language                            |  ROS, Realsense.FW.Update, Realsense.Viewer                         | 
| Segment			|  others                          | 

### Issue Description
After using the camera successfully for about 2 hours with ROS1 and RViz on Ubuntu 20.04, it stopped working. Since then we have tested it with another Ubuntu 20.04 ROS1 PC, Windows 11, Windows 10 and finally with Ubuntu 22.04 and ROS2. Nothing worked anymore. What does work though is the camera image with VLC as a Video4Linux device.

With Ubuntu 20.04 and ROS1 noetic we used the packages provided by ROS. With Windows we used the precompiled executables for SDK 2.50.

On Windows with RealSenseView no camera can be found, Realsense.FW.Update finds the camera but crashes:
```ps
PS C:\Users\sr4l\Downloads\Realsense L515>
PS C:\Users\sr4l\Downloads\Realsense L515> & '.\Intel.RealSense.FW.Update.exe' --version

C:\Users\sr4l\Downloads\Realsense L515\Intel.RealSense.FW.Update.exe  version: 2.50.0

PS C:\Users\sr4l\Downloads\Realsense L515> & '.\Intel.RealSense.FW.Update.exe'

Nothing to do, run again with -h for help

Connected devices:
 17/06 10:19:18,648 ERROR [17616] (l500-device.cpp:136) Failed to set units age
RealSense error calling rs2_create_device(info_list:000001DBED8250E0, index:0):
    hr returned: HResult 0x8007001f: "Ein an das System angeschlossenes Ger├ñt funktioniert nicht."
PS C:\Users\sr4l\Downloads\Realsense L515>
```

On Linux Ubuntu 20.04 dmesg output after connecting the device:
```bash
mrt@mrt-pc131:~$ dmesg 
[  273.750390] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[  273.771280] usb 2-1: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.58
[  273.771300] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  273.771309] usb 2-1: Product: Intel(R) RealSense(TM) 515
[  273.771316] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Camera
[  273.771322] usb 2-1: SerialNumber: FFFFFFFFFFFFFFFF
[  273.776334] usb 2-1: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[  273.777476] uvcvideo 2-1:1.1: Failed to set UVC probe control : -32 (exp. 48).
[  273.779192] uvcvideo 2-1:1.2: Failed to set UVC probe control : -32 (exp. 48).
[  273.780482] uvcvideo 2-1:1.3: Failed to set UVC probe control : -32 (exp. 48).
[  273.780822] input: Intel(R) RealSense(TM) 515: Int as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input34
[  273.781418] usb 2-1: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[  273.784319] hid-generic 0003:8086:0B64.0002: hiddev2,hidraw1: USB HID v1.10 Device [Intel(R) RealSense(TM) Camera Intel(R) RealSense(TM) 515] on usb-0000:00:14.0-1/input7
[  273.986117] hid-sensor-hub 0003:8086:0B64.0002: No report with id 0xffffffff found
```

On Linux Ubuntu 20.04 ROS1 noetic output after roslaunch, see attached file [roslaunch_output.txt](https://github.com/user-attachments/files/15864103/roslaunch_output.txt).

Summery: We bought the camera, tested it, it worked, it stopped working after 2 hours of use on about 3 different days. We DIDN'T try a firmware update or anything like that, we just use the ROS1 Noetic supplied packages for Intel Realsense. We also used 2 different USB-C cables that work with an OAK-D pro for example.

I think the SerialNumber in dmesg is strange, but I do not know the output from the working state. Also, we have not done anything that should change the serial number, if that is even possible.

Any and all help is appreciated.

MartyG-RealSense, 2024-06-17:
Hi @Sr4l  There are two previous cases of the ffffffffffffffff serial number on L515 at https://github.com/IntelRealSense/librealsense/issues/7302 and https://github.com/IntelRealSense/librealsense/issues/7311

This is a hardware issue and will not be fixable.  Please do an **RMA return** of your camera to the retailer that you purchased it from.  Guidance about doing so can be found at the link below.

https://support.intelrealsense.com/hc/en-us/articles/360051623953-How-to-return-a-faulty-Intel-RealSense-Camera

Sr4l, 2024-06-21:
I am sad to hear that. Thank you for the info.

closed 2024-06-21 by Sr4l