realsenseai/librealsense #13053: Bad face reconstruction quality with D435 camera

matteocarbonara1234, 2024-06-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

matteocarbonara1234, 2024-06-18:
Hi,

I am working on an algorithm which makes 3D face reconstruction starting from a set of images and in particular I am comparing the results obtained with both SR300 and D435 camera.
As you can notice in the image below (on the left D435 and on the right SR300), the result obtained with the SR300 camera is more accurate and detailed, in particular near the nose. 
I have tried different depth settings, different distances from the camera and different resolutions of the depth map, but the results are always better with the SR300 camera.

[image]

My question is: is it normal in this specific context i.e. face reconstruction to obtain worse results with the D435, even if it is a newer camera? and if not, what parameters should I modify in order to improve the quality of the surface reconstruction? 

Thanks for the help

MartyG-RealSense, 2024-06-19:
Hi @matteocarbonara1234  SR300 and D435 are different camera technologies.  SR300 is Coded Light (similar to a technology called Structured Light) and D435 is Stereo Depth (generating the depth image from a pair of left and right sensors).  In this particular application the Coded Light technology of a SR300 typically produces better solid facial masks of the type shown above.  

May I confirm please that you are not applying a post-processing **Decimation Filter** to the D435 data?  That could result in a blurring of the image as the Decimation filter 'downsamples' the resolution.

matteocarbonara1234, 2024-06-19:
Hi @MartyG-RealSense, thank you for the response.

I confirm that I am not applying a post-processing Decimation Filter to the D435 camera.

MartyG-RealSense, 2024-06-19:
You could next try increasing the Laser Power setting to its maximum value of '360'.  Doing so should increase the amount of detail on the depth image and make the camera's invisible dot-pattern projection more visible to the camera.  The camera uses the projection of the dots onto real-world surfaces to analyze those surfaces for depth information.

matteocarbonara1234, 2024-06-19:
I have just tried to increase the Laser Power setting to its maximum value, but unfortunately there is no improvement in the quality of the face reconstruction.

MartyG-RealSense, 2024-06-19:
Other than increasing the level of light illumination on the face to try to bring out further depth detail from the face (perhaps with an external lamp), there is likely to be no other action that can make the D435 mask more comparable to the SR300 one.

MartyG-RealSense, 2024-06-25:
Hi @matteocarbonara1234  Do you require further assistance with this case, please?  Thanks!

matteocarbonara1234, 2024-06-25:
Hi @MartyG-RealSense, I do not require further assistance. 
Thanks for the help.

MartyG-RealSense, 2024-06-25:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2024-06-25 by MartyG-RealSense