realsenseai/librealsense #13057: Python / RuntimeError: No device connected

frankahland, 2024-06-19:
Hello,

everything worked fine 3 month ago...
Now i am getting this error message even on the easiest way to implement the camera.
The software "RealSense Viewer" can find the camera and shows all streams.

Does anybody have an idea?

|**librealsense**|2.54.2 RELEASE|
Inten RealSense L515
Firmware 1.5.8.1
Windows
Python 3.11.9

Traceback (most recent call last):
  File "C:\Daten\OneDrive\Python\Objekterkennung\515Test.py", line 3, in <module>
    profile = pipe.start()
              ^^^^^^^^^^^^
RuntimeError: No device connected

import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()    
try:
  for i in range(0, 100):
    frames = pipe.wait_for_frames()
    for f in frames:
      print(f.profile)
finally:
    pipe.stop()

Same problem with this coding:
        self.pipeline = rs.pipeline()

        config = rs.config()
    #    config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
    #    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 20)
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 20)

        self.color_frame = True
        self.depth = None

        # Start streaming
        self.pipeline.start(config)

MartyG-RealSense, 2024-06-19:
Hi @frankahland  In your config instructions in the script above, the FPS is set to 20.  It should be 30.  I note that you have the same camera detection problem with the simple test script though.

Is the RealSense Viewer closed down when you launch your script, please?

Have you changed your Python 3 version since the last time that the script worked for you?

Is the L515 detected in Python if you unplug the micro-sized end of the USB cable from the base of the camera, turn the connector around the other way and re-insert it?

frankahland, 2024-06-19:
Thanks for your answer!

Sorry, the 20fps was only one of many tries to get it working. Also 30fps does not work as it was set this morning - the same code that used to work 3 month ago and which i find in any examples in google.

yes, RealSense Viewer is closed.

3 month ago i ran Python 3.11.4 (or so).
One week ago i deinstalled Python and installed 3.7 first and then 3.11 (because of another project).
So this morning i started it with python 3.11.
When today it did not work with the RealSense i upgraded to 3.11.9 (the latest 3.11 version)
And still not working.

The package pyrealsense2 was installed smoothly after the installation.

frankahland, 2024-06-19:
I turned the connector and still the same problem.

MartyG-RealSense, 2024-06-19:
Thank you very much for the information.  

Can you confirm whether you installed from a package with the **pip install pyrealsense2** installation method, please?

Did you uninstall Python 3.7 before installing Python 11 to ensure that you do not have two Python versions on the same computer that might confuse pyrealsense2 about which one one to use?

If you are okay with using Python 3.7, installing the 2.54.2 RealSense SDK on Windows with the **Intel. RealSense.SDK-WIN10** installer file will also install the pyrealsense2 wrapper for use with 3.7.

[image]

frankahland, 2024-06-19:
Thanks a lot Marty,

I installed it in the IDE PyCharm with right mouse click on the missing import. Now deleted it and installed it again with 
pip install pyrealsense2.

It installed this version. Is this correct for Python 3.11.9?
PS C:\Daten\OneDrive\Python\Objekterkennung> pip install pyrealsense2
Collecting pyrealsense2
  Using cached pyrealsense2-2.55.1.6486-cp311-cp311-win_amd64.whl.metadata (1.9 kB)
Using cached pyrealsense2-2.55.1.6486-cp311-cp311-win_amd64.whl (2.8 MB)
Installing collected packages: pyrealsense2
Successfully installed pyrealsense2-2.55.1.6486

Yes, i used uninstaller to remove Python 3.7. But after your idea i searched for files and found pyrealsense2 in other directories. I deleted them.

I need to go for Python 3.11.+, so 3.7 would not be an option.

I restarted Windows, used different USB ports, 

Still the same error.

Maybe of interest: After starting the python script it takes 10 seconds to get the error message "RuntimeError: No device connected"

Do you have another idea?

frankahland, 2024-06-19:
The device manager shows a very old driver...
[image]

frankahland, 2024-06-19:
I uninstalled the SDK and reinstalled it - now the viewer does not find the camera anymore. 
[image]

MartyG-RealSense, 2024-06-20:
I note that your RealSense Viewer version is 2.55.1.  Support for the L515 camera model has been removed from 2.55.1 onwards as L515 is a retired camera model.  You can install the 2.54.2 version of the SDK and Viewer that still supports L515 using the **Intel.RealSense.SDK-WIN10** installer file at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

[image]

If you only want the RealSense Viewer, download and run **Intel.RealSense.Viewer.exe** from the Assets list.  This Viewer must be run from the location that you downloaded the .exe file to and not with a Windows desktop launch shortcut icon.

frankahland, 2024-06-20:
Hello Marty,
thanks a lot. That version verion hepled. The viewer is viewing again.
And it was the same reason for python not to run anymore. The new installation of python made me install the latest pyrealsense2 which also does not support the L515 anymore.

pip install pyrealsense2==  
showed the available packages:
versions: 2.54.1.5216, 2.54.1.5217, 2.54.2.5684, 2.55.1.6486)

So i installed the version before the newest and it works!
 pip install pyrealsense2==2.54.2.5684

frankahland, 2024-06-20:
It would be great to raise an error message if a once supported device is found but not supported anymore.
I searched for hours.

closed 2024-06-20 by frankahland

MartyG-RealSense, 2024-06-20:
You are very welcome.  I'm pleased to hear that 2.54.2 is working for you.  Thanks very much for the update!

The Intel RealSense team has already received a request to display a message about L515 not being supported so that RealSense users do not lose time.  They are working on it.