realsenseai/librealsense #13058: No device connected(imshow example in wrappers)

Rhymakers, 2024-06-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.5.8.1) | 
| Operating System & Version |   {ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  (6.5.0-41-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {2.54.2}                          | 
| Language                            |  {C++, opnecv }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
When I create a build directory to run the imshow example, run cmake .., make, and then run the ./imshow command, the following error appears.

[image]

1. It is captured well in realsense-viewer, but what is the reason?

2. Are there any examples that can be used when using multiple l515s?

3. At first, I couldn't get realsense, so I modified the cmakelist as shown below. Is this correct?
[image]

MartyG-RealSense, 2024-06-20:
Hi @Rhymakers  The RealSense OpenCV wrapper's example programs are built by building the RealSense SDK from source code with CMake and including the flag **-DBUILD_CV_EXAMPLES=TRUE** in the CMake build instruction.

Are you uaing the imshow code in a new project that you have created, please?  If you are then doing so may result in necessary linkages to the librealsense library being absent from the CMakeLists.txt file.

A standalone C++ **Getting Started** project at the link below uses the same script as the rs-hello-realsense C++ example program but has a more complex CMakeLists.txt file than rs-hello-realsense.  Reviewing the additional linkages in the Getting Started project's CMakeLists.txt may be helpful for you if you are attempting to get imshow's code working in a new project.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

https://github.com/zivsha/librealsense/blob/getting_started_example/examples/getting-started/CMakeLists.txt

In regard to a multiple camera project, the RealSense SDK's **rs-multicam** C++ example should meet your needs.  If you have built the SDK with example programs included then there should be a pre-built executable version of rs-multicam in your **usr/local/bin** Ubuntu folder.

https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

Rhymakers, 2024-06-21:
Thank you for your response.
When I run rs-multicam in usr/local/bin, it works, or before it works, my laptop fails. I'm connecting several things, and I'm using a powered usb hub. This is even true for realsens-veiwer. The computer doesn't work when I do the same version in Windows, but in Ubuntu my computer stops while running a viewr or multicam example. Why is that?

MartyG-RealSense, 2024-06-21:
Does it make a difference if you launch Ubuntu programs in sudo admin permissions mode, such as **sudo realsense-viewer** and **sudo rs-multicam**

Rhymakers, 2024-06-21:
Unfortunately, it's the same. my laptop is Lenovo Legion Slim 7i 16 Gen 8 laptop.

MartyG-RealSense, 2024-06-21:
What other devices are connected to your laptop, please?  Are any of them wireless USB devices such as a wireless mouse?  If you do have wireless devices, does the computer behave better if they are unplugged?

Rhymakers, 2024-06-24:
Unfortunately, it's the same. but It works well realsense-viewer in Windows.

MartyG-RealSense, 2024-06-24:
Which OpenCV version are you using, please?  The RealSense OpenCV wrapper's example programs such as imshow were designed for use with OpenCV 3.4.

closed 2024-06-26 by Rhymakers

MartyG-RealSense, 2024-06-26:
You are very welcome.  Thanks very much for the update.  :)