realsenseai/librealsense #13061: sudo apt-get update failed due to 403 forbidden

IRONCROWY, 2024-06-20:
I want to install Intel® RealSense™ SDK 2.0 (v2.50.0) for my T265, so i followed the linux-installation from the link https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

But i sadly found it quite hard to install because of the error.

```
(base) gds@gds-System-Product-Name:~$ sudo mkdir -p /etc/apt/keyrings
curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | sudo tee /etc/apt/keyrings/librealsense.pgp > /dev/null
(base) gds@gds-System-Product-Name:~$ sudo apt-get install apt-transport-https
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
apt-transport-https is already the newest version (2.7.14build2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
(base) gds@gds-System-Product-Name:~$ echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | \
sudo tee /etc/apt/sources.list.d/librealsense.list
sudo apt-get update
deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo noble main
Hit:1 https://dl.google.com/linux/chrome/deb stable InRelease
Get:2 https://apt.atzlinux.com/atzlinux bookworm InRelease [55.5 kB]                                                                                                                                  
Hit:3 https://packages.microsoft.com/repos/code stable InRelease                                                                                                                                                
Hit:4 http://archive.ubuntu.com/ubuntu noble InRelease                                                                 
Hit:5 http://security.ubuntu.com/ubuntu noble-security InRelease    
Hit:6 http://archive.ubuntu.com/ubuntu noble-updates InRelease      
Err:7 https://librealsense.intel.com/Debian/apt-repo noble InRelease
  403  Forbidden [IP: 13.227.62.104 443]
Hit:8 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Reading package lists... Done
E: Failed to fetch https://librealsense.intel.com/Debian/apt-repo/dists/noble/InRelease  403  Forbidden [IP: 13.227.62.104 443]
E: The repository 'https://librealsense.intel.com/Debian/apt-repo noble InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
(base) gds@gds-System-Product-Name:~$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package librealsense2-dkms
(base) gds@gds-System-Product-Name:~$ 
```

Because of the 403 forbidden, the libs cannot be installed.

Before these operation, I also tryed the command:
```
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
```
And it failed because of failing to connect to the server and i found some issues metioning that it has been removed?

I'm quite puzzled when installing this software...

By the way, my system is Ubuntu 24 LTS

IRONCROWY, 2024-06-20:
I again try the command to add key, the logs are below:
```
(base) gds@gds-System-Product-Name:~$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key  F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key  F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.C3lLEqGPZ7/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: keyserver receive failed: Connection timed out
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.UzAaXnZ7wa/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: keyserver receive failed: Connection timed out

```

IRONCROWY, 2024-06-21:
Strangely is the fact that some hours later, I found the key worked, but the responsitory is still 403 forbidden.

```
(base) gds@gds-System-Product-Name:~$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key  F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key  F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.SjRVU9HHvk/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
(base) gds@gds-System-Product-Name:~$ sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
Repository: 'deb https://librealsense.intel.com/Debian/apt-repo noble main'
Description:
Archive for codename: noble components: main
More info: https://librealsense.intel.com/Debian/apt-repo
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Found existing deb entry in /etc/apt/sources.list.d/librealsense.list
Updating existing entry instead of using /etc/apt/sources.list.d/archive_uri-https_librealsense_intel_com_debian_apt-repo-noble.list
Found existing deb-src entry in /etc/apt/sources.list.d/librealsense.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/librealsense.list
Hit:1 https://apt.atzlinux.com/atzlinux bookworm InRelease
Hit:2 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                
Hit:3 https://packages.microsoft.com/repos/code stable InRelease                                                                                             
Hit:4 http://security.ubuntu.com/ubuntu noble-security InRelease                                                                                             
Err:5 https://librealsense.intel.com/Debian/apt-repo noble InRelease                            
  403  Forbidden [IP: 3.165.102.52 443]
Hit:6 http://archive.ubuntu.com/ubuntu noble InRelease
Hit:7 http://archive.ubuntu.com/ubuntu noble-updates InRelease
Hit:8 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Reading package lists... Done
E: Failed to fetch https://librealsense.intel.com/Debian/apt-repo/dists/noble/InRelease  403  Forbidden [IP: 3.165.102.52 443]
E: The repository 'https://librealsense.intel.com/Debian/apt-repo noble InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
(base) gds@gds-System-Product-Name:~$ echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | \
sudo tee /etc/apt/sources.list.d/librealsense.list
sudo apt-get update
deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo noble main
Hit:1 https://apt.atzlinux.com/atzlinux bookworm InRelease
Hit:2 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                
Hit:3 http://security.ubuntu.com/ubuntu noble-security InRelease                                                                                             
Hit:4 http://archive.ubuntu.com/ubuntu noble InRelease                                    
Hit:5 https://packages.microsoft.com/repos/code stable InRelease    
Hit:6 http://archive.ubuntu.com/ubuntu noble-updates InRelease      
Hit:7 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Err:8 https://librealsense.intel.com/Debian/apt-repo noble InRelease
  403  Forbidden [IP: 3.165.102.52 443]
Reading package lists... Done
E: Failed to fetch https://librealsense.intel.com/Debian/apt-repo/dists/noble/InRelease  403  Forbidden [IP: 3.165.102.52 443]
E: The repository 'https://librealsense.intel.com/Debian/apt-repo noble InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
(base) gds@gds-System-Product-Name:~$ 
```

IRONCROWY, 2024-06-26:
Problem not solved, but realsense-viewer is installed by manual installation on Ubuntu 20.04 LTS with the kernel  of 5.15.0

closed 2024-06-26 by IRONCROWY

jiaming-ai, 2024-08-06:
Any solutions for this?

I got the same problem on ubuntu24.

jiaming-ai, 2024-08-06:
Hi @MartyG-RealSense , thanks for your prompt response. However, I’m in Singapore, not China. The internet is definitely not blocked.

MartyG-RealSense, 2024-08-06:
@jiaming-ai  Thanks very much for the clarification of your location.  It could be an issue with the Ubuntu 24.04 version that you are using then.  There was a past 403 error case at https://github.com/IntelRealSense/librealsense/issues/10792#issuecomment-1220320263 that was experienced by someone who was using 22.04 before official support of it was introduced in librealsense.  

Likewise, official support for 24.04 Noble Numbat is not in the current latest 2.55.1 SDK but will be introduced in the next SDK version planned for release around the end of September 2024.  More information about this can be found at https://github.com/IntelRealSense/librealsense/discussions/13072