realsenseai/librealsense #13065: Regarding the issue of the inability to install realsense in Ubuntu 24.04 version

987640673, 2024-06-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 24) | 
| Kernel Version (Linux Only)    | 6.8.0-1005-raspi                                    | 
| Platform                                 | Raspberry P |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | python                          | 
| Segment			|  Robot                          | 

### Issue Description
<My Raspberry Pi has the detailed configuration shown above. When installing realsense, I used the "./scripts/patch realsense ubuntu lts hwe. sh" command to compile, but the terminal prompted that my system's distribution version is not supported. I need to install realsense to use the D435I camera. What should I do?>

MartyG-RealSense, 2024-06-21:
Hello, the best way to install the librealsense SDK on Ubuntu on Raspberry Pi is to use the **libuvc backend** build script at the link below.  This installation method bypasses the kernel and so a patch script does not need to be applied.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

987640673, 2024-06-25:
I try to use the **libuvc backend** build script at the link below,but the terminal displays the following prompt:

+ exec
+ exec
+ '[' 19 -ne 0 ']'
+ echo -e '\e[32m'

+ read -p 'Remove all RealSense cameras attached. Hit any key when ready'
Remove all RealSense cameras attached. Hit any key when ready
+ echo -e '\e[0m'

+ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble
++ uname -r
+ echo 'Kernel version 6.8.0-1005-raspi'
Kernel version 6.8.0-1005-raspi
+ sudo apt-get update
Warning: The unit file, source configuration file or drop-ins of apt-news.service changed on disk. Run 'systemctl daemon-reload' to reload units.
Warning: The unit file, source configuration file or drop-ins of esm-cache.service changed on disk. Run 'systemctl daemon-reload' to reload units.
命中:1 https://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports noble InRelease
命中:2 https://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports noble-updates InRelease
命中:3 https://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports noble-backports InRelease
命中:4 http://packages.ros.org/ros2/ubuntu noble InRelease                     
命中:5 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease           
忽略:6 https://ppa.launchpadcontent.net/canonical-chromium-builds/stage/ubuntu noble InRelease
错误:7 https://librealsense.intel.com/Debian/apt-repo noble InRelease
  403  Forbidden [IP: 108.138.246.11 443]
错误:8 https://ppa.launchpadcontent.net/canonical-chromium-builds/stage/ubuntu noble Release
  404  Not Found [IP: 185.125.190.80 443]
正在读取软件包列表... 完成
E: 无法下载 https://librealsense.intel.com/Debian/apt-repo/dists/noble/InRelease  403  Forbidden [IP: 108.138.246.11 443]
E: 仓库 “https://librealsense.intel.com/Debian/apt-repo noble InRelease” 没有数字签名。
N: 无法安全地用该源进行更新，所以默认禁用该源。
N: 参见 apt-secure(8) 手册以了解仓库创建和用户配置方面的细节。
E: 仓库 “https://ppa.launchpadcontent.net/canonical-chromium-builds/stage/ubuntu noble Release” 没有 Release 文件。
N: 无法安全地用该源进行更新，所以默认禁用该源。
N: 参见 apt-secure(8) 手册以了解仓库创建和用户配置方面的细节。

how should I do?

MartyG-RealSense, 2024-06-25:
The Ubuntu version should not be a problem with libuvc, since it not only bypasses the kernel but is not dependent on particular Linux versions.  There have been few people using librealsense with Ubuntu 24.04 at the time of writing this though, so there may be unpredictable compatibility issues.  Official support for 24.04 is being added in the next version of librealsense after the current 2.55.1 one.

Another, more complicated method of building the SDK with a kernel bypass like libuvc's is **RSUSB backend**.  Instructions for an RSUSB build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

MartyG-RealSense, 2024-07-04:
Do you require further assistance with this case, please?  Thanks!

Schwarzkopf-Henkal, 2024-07-09:
encountered same issue, also got 403 ERROR when accessing apt repository when trying to install via libuvc backend

MartyG-RealSense, 2024-07-09:
Hi @Schwarzkopf-Henkal  Were you also using Ubuntu 24.04 with the libuvc backend script please?

Schwarzkopf-Henkal, 2024-07-09:
Yes, Ubuntu24.04 on Raspberry Pi5. Got almost identical error logs with above mentioned one when using libuvc backend script. But successfully installed and launched realsense-viewer via RSUSB backend.

MartyG-RealSense, 2024-07-09:
Thanks so much @Schwarzkopf-Henkal for confirmation that using RSUSB backend (the successor to libuvc backend) worked for you on your Pi.

Would you be able to share with @987640673 the details of the installation procedure that you used,please?  Thanks again!

Schwarzkopf-Henkal, 2024-07-10:
I followed exactly the same procedure described in #9931 and I was good to go

Schwarzkopf-Henkal, 2024-07-10:
My hypothesis is that libuvc backend is technically ok with ubuntu24.04, the problem is the apt repository or network issue. I got 403 error too when trying to install via apt as described in doc/distribution_linux.md. The installation was already erring when executing `sudo apt update`, got 403 error connecting to intel realsense apt repository. Then I got error locating package `librealsense2-dkms`. Also, the Great Firewall of Mainland China could cause connection issue and that could also be the case.

987640673, 2024-07-11:
I used the method @Schwarzkopf-Henkal  provided to build the software and it was successful. Thank you very much!

987640673, 2024-07-11:
But when I was building using libuvc, I had already used VPN, but still received a 403 prompt, which is clearly not a problem with the Chinese firewall.

MartyG-RealSense, 2024-07-11:
Thanks very much @987640673 and @Schwarzkopf-Henkal for the updates.  I'm pleased to hear that you were successful!