realsenseai/librealsense #13071: error in Realsence Viewer

nohaalaref, 2024-06-23:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-06-23:
Hi @nohaalaref  Does the error appear when the RealSense Viewer first opens, or does it appear when an option is selected?

This error sometimes occurs when trying to access the RGB camera from the **Add Source** option, like in https://github.com/IntelRealSense/librealsense/issues/11338

nohaalaref, 2024-06-23:
yes exactly this is when I was trying to open the RGB camera (platform camera) from Add Source. However, my main issue is that the Realsense viewer is not reading my camera lidar L515 at all. There is an access to the laptop , tried changing the cable and reinstalled the SDK and viewer more than one time.

MartyG-RealSense, 2024-06-23:
Support for L515 has been removed from the RealSense SDK from version 2.55.1 onwards as it is a retired camera model.  If you install the **2.54.2** version then the L515 should be detected.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.2

nohaalaref, 2024-06-23:
I tried the new version but still, it couldn't read it

MartyG-RealSense, 2024-06-23:
Platform Camera is usually used for accessing a non-RealSense RGB camera such as the webcam on a laptop.  Not all webcam laptops will work with the Platform Camera feature though, unfortunately.  Some laptops work with it and others do not.

closed 2024-07-04 by MartyG-RealSense