realsenseai/librealsense #13085: Steps between depth

oscarthorn, 2024-06-25:
Hi!
For the D455 in 640x480 resolution mode how can I calculate the depth resolution / step length? There are only certain values that are possible due to the pixelsize / disparity values, but I'm not sure how to calculate it for the d455. Say 6 meters exactly was on value what is the next possible value?

MartyG-RealSense, 2024-06-25:
Hi @oscarthorn  It sounds as though you are describing the depth **step size**, detailed at the link below.  Is that correct, please?

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#change-the-depth-step-size

The default step size is 1000um (1mm).  In most cases it is not necessary to know this value or to change it, except when operating the camera at very close range to the observed target object / surface or wanting to extend the range of the camera beyond its default maximum.

Whilst maximum range can be theoretically increased, in practice  there is a physical limit to how far the camera's sensor components can see.

The step size value does not change once set and is not affected by how distant something is from the camera.

The **intrinsics** of the camera (such as pixel height 'fx' and pixel width 'fy') will be affected by a change in resolution, and you can retrieve these using program code or with the **rs-enumerate-devices** tool, as described at https://github.com/IntelRealSense/librealsense/issues/10663#issuecomment-1178804695

The step size can be configured in the **Stereo Module > Advanced Controls > Depth Table** section of the RealSense Viewer tool's options side-panel, though there is usually no need to change its default value.

[image]

It is better to configure the other 'Depth Units' option in **Stereo Module > Controls** instead if you need to change the depth scale.   Like the Advanced Controls version of this setting, there is usually no need to change this value from its default, which is 0.001 (m) instead of 1000 (um).

[image]

If I have misunderstood the original question or you require further information, please do let me know.  Thanks!

oscarthorn, 2024-07-02:
Hi @MartyG-RealSense! No I'm not talking about the depth units, though maybe it is related, sorry for the confusion. I'm talking about the distances that are possible to calculate based on the disparity. Since the depth is inversely proportional to the disparity a disparity of 1 pixel means something is far away. The jump is distance from disparity 1 to disparity 2 is significantly bigger than from say 100 to 101. But I'm not sure how to calculate the "jump" taking into account that the realsense also uses subpixel calculations.

MartyG-RealSense, 2024-07-02:
Maximum depth distance is limited by the physical capabilities of the camera hardware.  So if a particular RealSense camera model's maximum observable depth distance is rated at 10 meters then it will not be able to be configured to render depth information that is further away than that maximum.

Intel's **subpixel linearity** white-paper guide states that " the depth resolution will be determined by the smallest step size of the Disparity, DisparityStep.  We also call this the Subpixel".

The guide describes how to measure subpixel linearity.

https://dev.intelrealsense.com/docs/white-paper-subpixel-linearity-improvement-for-intel-realsense-depth-cameras#measurement-and-tuning-of-subpixel-linearity

closed 2024-07-03 by oscarthorn