realsenseai/librealsense #13093: rs_error was raised when calling rs2_get_option "failed to set power state"

garretthilyer, 2024-06-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer -->2.55) | 
| Operating System & Version |  Ubuntu 24.04, Arm64 | 
| Kernel Version (Linux Only)    |  6.8.0-1005-raspi                                    | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C}                         | 
| Segment			|  {others}                          | 

### Issue Description
I used this [code](https://github.com/IntelRealSense/librealsense/issues/11506#issuecomment-1599168850) to build realsense.
My code was working completely fine, I could capture frames for long periods of time. 
I started messing around with sensors and trying to adjust exposure, contrast, white balance, etc. 

Everytime I would use either: 
`rs2_get_option(sensor, RS2_OPTION_CONTRAST, &e)` or `rs2_set_option(sensor, RS2_OPTION_CONTRAST, 50.0, &e)`

I would get this error:
`rs_error was raised when calling rs2_get_option(options:0xaaaada8ea640, option_id:2):
    failed to set power state`

I tried all the different types of options and the option_id tag would update depending on the parameter I wanted to change. Assume I am setting up the ctx, config, pipeline, device correctly...

This is the code I added for the sensor setting:
```
// Request the entire list of sensors
    rs2_sensor_list* sensor_list = rs2_query_sensors(dev, &e);
    check_error(e);

    // Request access to the total number of sensors
    int num_sensor = rs2_get_sensors_count(sensor_list, &e);
    check_error(e);

    for (int i = 0; i < num_sensor; i++) {
        // iterate through all sensors
        rs2_sensor* sensor = rs2_create_sensor(sensor_list, i, &e);
        check_error(e);
        // until you find color sensor
        if (rs2_is_sensor_extendable_to(sensor, RS2_EXTENSION_COLOR_SENSOR, &e) && !e) {
            
            // get color params
            printf("exposure default: %f \n", rs2_get_option(sensor, RS2_OPTION_CONTRAST, &e));
            check_error(e);

            // potential set of other params here as well ....

        }
        // free sensor
        rs2_delete_sensor(sensor);
    }
    // free sensor list
    rs2_delete_sensor_list(sensor_list);
```
I tried increasing the USB power output, sleeping for a couple of seconds before I reach this code, and even capturing a couple of data frames before running this. None of that has worked. 

I am open to all possible solutions. Thanks in advanced!

garretthilyer, 2024-06-26:
### Solution!

I was trying to manipulate the sensors **after** I created the pipeline/enable_stream(). I moved the sensor setting code to directly below the device initialization and it worked completely fine.

MartyG-RealSense, 2024-06-27:
Hi @garretthilyer  It's excellent to hear that you achieved a solution.  Thanks very much for the update!

closed 2024-07-04 by MartyG-RealSense