realsenseai/librealsense #13094: fastrtps is ALREADY outdated and replaced by Fast-DDS ??

jiapei100, 2024-06-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

1) Can we replace all packages under [third-party](https://github.com/IntelRealSense/librealsense/tree/master/third-party) from **local** to **system**?
2) For [third-party realdds](https://github.com/IntelRealSense/librealsense/tree/master/third-party/realdds), [https://github.com/canonical/Fast-RTPS](https://github.com/canonical/Fast-RTPS) is **ALREADY** outdated and replaced by [https://github.com/eProsima/Fast-DDS](https://github.com/eProsima/Fast-DDS) ... Can anybody help to update it?

MartyG-RealSense, 2024-06-27:
Hi @jiapei100  When building the librealsense SDK from source code with CMake, there is a configuration flag called **BUILD_WITH_DDS** that requires CMake 3.16.3 and is set to Off by default.  

https://github.com/IntelRealSense/librealsense/blob/master/CMake/lrs_options.cmake#L49

The main **CMakeLists.txt** file of librealsense has a provision that if the BUILD_WITH_DDS flag is set to On then the CMake file **external_fastdds.cmake** is included in the build, 

https://github.com/IntelRealSense/librealsense/blob/master/CMakeLists.txt#L62-L75

This places the FastDDS project in the **3rd Party/fastdds** folder.

When building with CMake, the configuration flags have -D placed on the front of them, so using **-DBUILD_WITH_DDS=ON** in the CMake build instruction should enable it.  

*********

In regard to your other question, can you provide information about the difference between local and system packages, please?

MartyG-RealSense, 2024-07-04:
Hi @jiapei100  Do you require further assistance with this case, please?  Thanks!

jiapei100, 2024-07-05:
@MartyG-RealSense 
Thank you Marty. I'm closing it. I'll build **librealsense** without `Fast-DDS`. Thank you ...

closed 2024-07-05 by jiapei100

MartyG-RealSense, 2024-07-05:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!