realsenseai/librealsense #13100: Getting mutiple issues in Intel Realsense Camera.

ShreyasKuntnal, 2024-07-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.15.0-113-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            | LibRealSense v2.50.0                     | 

I tried mutiple issue threads but fed up with these mutiple issues.
When I do reaslsense launch, i am getting these issues:
`roslaunch realsense2_camera rs_camera_jetsoncopy.launch global_time_enabled:=true initial_reset:=true`

```
SUMMARY
========

PARAMETERS
 * /camera/l500_depth_sensor/global_time_enabled: True
 * /camera/motion_module/global_time_enabled: True
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
[... 120 lines cut ...]
 01/07 11:46:21,258 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:21,308 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:21,359 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:21,411 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:21,664 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:45,644 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:45,694 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:46,746 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:46,796 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:48,076 ERROR [139920773478144] (ds5-options.cpp:88) Asic Temperature value is not valid!
 01/07 11:46:48,849 WARNING [139921805256448] (messen 01/07 11:50:49,297 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:51:11,369 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
ger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:49,081 ERROR [139920773478144] (ds5-options.cpp:88) Asic Temperature value is not valid!
 01/07 11:46:49,901 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:46:50,087 ERROR [139920773478144] (ds5-options.cpp:88) Asic Temperature value is not valid!
 01/07 11:47:09,968 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:47:26,029 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:48:47,145 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:50:49,297 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/07 11:51:11,369 WARNING [139921805256448] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

Can you please help me with resolving these issues?

ShreyasKuntnal, 2024-07-01:
`realsense-viewer `:

 ```
01/07 13:26:32,536 WARNING [139704330610432] (sensor.cpp:1147) RGB Camera: skipping Backlight Compensation control. descriptor: [min/max/step/default]= [0/0/0/0]
 01/07 13:26:32,544 WARNING [139704330610432] (sensor.cpp:1147) RGB Camera: skipping Exposure control. descriptor: [min/max/step/default]= [0/0/0/0]
 01/07 13:26:32,567 ERROR [139704330610432] (librealsense-exception.h:52) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 01/07 13:26:32,567 WARNING [139704330610432] (sensor.cpp:1173) Failed to add Enable Auto Exposure control for RGB Camera
```
How to fix these error when realsense camera is activated?

MartyG-RealSense, 2024-07-01:
Hi @ShreyasKuntnal  You do not need to set **global_time_enabled** to true as it is already enabled by default on 400 Series cameras such as D435i.

The `control_transfer returned` warning can indicate that there is a communication problem between the camera and the computer / computing board, such as an issue with the USB port or USB cable.  The **Resource temporarily unavailable** type of this warning could indicate that the computer's CPU is struggling to keep up with processing the frames that are arriving at the USB port from the camera.

If you do not need the Infra2 stream then you can disable it to reduce the processing burden, as it is not required for streaming Infra1.

Likewise, if you do not need depth-color aligned images then set **align_depth** to False to further reduce processing, as alignment is calculated on the computer and not the camera hardware.

Please also check whether the depth and color streams are being published in RViz, as sometimes they can be displayed normally in RViz even if control_transfer warnings are generating continuously in the log.

ShreyasKuntnal, 2024-07-01:
Have i installed right version of dkms for particular kernel and librealsense version?

[image]

I get these errors as soon as I start realsense-viewer. What are these, Is there need to fix these or ignoring can be ok?

MartyG-RealSense, 2024-07-01:
The firmware version 5.13.0.50 is correct for librealsense 2.50.0.  As you have 2.50.0, it looks as though you have the correct DKMS package version.

Are you using the official 1 meter USB cable supplied with the camera or a cable of your own choice?

Is the RGB stream of the Viewer able to be enabled despite the RGB-related errors in the Viewer log?

Is there improvement if you reset your camera's calibration to its factory-new default settings in the Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

ShreyasKuntnal, 2024-07-01:
I use the default cable that's provided by intel.
RGB stream is able to be enabled,
With respect to camera calibration, this is how it looks.
[image]

MartyG-RealSense, 2024-07-01:
To unlock the **Write Table** button on the Camera Calibration window, you need to enable the Stereo Module (the depth stream) before opening the window.  You can then click on Write Table to reset the camera's calibration to its defaults.

In general, if streams are behaving normally when enabled despite warning / error messages in the log then the messages can be ignored.

ShreyasKuntnal, 2024-07-01:
If I do camera calibration reset, is there a need to do imu calibration again.
Sorry for very vague question, I feel there might be some link.

MartyG-RealSense, 2024-07-01:
There is not information available on whether the reset affects IMU calibration.

However, if data stored in the camera called a **calibration table** has become corrupt then it can prevent the camera from working correctly.  Clicking the Write Table button overwrites the old calibration table with a new one, so it can enable the camera to work correctly if a corrupted calibration table was the cause of the problems.

ShreyasKuntnal, 2024-07-01:
Did Camera calibration 
[image]
and restarted the realsense-viewer still these errors persist.
[image]

If we can fix the same the lets work on it or else as u said we can ignore lets ignore

MartyG-RealSense, 2024-07-01:
Yes, I recommend ignoring the errors if performance of the streams is not being negatively affected.

ShreyasKuntnal, 2024-07-01:
Oh ok so even this error `(librealsense-exception.h:52) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument` can be ignored then?
Thank you so much for the faster replies.

closed 2024-07-01 by ShreyasKuntnal

MartyG-RealSense, 2024-07-01:
You are very welcome.  The VIDIOC error suggests a conflict between librealsense and the kernel, but if the streams are performing normally then it is likely just a momentary hiccup that can be ignored.  Thanks very much for the update!