realsenseai/librealsense #13101: How to calculate real-world distance from object mask and centroid (D405)?

qidihan, 2024-07-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                How to calculate real-world distance from object mask and centroid (D405)? |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D405 } | 
| Firmware Version                   | (The newest) | 
| Operating System & Version |   { Linux 22.04  | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { The newest }                          | 
| Language                            |  {Python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hello,

I am working on a computer vision project where I need to calculate the real-world distance that an object has moved based on its mask size and centroid coordinates in the image. I have the following information available:

I can extract the object mask (binary segmentation) from the image.
I have the centroid coordinates (x, y) of the object within the image.
I use 640*360 resolution. 
However, I am unsure how to convert these image-based measurements into real-world distances (e.g., in millimeters or meters). Can you help me on this?

Thanks,
Qidi

MartyG-RealSense, 2024-07-01:
Hi @qidihan  If you have XY image pixel coordinates then you can convert those coordinates to real-world distance in meters by converting an RGB color pixel to a depth pixel with the instruction **rs2_project_color_pixel_to_depth_pixel**.  Python examples of implementing this instruction can be found at https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008 and https://github.com/IntelRealSense/librealsense/issues/9945#issuecomment-972539640

qidihan, 2024-07-08:
I may not have been clear. What I meant was that I am using a 640*360 RGB image, and I want to know approximately how many millimeters a pixel is in this image, not the distance of my object from the camera. I found online that millimeters can be calculated as pixels multiplied by (25.4 / PPI), but I don't know the PPI value.

MartyG-RealSense, 2024-07-08:
There is not RealSense-specific information available about this.  My general non-RealSense research of the subject indicates though that in order to calculate PPI, you need to know the diagonal size of the screen that you are using.

https://www.calculatorsoup.com/calculators/technology/ppi-calculator.php

The **DPI** (Dots Per Inch) value is knowable though.  At 640x360 resolution, DPI should be **160**.

The guide below asserts in the section headed "Are PPI & DPI the same?" that PPI and DPI can be used interchangeably (1 PPI = 1 DPI), so that 160 DPI would be equal to 160 PPI.  Other resources back up the assertion that DPI and PPI are interchangeable.

https://cssunitconverter.com/pixels-to-inches/

qidihan, 2024-07-09:
Thanks for your information!!! I will close the issue right now.

closed 2024-07-09 by qidihan

MartyG-RealSense, 2024-07-09:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!