realsenseai/librealsense #13107: which sdk release to build on 24.04 ubuntu with d415 camera? with  force rsusb=true; thanks

AndreV84, 2024-07-01:
folks pointed out some instability with camera getting disappeared after some time
@MartyG-RealSense

MartyG-RealSense, 2024-07-02:
Hi @AndreV84  24.04 support will be in the next SDK release, as announced recently by my Intel RealSense colleagues at https://github.com/IntelRealSense/librealsense/discussions/13072

Whilst RSUSB = true can be used in situations where a Linux version is not yet officially supported, there have been a couple of recent cases where an RSUSB source code build failed on 24.04 whilst it worked on 22.04.

AndreV84, 2024-07-02:
@MartyG-RealSense thank you for following up!
could you also please answer to the question of my colleague "CUDA accelerates only point cloud generations?" ?
 Given I have built from sources like that 
`cmake ../ -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=true -DBUILD_WITH_CUDA=true -DCMAKE_CUDA_ARCHITECTURES=86`

Thanks
AV

MartyG-RealSense, 2024-07-02:
The CUDA support of librealsense accelerates point clouds, depth-color alignment and YUY to RGB color conversion.  It does not affect any other processes.

CUDA support is included and enabled by default in Intel's Jetson package installation (librealsense2-utils and librealsense2-dev) and in RSUSB = true source code installations.

MartyG-RealSense, 2024-07-02:
No, CUDA support was removed from the packages on the **distribution_linux** instruction page for x86 / x64 computers.

closed 2024-07-09 by AndreV84

MartyG-RealSense, 2024-07-09:
That's great to hear.  Thanks very much for the update!

AndreV84, 2024-07-09:
the Dockerfile  below did the trick along with passing arguments like privileged / --runtime=nvidia --gpus all; maybe further optimization could be to find a way for realsense camera to be accessed without using the privileged mode  :
```
FROM nvidia/cuda:12.5.0-devel-ubuntu22.04
ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    lsb-release \
    software-properties-common \
    cmake \
    wget \
    build-essential \
    libgtk-3-dev \
    libjpeg-dev \
    libopenexr-dev \
    unzip \
    libopencv-dev \
    git \
    gcc \
    openssl \
    libssl-dev  libusb-1.0-0-dev pkg-config libgtk-3-dev libusb-dev libssl3  libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev at
RUN git clone https://github.com/IntelRealSense/librealsense
RUN cd librealsense &&  mkdir build && cd build && cmake -DFORCE_RSUSB_BACKEND=ON  -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true  -DBUILD_WITH_CUDA=true -DCMAKE_CUDA_ARCHITECTURES=86 .. && make -j6 && make install
```

MartyG-RealSense, 2024-07-09:
Thanks so much for sharing your Dockerfile!