realsenseai/librealsense #13108: xioctl (VIDIOC_S_FMT) Error while running "ros2 run realsense2_camera realsense2_camera_node"

mosi3707, 2024-07-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | 5.12.14.50                                               | 
| Operating System & Version |  Linux (Ubuntu 22)                                  | 
| Kernel Version (Linux Only)    | 5.15.0-1046-raspi                                   | 
| Platform                                 | Raspberry Pi                                           |
| SDK Version                            |  2.55.1-1jammy                                     | 
| Segment			         |  Robot                                                   | 

When running the command "ros2 run realsense2_camera realsense2_camera_node", the following error screen appears.
[image]

I am trying to get a turtlebot3 system running the D435i camera, using a Raspberry Pi, and ROS2 humble. ROS2 humble has been setup following the steps on the official guide (https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html).

The realsense libraries where installed with the following steps.
1.	sudo add-apt-repository universe
2.	sudo add-apt-repository multiverse
3.	sudo add-apt-repository restricted
4.	sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
5.	curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add –
6.	sudo apt install ros-humble-librealsense2*
7.	sudo apt install ros-humble-realsense2-*

None of the above commands gave warnings. Then, on the terminal with no other ROS2 script running, using either "ros2 run realsense2_camera realsense2_camera_node" or "ros2 launch realsense2_camera rs_launch.py" both produce the same output.

The Line above the Error says "Connection timed out", but I am not sure why this would be happening. I have a USB - USB C 3.2 connection to the Pi's USB 3.0 port, so there shouldn't be any speed issues there. There are no other ROS2 nodes running, so from what I can tell, there are no other processes taking up processing power.

mosi3707, 2024-07-01:
Adding additional screenshots of version info
[image]

MartyG-RealSense, 2024-07-02:
Hi @mosi3707  Your second image lists your librealsense SDK and ROS2 wrapper versions as 2.55.1 and 4.55.1, which is a correct match.  However, the camera firmware driver version is listed as 5.12.14.50.  This firmware is designed for use with librealsense SDK 2.48.0.  The required firmware for librealsense 2.55.1 is **5.16.0.1**

[image]

mosi3707, 2024-07-02:
Ok, how would I upgrade to that exact version of firmware?

MartyG-RealSense, 2024-07-02:
If you are not able to use graphical tools such as **realsense-viewer** on your Pi then you can download the firmware **.bin** file for firmware 5.16.0.1 from the link below.

https://dev.intelrealsense.com/docs/firmware-releases-d400#d400-series-firmware-downloads

[image]

You can then use a text-based tool called **rs-fw-update** to perform the firmware update.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

Then initiate the firmware update using the downloaded .bin file with a command like the one below.

`rs-fw-update -s 725112060411 -f Signed_Image_UVC_5_16_0_1.bin`

Change the serial number after **-s** to the serial number of your particular camera.

*******

If you do have access to the realsense-viewer tool then once you have downloaded the 5.16.0.1 .bin file, you can go to the 'More' option near the top of the Viewer's options side-panel, select 'Update Firmware' from its drop down menu and navigate to the .bin file's location on your computer.  Then select the file to automatically initiate the firmware update.

mosi3707, 2024-07-02:
I downloaded the .bin file to the pi, but have a permission denied error when updating. Is there a preferred way to resolve this? Updating the permissions for accessing the camera, or adding rs-fw-update to the sudo path and running the command as sudo?

[image]

MartyG-RealSense, 2024-07-03:
Does it make a difference if you run rs-fw-update in sudo admin permissions mode by putting **sudo** on the front of the command?  For example:

`sudo rs-fw-update -s 725112060411 -f Signed_Image_UVC_5_16_0_1.bin`

mosi3707, 2024-07-03:
I have attempted several solutions to get the command to run as sudo. First, adding sudo to the rs-fw-update command will give a "Command not found" error. I discovered that this is because the sudo $PATH is not the same as default user's $PATH. To solve this, I added the path (/opt/ros/humble/bin) to the rs-fw-update command to the visudo file. I also added the path (/opt/ros/humble/lib/aarch64-linux-gnu) to .... the visudo file. 

[image]

Adding these paths to the visudo file allows the command to be run, but it will get the error "error while loading shared libraries: librealsense2.so.2.55: cannot open shared object file: No such file or directory"

I originally tried to solve this by adding the path to those libraries to the visudo file, but that did not fix it. I found that others had a similar issue when several versions of librealsense2 were installed. I attempted to move the librealsense2.so.2.55.1 file to a different directory to "hide" it so the command does not get confused on versions, but that had no change.

[image]

MartyG-RealSense, 2024-07-04:
As you are using a Raspberry Pi, the sudo-apt method `install ros-humble-librealsense2*` that was used to install librealsense would not be the best choice for Pi as this install method is suited to desktop and laptop computers with an x86 or x64 chip architecture rather than the Arm architecture of Pi.

I would recommend first focusing on getting librealsense working correctly on your Pi and look at the problem of installing the ROS wrapper afterwards.

A reliable way of installing librealsense on Pi is the **libuvc backend** method.  A build script for installing librealsense this way can be found at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Before running the build script, I also recommend uninstalling all packages on your computer with 'realsense' in their name using this command:

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

mosi3707, 2024-07-06:
It appears we are almost there to successfully running "ros2 run realsense2_camera realsense2_camera_node". Currently that command gives the following error

[image]

Here are the steps that I have taken and the current state of the Pi.

1. Running the steps from https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md failed due to lingering package conflict even after the purge command. I restarted and entirely remade the Pi along with ROS.
2. sudo add-apt-repository universe
3. sudo add-apt-repository multiverse
4. sudo add-apt-repository restricted
5. wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
6. chmod +x ./libuvc_installation.sh
7.  ./libuvc_installation.sh
8. Downloaded Signed_Image_UVC_5_16_0_1 from https://dev.intelrealsense.com/docs/firmware-releases-d400#d400-series-firmware-downloads
9. rs-fw-update -l    (to get the SN)
10. sudo rs-fw-update -s 048622071139 -f Signed_Image_UVC_5_16_0_1.bin
11. ros2 run realsense2_camera realsense2_camera_node

All above steps except the final one worked without errors. I suspect there is no realsense2_camera package installed since I skipped the two following steps 

sudo apt install ros-humble-librealsense2*
sudo apt install ros-humble-realsense2-*

Is there a way to 1. check if I have the realsense2_camera package installed and I simply do not have it on my PATH, and 2. If it is not installed, a way to install it without causing conflict with the libuvc workaround?

MartyG-RealSense, 2024-07-07:
You can list all installed packages in the Ubuntu terminal with the command `apt list --installed`

To list only packages with 'realsense' in their name, try this:

`apt list --installed | grep "realsense"`

***********

Regarding libuvc, that method builds librealsense from source code.  You should be able to install the ROS wrapper from packages with `sudo apt install ros-humble-realsense2-*` without causing a conflict with the libuvc installation of librealsense.

mosi3707, 2024-07-08:
It appears to have worked once. I ran "sudo apt install ros-humble-realsense2-*", then ran the ros2 run command. This created the exact same error messages as the start of this thread. Then I tried running ros2 launch, and that seemed to work.

Then, on a linux PC I ran "ros2 launch turtlebot3_bringup rviz2.launch.py" to try to see any output. I could get the image, but some of the features did not seem to be working. I checked back on the terminal connected to the pi, and I saw the same xioctl(VIDIOC_S_FMT) error as the ros2 run command.

[image]

I am not sure what this error message is relating to, since I am able to get an image in rviz that was successfully updating. Listing all topics on my linux PC gives

[image]

Furthermore, after trying ros2 run or ros2 launch a second time, I am met with the same error, and I cannot get it to run again. A full power cycle of the Pi does not resolve the issue. Below is a full copy of the error message for the ros2 launch command after a power cycle

`ubuntu@ubuntu:~$ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-07-08-01-59-41-845740-ubuntu-1166
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1168]
[realsense2_camera_node-1] [INFO] [1720403987.798943309] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1720403987.799774930] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1720403987.800019798] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1720403990.233124769] [camera.camera]: Device with serial number 048622071139 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1720403990.233585468] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1720403990.233783929] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1720403990.236561085] [camera.camera]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1720403990.236921766] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1720403990.241352147] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1720403990.255181596] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1720403990.255533926] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1720403990.255721535] [camera.camera]: Device Serial No: 048622071139
[realsense2_camera_node-1] [INFO] [1720403990.255879663] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1720403990.256044272] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1720403990.256202937] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1720403990.256347621] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1720403990.452767201] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1720403990.459378910] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1720403990.520301651] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1720403990.572262743] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1720403990.584128158] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1720403990.585800011] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [INFO] [1720403990.651927697] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1720403990.652688374] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1720403990.959159131] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1720403990.976948077] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1720403990.993707181] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [WARN] [1720403991.101734589] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[realsense2_camera_node-1]  08/07 01:59:56,222 ERROR [281472199878880] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/07 01:59:56,237 ERROR [281472199878880] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/07 01:59:56,241 ERROR [281472208333024] (librealsense-exception.h:52) xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
[... cut ...]
[realsense2_camera_node-1]  08/07 02:00:21,843 ERROR [281472174516448] (error-handling.cpp:99) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  08/07 02:00:26,963 ERROR [281472199878880] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/07 02:00:26,964 ERROR [281472199878880] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
`

MartyG-RealSense, 2024-07-08:
Does it make a difference if you add the **initial_reset:=true** parameter to your launches to reset the camera at launch?

**ros2 run**
`ros2 run realsense2_camera realsense2_camera_node --ros-args -p  initial_reset:=true`

**ros2 launch**
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

mosi3707, 2024-07-08:
No difference that I can see

```
ubuntu@ubuntu:~$ ros2 launch realsense2_camera rs_launch.py initial_reset:=true
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-07-08-15-17-38-957021-ubuntu-1390
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1391]
[realsense2_camera_node-1] [INFO] [1720451878.107866415] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1720451878.108587278] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1720451878.108825085] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1720451880.549869719] [camera.camera]: Device with serial number 048622071139 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1720451880.550325611] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1720451880.550526790] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1720451880.553569539] [camera.camera]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1720451880.553943619] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1720451880.554124946] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1720451889.084386044] [camera.camera]: Device with serial number 048622071139 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1720451889.084771088] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1720451889.084956731] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1720451889.086943506] [camera.camera]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1720451889.087402197] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1720451889.088453573] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1720451889.100792009] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1720451889.101129869] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1720451889.101359362] [camera.camera]: Device Serial No: 048622071139
[realsense2_camera_node-1] [INFO] [1720451889.101554560] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1720451889.101749703] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1720451889.101926827] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1720451889.102100507] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1720451889.296164145] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1720451889.302273578] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1720451889.345315393] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1720451889.399863686] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1720451889.411779004] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1720451889.414636180] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [INFO] [1720451889.478805006] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1720451889.479615890] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1720451889.607218250] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1720451889.625139503] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1720451889.639421956] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [WARN] [1720451889.749413072] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[realsense2_camera_node-1]  08/07 15:18:14,820 ERROR [281472132770016] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/07 15:18:14,820 ERROR [281472132770016] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/07 15:18:14,838 ERROR [281472342550752] (librealsense-exception.h:52) xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
[realsense2_camera_node-1]  08/07 15:18:14,839 ERROR [281472342550752] (rs.cpp:237) [rs2_open_multiple( sensor:0xaaaaecef2cd0, profiles:0xffff5805fb80, count:1 ) Backend] xioctl(VIDIOC_S_FMT) failed, errno=5 Last  [... cut ...]
[ERROR] [realsense2_camera_node-1]: process[realsense2_camera_node-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [realsense2_camera_node-1]: sending signal 'SIGKILL' to process[realsense2_camera_node-1]
[ERROR] [realsense2_camera_node-1]: process has died [pid 1391, exit code -9, cmd '/opt/ros/humble/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_w_fuutzg --params-file /tmp/launch_params_lg82mxs9'].
ubuntu@ubuntu:~$
```

mosi3707, 2024-07-10:
After a couple tries with different iterations of ros2 run and ros2 launch with the flags global_time_enabled:=false and initial_reset:=true, I had it function once, but I have not been able to reproduce it. Here is the output from the temporarily successful run

```
ubuntu@ubuntu:~$ ros2 run realsense2_camera realsense2_camera_node --ros-args -p global_time_enabled:=false
[INFO] [1720540134.096741712] [camera.camera]: RealSense ROS v4.55.1
[INFO] [1720540134.097143170] [camera.camera]: Built with LibRealSense v2.55.1
[INFO] [1720540134.097247873] [camera.camera]: Running with LibRealSense v2.55.1
[INFO] [1720540136.435619901] [camera.camera]: Device with serial number 048622071139 was found.

[INFO] [1720540136.436081470] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[INFO] [1720540136.436271190] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [1720540136.439540764] [camera.camera]: Device with port number 2-2 was found.
[INFO] [1720540136.439817779] [camera.camera]: Device USB type: 3.2
[INFO] [1720540136.440387329] [camera.camera]: getParameters...
[INFO] [1720540136.450740933] [camera.camera]: JSON file is not provided
[INFO] [1720540136.451079429] [camera.camera]: Device Name: Intel RealSense D435I
[INFO] [1720540136.451269612] [camera.camera]: Device Serial No: 048622071139
[INFO] [1720540136.451446499] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0
[INFO] [1720540136.451998956] [camera.camera]: Device FW version: 5.16.0.1
[INFO] [1720540136.452338545] [camera.camera]: Device Product ID: 0x0B3A
[INFO] [1720540136.452506617] [camera.camera]: Sync Mode: Off
[WARN] [1720540136.677615541] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[INFO] [1720540136.801759075] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [1720540136.802630676] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [1720540136.802904840] [camera.camera]: Stopping Sensor: Motion Module
[INFO] [1720540136.957375548] [camera.camera]: Starting Sensor: Depth Module
[INFO] [1720540136.995018657] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1720540136.995436152] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1720540136.995715501] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1720540137.008799667] [camera.camera]: Starting Sensor: RGB Camera
[INFO] [1720540137.038432497] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
 09/07 15:48:57,048 WARNING [281473139402976] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[INFO] [1720540137.064654014] [camera.camera]: Starting Sensor: Motion Module
[INFO] [1720540137.090875365] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[INFO] [1720540137.091230972] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[INFO] [1720540137.136630864] [camera.camera]: RealSense Node Is Up!
[WARN] [1720540137.959984588] [camera.camera]: XXX Hardware Notification:Motion Module failure,1.72054e+12,Error,Hardware Error
[WARN] [1720540137.960602322] [camera.camera]: Hardware Notification:Motion Module failure,1.72054e+12,Error,Hardware Error
 09/07 15:50:05,758 ERROR [281473130948832] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:05,758 ERROR [281473130948832] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:10,898 ERROR [281472896329952] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:10,898 ERROR [281472896329952] (error-handling.cpp:99) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:16,024 ERROR [281473130948832] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:16,025 ERROR [281473130948832] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:21,138 ERROR [281472896329952] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:21,138 ERROR [281472896329952] (error-handling.cpp:99) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:26,259 ERROR [281473130948832] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 09/07 15:50:26,260 ERROR [281473130948832] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[... cut ...]
[ERROR] [realsense2_camera_node-1]: process[realsense2_camera_node-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [realsense2_camera_node-1]: sending signal 'SIGKILL' to process[realsense2_camera_node-1]
[ERROR] [realsense2_camera_node-1]: process has died [pid 2326, exit code -9, cmd '/opt/ros/humble/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_n_g66i9r --params-file /tmp/launch_params_4_ay44mg'].
```

MartyG-RealSense, 2024-07-10:
it is encouraging that you were able to launch the ROS wrapper successfully at least once.  THe problem may be with the Raspberry Pi board itself rather than the camera or the SDK / wrapper software.  It is difficult to get librealsense working on Pi boards, and getting the ROS wrapper working on Pi afterwards can be even more complicated.

In these situations, I recommend focusing on getting librealsense working first and then looking at the wrapper installation once you have s stable librealsense build.  On Pi though the best that can be expected is streaming basic depth and RGB streams.  The RealSense IMU is usually not accessible on Pi.

mosi3707, 2024-07-11:
Perfect, I don't really need IMU data. Do we suspect that the access to IMU is causing these issues? Is there a way to run the node to only access RGB? Once that is working, is there a way to only attempt to access RBG + Depth information, since that would be the extent of what I would like working for my project? Possibly even a solution to do a ros2 launch command to start an RGB camera topic, then open another terminal and do a ros2 launch command to start a depth topic?

MartyG-RealSense, 2024-07-11:
You can control which streams are enabled and which are left disabled by including configuration statements in your ROS2 launch instruction.

You do not need to disable the IMU and infrared streams as they are already disabled in the launch file by default.

**RGB only**
`ros2 launch realsense2_camera rs_launch.py enable_color:=true enable_depth:=false`

**RGB and Depth**
`ros2 launch realsense2_camera rs_launch.py enable_color:=true enable_depth:=true`

mosi3707, 2024-07-12:
I can have repeatable success, with some strange side effects. After playing around with only enabling one sensor at a time, I saw that I could get any single sensor running (RBG, Depth, even infra, even though my rviz would not show it). If I enabled any two, it would fail. I then found #5628 and decided to try to change the timeouts. I'm not sure if I'm using the correct settings here, but I changed wait_for_device_timeout and reconnect_timeout and managed to get depth and RBG running together. I'm not sure how many I can stream at the same time, but for this project I only need those two.

Here is the command I used to successfully run the node.

`ros2 launch realsense2_camera rs_launch.py enable_color:=true enable_depth:=true wait_for_device_timeout:=10. reconnect_timeout:=10.`

And here is the successful output
```

ubuntu@ubuntu:~$ ros2 launch realsense2_camera rs_launch.py enable_color:=true enable_depth:=true wait_for_device_timeout:=10. reconnect_timeout:=10.
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-07-12-02-48-40-374088-ubuntu-1691
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1692]
[realsense2_camera_node-1] [INFO] [1720752524.471429327] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1720752524.472158633] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1720752524.472387057] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1720752526.907759294] [camera.camera]: Device with serial number 048622071139 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1720752526.908219826] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1                  .0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1720752526.908422879] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1720752526.912301114] [camera.camera]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1720752526.912679925] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1720752526.913255752] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1720752526.930044883] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1720752526.930389897] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1720752526.930610117] [camera.camera]: Device Serial No: 048622071139
[realsense2_camera_node-1] [INFO] [1720752526.930806708] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0                  /video4linux/video0
[realsense2_camera_node-1] [INFO] [1720752526.931043705] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1720752526.931243370] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1720752526.931430201] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1720752527.136857226] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1720752527.144111424] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1720752527.187952092] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequen                  cy' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1720752527.242561215] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1720752527.258686687] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1720752527.260779275] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [INFO] [1720752527.352586546] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1720752527.353366945] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1720752527.483521944] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1720752527.502941231] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1720752527.519345125] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1720752527.551840435] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1720752527.579764777] [camera.camera]: RealSense Node Is Up!
```
It is possible to sometimes get this warning message

`[realsense2_camera_node-1] [WARN] [1720754006.997548686] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.`

and when this warning message appears, then this error can follow, and it the pi/camera will get into a bad state where I cannot get RGB+Depth to run again.

```
[realsense2_camera_node-1] [WARN] [1720754006.997548686] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[... cut ...] with the timeout values set, and it will run successfully. If I attempt to run RGB+Depth immediately after a power cycle without running individually first it will fail every time.

Overall, I can make this work for my project, I will just have to power cycle and run a weird setup procedure . Is there any settings you would recommend playing with to make this more robust? I would prefer to see if there is a more robust and straight forward method than the one I created before I close this issue.

MartyG-RealSense, 2024-07-12:
Thanks so much for sharing the details of the adjustments that enabled you to achieve two streams on your Raspberry Pi.

There is an alternative method to an external power source for powering a Pi called a Power Over Ethernet (PoE) Hat, as described at https://github.com/IntelRealSense/librealsense/issues/8274#issuecomment-769807239

mosi3707, 2024-07-14:
Since the method I described has been stable these last few days, I think this issue is resolved and can be closed.

closed 2024-07-14 by mosi3707

MartyG-RealSense, 2024-07-15:
That's great to hear that your system has been stable.  Thanks very much for the update!