realsenseai/librealsense #13110: Install librealsense in podman

1005088h, 2024-07-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.16.0.1) | 
| Operating System & Version |   Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15.0-107-generic                                     | 
| Platform                                 | PC  |
| SDK Version                            |  5.16 }                          | 
| Language                            |  C++                          | 
| Segment			|  {Robot }                          | 

### Issue Description

I want to install librealsense in a docker container, but there is no very clear instructions how to configure host and container.

Understand that if we install the pre-build package following https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. We need to install `librealsense2-dkms`. However, if we build from source following https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md, there is not mention that we need to install `librealsense2-dkms`. Instead, there is a script `./scripts/patch-realsense-ubuntu-lts-hwe.sh` need to patch the kernel when build from source.

In the dockerfile here https://github.com/2b-t/realsense-ros2-docker?tab=readme-ov-file#12-installation-from-source and https://github.com/IntelRealSense/librealsense/blob/master/scripts/Docker/Dockerfile. There are not requirement to install the dkms or patch the kernel.

I also saw adding ` -DFORCE_RSUSB_BACKEND=TRUE ` https://github.com/IntelRealSense/librealsense/issues/12158#issuecomment-1702787288 then no need to install dkms?

I have some questions:

1. Do we need to install `librealsense2-dkms` or run  `./scripts/patch-realsense-ubuntu-lts-hwe.sh` in the host or containe and why.
2. How does '99-realsense-libusb.rules ' in the container affect the host, is it necessary to copy it to the container?
3. Do we need ` -DFORCE_RSUSB_BACKEND=TRUE ` while I saw both dockerfile do not set it
4. What exactly we need to configure for the host and container?

MartyG-RealSense, 2024-07-02:
There are not many references available for installing only librealsense in a Docker container.  The Intel guide that you linked to is one of the best available,

https://github.com/IntelRealSense/librealsense/blob/master/scripts/Docker/Dockerfile

Guides to Docker container installation that are shared by RealSense users are usually for librealsense + ROS, like the one at https://github.com/IntelRealSense/realsense-ros/issues/388#issuecomment-410672421

******************

Packages such as `librealsense2-dkms` are not used when building the librealsense SDK from source code with CMake with the **installation.md** instructions.  This includes source code builds that use the flag `-DFORCE_RSUSB_BACKEND=TRUE`

The SDK should be built either from packages or source code but not both installation methods on the same computer, otherwise there could be more than one librealsense installation on the same computer that conflict with each other.  

Installing with both packages and source code on the same computer can also result in two different sets of **udev** device handling rules being installed, which can cause an error called **Multiple realsense udev-rules were found!**

The two rules are named `/etc/udev/rules.d/99-realsense-libusb.rules` and `/lib/udev/rules.d/60-librealsense2-udev-rules.rules`.  The **/lib** '60' rule is created by a dkms package installation and the **/etc** '99' rule is created by a source code installation.

If you have both types of rules on your computer then the **/etc** one is the one that should be deleted, as deleting the /lib rule can result in dkms packages becoming broken.

MartyG-RealSense, 2024-07-09:
Do you require further assistance with this case, please?  Thanks!

1005088h, 2024-07-09:
Thanks for your replied, I have done the experiment that, `99-realsense-libusb.rules` has to be installed in the host to setup the permission.

1005088h, 2024-07-09:
I have another question. How to determine whether the camera is D435IF or D435I from the SDK? We can get device name from `RS2_CAMERA_INFO_NAME` but both  D435IF or D435I displayed as D435I

MartyG-RealSense, 2024-07-10:
Cameras with a filter on their sensors do not have 'f' in their name, so a D435f will be D435 and D435if will be D435i.  The camera's firmware driver works out the name of the product based on the components that it can detect, such as RGB sensor and IMU.  A filter is not a component that the firmware can detect.

closed 2024-07-24 by MartyG-RealSense